Koenkk/zigbee2mqtt #15010: Legrand 600086: the position is not updated either !

novisys, 2022-11-15:
### What happened?

        @Koenkk 
I tested under docker on the image koenkk/zigbee2mqtt:latest-dev and it does not work. 
Initial status with the shutter open:
[image]

When I click on the closed button:
[image]

When I click on the synchronization button:
[image]

_Originally posted by @Alindil in https://github.com/Koenkk/zigbee2mqtt/issues/13566#issuecomment-1219599525_
     Still have the same problem, any update about this ?
Thanks

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

novisys, 2022-11-19:
Any help ? @Koenkk

Koenkk, 2022-11-20:
Can you try changing the reporting of `currentPositionLiftPercentage` of the `closuresWindowCovering` in the "reporting" tab of the device?

novisys, 2022-11-20:
> Can you try changing the reporting of `currentPositionLiftPercentage` of the `closuresWindowCovering` in the "reporting" tab of the device?

hi, thanks for the response
but this option don't exist on the list of the Closures Options 
i have: 
- windowCoverrigType
- windowCoverrigMode

novisys, 2022-11-20:
Here is the options possible and working on ConsoleDev:
`2022-11-20 12:23:22Read result of 'closuresWindowCovering': {"windowCoveringMode":0,"windowCoveringType":2,"configStatus":3,"currentPositionLiftPercentage":0,"installedOpenLimitLiftCm":0,"installedClosedLimitLiftCm":65535}`
[image]

Koenkk, 2022-11-20:
Not the dev tab but the reporting one

novisys, 2022-11-20:
> Not the dev tab but the reporting one

I know !
to be more clear, here is a screenshot withe error get when trying to change to the value you telled:
[image]
[image]
[image]

Koenkk, 2022-11-20:
Seems this cover cannot reports its position. One thing we can do is to always return position 0 when you send CLOSE and 100 when you send OPEN.

novisys, 2022-11-20:
> Seems this cover cannot reports its position. One thing we can do is to always return position 0 when you send CLOSE and 100 when you send OPEN.

you are true,
i passed my week learning Legrand documentation and other Deconz ressources and conclude that this one don't support position %  👎

closed 2022-12-28 by github-actions[bot]