Koenkk/zigbee2mqtt #15014: [New device support]: 45856GE Energy Monitoring Missing

coolspot18, 2022-11-16:
### Link

https://www.zigbee2mqtt.io/devices/45856GE.html

### Database entry

{ 	"id": 33, 	"type": "Router", 	"ieeeAddr": "0x0022a300002535d9", 	"nwkAddr": 62324, 	"manufId": 4388, 	"manufName": "Jasco Products", 	"powerSource": "Mains (single phase)", 	"modelId": "45856", 	"epList": [1, 2], 	"endpoints": { 		"1": { 			"profId": 260, 			"epId": 1, 			"devId": 256, 			"inClusterList": [0, 3, 4, 5, 6, 2821, 1794], 			"outClusterList": [10, 25], 			"clusters": { 				"genBasic": { 					"attributes": { 						"modelId": "45856", 						"manufacturerName": "Jasco Products", 						"powerSource": 1, 						"zclVersion": 1, 						"appVersion": 18, 						"stackVersion": 80, 						"hwVersion": 1 					} 				} 			}, 			"binds": [{ 				"cluster": 6, 				"type": "endpoint", 				"deviceIeeeAddress": "0x00124b0026b698e5", 				"endpointID": 1 			}], 			"configuredReportings": [{ 				"cluster": 6, 				"attrId": 0, 				"minRepIntval": 0, 				"maxRepIntval": 3600, 				"repChange": 0 			}], 			"meta": {} 		}, 		"2": { 			"profId": 260, 			"epId": 2, 			"devId": 259, 			"inClusterList": [0, 3, 2821], 			"outClusterList": [3, 6], 			"clusters": { 				"genBasic": { 					"attributes": {} 				} 			}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"appVersion": 18, 	"stackVersion": 80, 	"hwVersion": 1, 	"zclVersion": 1, 	"interviewCompleted": true, 	"meta": { 		"configured": 1 	}, 	"lastSeen": 1668590701451, 	"defaultSendRequestWhen": "immediate" }

### Comments

This device is supported for basic on/off, but energy monitoring information is not reported by Zigbee2MQTT. 

According to the BlackAdder documentation, deConz supports energy monitoring.

Is it possible to have energy information added to Z2M?

Koenkk, 2022-11-17:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-17 by Koenkk

coolspot18, 2022-11-18:
> Added, thanks!
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi @Koenkk I noticed OnOff reporting is not enabled on the switch, but even though I added it to the converter? The convert might need a small fix.

[image]

Koenkk, 2022-11-20:
The onOff reporting was not in your converter, only the bind. I added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)