Koenkk/zigbee2mqtt #15025: Z2M restart every 24 hour, possible due to UNSUPPORTED_ATTRIBUTE in Sunricher ZG9092A thermostat

Tenkc, 2022-11-16:
### What happened?

Every 24 hours Z2M restarts, I believe it's because a Sunricher ZG9092A thermostat triggers this with an UNSUPPORTED_ATTRIBUTE .

Already tried resetting the thermostat to factory and rejoin and upgrading the adapter firmware to the latest dev version.

### What did you expect to happen?

No unsupported attribute and no restarting.

### How to reproduce it (minimal and precise)

Just wait 24 hours...

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20221102

### Adapter

Slaesh CC2652RB

### Debug log

Cleaned up the log a bit...

Error: Write 0xffff030201030259/1 genTime({"time":721819591}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22)
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-11-15 09:26:36: Logging to console and directory: '/app/data/log/2022-11-15.09-26-36' filename: log.txt
Zigbee2MQTT:info  2022-11-15 09:26:36: Starting Zigbee2MQTT version 1.28.2 (commit #360a777)
Zigbee2MQTT:info  2022-11-15 09:26:36: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:info  2022-11-15 09:26:50: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-11-15 09:26:50: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20221102,"transportrev":2},"type":"zStack3x0"}'

Zigbee2MQTT:info  2022-11-15 09:26:50: Keuken vloerverwarming (0xffff030201030259): SR-ZG9092A - Sunricher Touch thermostat (Router)

Zigbee2MQTT:warn  2022-11-15 09:26:50: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-11-15 09:26:50: Allowing new devices to join.
Zigbee2MQTT:warn  2022-11-15 09:26:50: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-11-15 09:26:50: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-11-15 09:26:50: Connecting to MQTT server at mqtt://192.168.50.26
Zigbee2MQTT:info  2022-11-15 09:26:50: Connected to MQTT server
Zigbee2MQTT:info  2022-11-15 09:26:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-11-15 09:26:51: Configuring 'Keuken vloerverwarming'
Zigbee2MQTT:error 2022-11-15 09:26:52: Failed to configure 'Keuken vloerverwarming', attempt 1 (Error: ConfigureReporting 0xffff030201030259/1 hvacUserInterfaceCfg([{"attribute":"keypadLockout","minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":null}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.thermostatKeypadLockMode (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:143:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/sunricher.js:465:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.start (/app/lib/extension/configure.ts:68:13)
    at Controller.callExtensions (/app/lib/controller.ts:314:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:109:5))
Zigbee2MQTT:info  2022-11-15 09:26:52: Started frontend on port 0.0.0.0:8889

Koenkk, 2022-11-17:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/d844713c2fe84ae40163241e38f7affe
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Tenkc, 2022-11-18:
Thanks a lot! Looks look this fixed it, there was no restart, and I don't see the UNSUPPORTED_ATTRIBUTE error in the logs.

I'll wait another 24 hrs to confirm.

Koenkk, 2022-11-20:
Great, let me know

Tenkc, 2022-11-21:
So, a bit more than 24 hours, still no restarts.

Thanks for the fix!

Koenkk, 2022-11-21:
pushed the fix! You can remove the external converter with the next release on 1 December.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-21 by Koenkk