Koenkk/zigbee2mqtt #15027: Sonoff SNZB-04 devices periodically leave network.

SeanCline, 2022-11-16:
### What happened?

I have a very small network of 4 devices connected to a [Sonoff ZBDongle-E](https://www.zigbee2mqtt.io/guide/adapters/#based-on-silicon-labs-efr32mg2x-mgm21x-and-efr32mg1x-mgm1x-series). The two [Sonoff SNZB-01](https://www.zigbee2mqtt.io/devices/SNZB-01.html) buttons have been rock solid, but the [Sonoff SNZB-04](https://www.zigbee2mqtt.io/devices/SNZB-04.html) contact sensors periodically fall off the network about once a day and need to be re-paired with the coordinator.

I've taken the steps documented in [Improve network range and stability](https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html). Picking a zigbee channel away from my wifi channel, using a USB extension cable, and changing adapter orientation all have no effect. Link quality typically stays above 150, yet the SNZB-04's still leave the network, often at the same time.

Below, I've attached the full zigbee-herdsman log along with the snippet of of the devices leaving the network.

I am unsure whether there is something wrong with my setup or whether there is some software issue at play. Any ideas that would get to the root of the problem would be welcome.

### What did you expect to happen?

The SNZB-04 devices should be as stable as the SNZB-01 devices on the same network.

### How to reproduce it (minimal and precise)

Pair an SNZB-04 to a ZBDongle-E and wait about a day.

### Zigbee2MQTT version

1.28.2-dev commit: 16398b7

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

### Full log:
[z2m_edge.log](https://github.com/Koenkk/zigbee2mqtt/files/10023024/z2m_edge.log)

### Snippet of devices leaving:
```
2022-11-16T07:41:23.185Z zigbee-herdsman:adapter:ezsp:uart <-- [2350b1a9772a15b2d2a9003de77092028e4e233e8c7e]
2022-11-16T07:41:23.185Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,3,0): 2350b1a9772a15b2d2a9003de77092028e4e233e8c7e
2022-11-16T07:41:23.186Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (3)
2022-11-16T07:41:23.186Z zigbee-herdsman:adapter:ezsp:uart --> [83401b7e]
2022-11-16T07:41:23.186Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 129001230000008b3d4a184d25004b120004
2022-11-16T07:41:23.187Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x23: {"_id_":35,"_cls_":"childJoinHandler","_isRequest_":false,"index":0,"joining":0,"childId":15755,"childEui64":{"type":"Buffer","data":[0,18,75,0,37,77,24,74]},"childType":4}
2022-11-16T07:41:23.187Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 15755 00124b00254d184a
2022-11-16T07:41:23.187Z zigbee-herdsman:controller:log Device leave '0x00124b00254d184a'
2022-11-16T07:41:23.188Z zigbee-herdsman:controller:log Removing device from database '0x00124b00254d184a'
Zigbee2MQTT:warn  2022-11-16 02:41:23: Device 'FrontDoor' left the network
Zigbee2MQTT:info  2022-11-16 02:41:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"FrontDoor","ieee_address":"0x00124b00254d184a"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-11-16 02:41:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b00254d184a"},"type":"device_removed"}'
2022-11-16T07:41:23.989Z zigbee-herdsman:adapter:ezsp:uart <-- [3350b1a9772a14b25eec75e5e27092028e4e230c347e]
2022-11-16T07:41:23.990Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (3,3,0): 3350b1a9772a14b25eec75e5e27092028e4e230c347e
2022-11-16T07:41:23.990Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (4)
2022-11-16T07:41:23.990Z zigbee-herdsman:adapter:ezsp:uart --> [8430fc7e]
2022-11-16T07:41:23.990Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1290012300010007783fc04825004b120004
2022-11-16T07:41:23.990Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x23: {"_id_":35,"_cls_":"childJoinHandler","_isRequest_":false,"index":1,"joining":0,"childId":30727,"childEui64":{"type":"Buffer","data":[0,18,75,0,37,72,192,63]},"childType":4}
2022-11-16T07:41:23.991Z zigbee-herdsman:adapter:ezsp:debg Device left network request received: 30727 00124b002548c03f
2022-11-16T07:41:23.991Z zigbee-herdsman:controller:log Device leave '0x00124b002548c03f'
2022-11-16T07:41:23.991Z zigbee-herdsman:controller:log Removing device from database '0x00124b002548c03f'
Zigbee2MQTT:warn  2022-11-16 02:41:23: Device 'FrontDoorLock' left the network
Zigbee2MQTT:info  2022-11-16 02:41:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"FrontDoorLock","ieee_address":"0x00124b002548c03f"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-11-16 02:41:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x00124b002548c03f"},"type":"device_removed"}'
```

SeanCline, 2022-11-23:
### Update
I've recreated the Zigbee network with a new panID, and re-paired all 4 devices.  Now, even the SNZB-04 buttons, which were stable, fall off the network exactly 256 minutes after they were last pressed. I've since disabled Z2M and added my 4 devices to ZHA, where they've stated on the network for the last week.

This is appears to be regression in ZBDongle-E support, perhaps related to #15066.

marinus61, 2022-11-24:
Hi SeanCline

I experience the same on one of my HA z2m installations. But I have since 8.11.2022 filed a error report and refiled it monday, but no one seems to be looking into these error reports.

filed under 14818 and 15068 (i though I had cancelled the first one, but seems still active)

So if someone does comment on yours, I would appriciate a message
Marinus

kirovilya, 2022-11-24:
@SeanCline @marinus61  I saw your error messages. but I can’t quickly restore the adapter’s work, because I don’t find these errors in myself installation.

@SeanCline I see you tried to learn something about this error. can you test the guess and restore setup CONFIG_END_DEVICE_POLL_TIMEOUT = 14 setting? then test.
I removed the parameter settings in order to use the parameters from the firmware, I hoped that this would be more correct.

kirovilya, 2022-11-24:
Made changes, in the hope that the loss of devices will leave

kirovilya, 2022-11-24:
the changes are merged. try to check the work on the updated dev / edge version of z2m

marinus61, 2022-11-25:
@kirovilya the CONFIG_END_DEVICE_POLL_TIMEOUT = 14 is that in z2m config file or in the GUI to be added?

Marinus

kirovilya, 2022-11-25:
@marinus61 no. this needs to be edited in the zigbee-herdsman file... the changes have already been merged into the dev|edge branch and you can update, without file editing

marinus61, 2022-11-25:
@kirovilya 
thks for the quit reply, I will try when returning home tonight.
Appriciated that you assist us here :)

marinus61, 2022-11-25:
@kirovilya
thks for the quit reply, i have looked intot he github and see that there is an update to 0.14.76. I am on 0.14.68
However I am unable to find out howto update to latest level, since I installed Z2M via HACS and see no update here that has not been performed.
Sorry to bather with low level questions
Appriciated that you assist us here :)

redsam69, 2022-11-26:
Hi, I also join the question how to upgrade zigbee-herdsman to version 0.14.76?

marinus61, 2022-11-26:
@redsam69 

As I havebeen able to find on the github written by Koenkk, seems we need to use Z2M EDGE (experimental/beta) to be able touse the "in between" updates.

marinus61, 2022-11-26:
> ### Update
> I recreated the Z2M network with a new panID, and re-paired all 4 devices. Now, even the SNZB-01 buttons, which were stable, fall off the network exactly 256 minutes after they were last pressed. I've since disabled Z2M and added my 4 devices to ZHA, where they've stayed on the network for the last week.
> 
> This is appears to be regression in ZBDongle-E support, perhaps related to #15066.

@SeanCline 
Yes it is the same error I face and several other. Funny though that after SB-02 "left" the system, it suddenly reappeared, although with an error
In order to repair with success, i need to pair, remove and new pair

Dominator86, 2022-11-29:
I switched to Z2M Edge and my SNZB-01 buttons are still leaving the network.
Interestingly one of the buttons seems to have rejoined on its own although I haven't permitted joining.
It's showing a red exclamation mark which says "Interview failed" when hovering.
But the weirdest thing by far is that it seems to be bound to the closest Aqara smart plug now.
You can probably imagine my surprise when I pressed the button and my PC turned off.

marinus61, 2022-11-29:
> I switched to Z2M Edge and my SNZB-01 buttons are still leaving the network. Interestingly one of the buttons seems to have rejoined on its own although I haven't permitted joining. It's showing a [...]

@Dominator86 
thks for the update, I have chosen not to go for EDGE and wait a little longer for an update from @koenkk to the stable version. If no solution is found I am thinking switching to ZHA 

Meantime I installed some IKEA bulbs, and the SNZB-02 that suddently reappeared after some time had the same exclamation mark on it. In order to rejoin correct, I have to join the unit, delete it and join again, then it is back to normal for 256 minutes

marinus61, 2022-11-29:
@Dominator86 

Seems that the error only exists when pairing direct to the Sonoff dongle E, the system infected is not my main residens, so I will test further when I get back there, but the units paired to the IKEA bulp seems to stay online but with the exclamtion mark on it (did not remove and repair it, so I will try that, to see if the X mark vanished)

Dominator86, 2022-11-30:
> Yes, if i pair with aqara plug, everything works seeamless

The problem with that is that in my case the button then toggles the plug (even though it's bound to the coordinator, not the plug) but I need my buttons to control lights.

When I pair my buttons with the dongle or my Hue and Tradfri bulbs instead they don't stay in the network.

marinus61, 2022-11-30:
@kirovilya 

any chance that you could assist in how we can make this work or how to update the files without going via EDGE 
Marinus

kirovilya, 2022-12-08:
@marinus61 I'm at a loss... Sonoff sensors are so capricious.
What firmware version do you have? 7.1.1

kirovilya, 2022-12-08:
@marinus61 Not this. 
[image]

marinus61, 2022-12-08:
@kirovilya 

thanks for fast reply.
[image]

marinus61, 2022-12-09:
@kirovilya 

I found the version you had, the one I had installed was the one I had foundon Sonoff link.

Update went well, SNZB-2 and -4 are now showing all info after delete and repair.

I will update tomorrow if the link stays on

Thks for your support
Marinus

marinus61, 2022-12-10:
@kirovilya 

Sonoff SNZB-04 still drop connection to system, so not even the latest updates secure the stability of the connection

Any further ideas

Marinus

marinus61, 2022-12-20:
@kirovilya 

Hi again, have you had a chance to look into this persisting problem

@Dominator86 is your SNZB still online, mine drop out even when paired with other Zigbee units that the Sonoff stick

Marinus

kirovilya, 2022-12-20:
@marinus61 I don't have a solution yet, but I'm trying to find it. I'm seeing a similar problem with Legrand buttons

marinus61, 2022-12-20:
@kirovilya 

thks for the update, i will cross my fingers, let me know if you need any log files
Marinus

Dominator86, 2022-12-20:
> @Dominator86 is your SNZB still online, mine drop out even when paired with other Zigbee units that the Sonoff stick

@marinus61 I "fixed" my issue by flashing my old Sonoff ZBBridge with router firmware and pairing the buttons with that instead of the Dongle-E.

kminhtri, 2022-12-20:
Hello everybody,

Same issue here with a SNZB-04 (multiple disconnection) and sometimes the SNZB-02 disappears too.
I also have the Sonoff Dongle-E, here the firmware version :
[image]
and fyi, I also sync an aqara door sensor -> zero disconnection, seems the sonoff sensors are really capricious !
If I can provide any help or test with my devices, let me know :)

marinus61, 2022-12-21:
> > @Dominator86 is your SNZB still online, mine drop out even when paired with other Zigbee units that the Sonoff stick
> 
> @marinus61 I "fixed" my issue by flashing my old Sonoff ZBBridge with router firmware and pairing the buttons with that instead of the Dongle-E.

@Dominator86 , I am thinking to go same way or maybe go to ZHA, I prefer Z2M, but seems this problem might be there to stay, thks for the update

marinus61, 2022-12-22:
> Hello everybody,
>
> Same issue here with a SNZB-04 (multiple disconnection) and sometimes the SNZB-02 disappears too. I also have the Sonoff Dongle-E, here the firmware version : [image] and fyi, I [...]

@kminhtri, updating firmware to latest ersion (see above) solved issues with  SNZB-02 dropping out, only SNZB-02 is still an issue

Casper-O, 2023-01-03:
Also having issues with SNZB-04 showing up with red exclamation mark which says "Interview failed" when hovering and later disapear.

Dongle: ZBDongle-E

[image]

VS-X, 2023-01-04:
I found a workaround for my case. For all affected so off devices I went into the Reporting tab, and changed the max rep interval to 7200 for the battery attribute. This way buttons are reporting something at least every 2 hours, and are not disappearing from the network.
Not ideal, but seems to work

Koenkk, 2023-01-04:
Seems similar to https://github.com/Koenkk/zigbee2mqtt/issues/13600 which is fixed now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Casper-O, 2023-01-04:
I have installed Edge version now and installed a test sensor that is laying on my desk. I will come back with result tomorrow

marinus61, 2023-01-05:
> Seems similar to #13600 which is fixed now. Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@Koenkk 
any idea when it will be on general update, for us not so super programmers like you
Marinus

silvinojr, 2023-01-05:
I hope I'm not hijacking the thread.
My setup has a ZBDongle-P as coordinator and the SNZB-04s are also dropping from the network after a few hours.
Same results with a few router models, including a Sonoff Bridge flashed as router.
Once it becomes unresponsive, the only way to make it work again is to force remove the device and re-pair.

Zigbee2MQTT version: 1.29.0 commit: [unknown]
Coordinator type: zStack3x0
Coordinator revision: 20220219
Frontend version: 0.6.121

Casper-O, 2023-01-05:
Update:

I been on the Edge version since yesterday evening - it seems to solve the issue.

marinus61, 2023-01-06:
@Koenkk 
thank you for that.
marinus

Koenkk, 2023-01-06:
Release is out (1.29.1), assuming this can be closed now.

closed 2023-01-06 by Koenkk

marinus61, 2023-01-06:
> Release is out (1.29.1), assuming this can be closed now.

Hi @Koenkk 
installed, great work, thanks for the support. I will revert if problem re surfaces.

kminhtri, 2023-01-06:
Hello,

I'm on edge version : 1.29.0-dev commit: [cf80134](https://github.com/Koenkk/zigbee2mqtt/commit/cf80134)
Looks to be stable for sonoff [SNZB-02] and [SNZB-04] !!!
I saw the release, I will update to 1.29.1 now :)

Great work @Koenkk !

Thanks

Koenkk, 2023-01-07:
All credits go to @blacknell