Koenkk/zigbee2mqtt #15031: Devices not assigned to a group are switched on/off

Teberon, 2022-11-16:
### What happened?

I have set up Z2M with Mosquitto and MQTT. I have then added the devices and set up groups. Unfortunately, devices outside of one group are also being switched off/on

I have already checked with MQTT explorer and removed all groups there and verified that there are also gone in Z2M as well as in Alexa. But after rebooting and creating new groups the issue comes back, the lights that are being controlled when calling one group is different.

### What did you expect to happen?

When asking Alexa or a card in the lovelace to only switch on the light in that group and not others.

### How to reproduce it (minimal and precise)

Install Z2M, configure Mosquitto and MQTT, add devices, create groups and add some devices to their group

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

26780700

### Adapter

 ConBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10024717/log.txt)

Teberon, 2022-11-16:
Since collecting logs as described in the debug article please find here the last snippet

```Debug 2022-11-16 19:37:17Received MQTT message on 'zigbee2mqtt/Büro/set' with data '{"state":"ON"}'
Debug 2022-11-16 19:37:17Publishing 'set' 'state' to 'Büro'
Debug 2022-11-16 19:37:17Received Zigbee message from 'Coordinator', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 2, ignoring since it is from coordinator
Info 2022-11-16 19:37:17MQTT publish: topic 'zigbee2mqtt/office_01', payload '{"brightness":254,"last_seen":"2022-11-16T18:36:22.246Z","linkquality":255,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'
Info 2022-11-16 19:37:17MQTT publish: topic 'zigbee2mqtt/Büro', payload '{"state":"ON"}'
Debug 2022-11-16 19:37:17Received Zigbee message from 'kitchen_door_02', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2022-11-16 19:37:17MQTT publish: topic 'zigbee2mqtt/Küche', payload '{"brightness":254,"state":"ON"}'
Info 2022-11-16 19:37:17MQTT publish: topic 'zigbee2mqtt/kitchen_door_02', payload '{"brightness":254,"last_seen":"2022-11-16T18:37:17.882Z","linkquality":255,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'
Debug 2022-11-16 19:37:17Received Zigbee message from 'kitchen_table_01', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info 2022-11-16 19:37:17MQTT publish: topic 'zigbee2mqtt/kitchen_table_01', payload '{"brightness":254,"last_seen":"2022-11-16T18:37:17.918Z","linkquality":255,"power_on_behavior":"on","state":"ON","update":{"state":"idle"},"update_available":false}'```
![image](https://user-images.githubusercontent.com/25689158/202265200-7ac20cb5-ac2a-4a1f-88bb-0cc997a2648c.png)

![image](https://user-images.githubusercontent.com/25689158/202265275-05b844de-7ff6-4b00-8aec-58512c434a5f.png)

darko-mijic, 2022-11-21:
I am having the same issue.
I have 16 lights in `lights.office` group and 1 light in `lights.office-accent` group. When I issue a change to the accent group, 3 lights from the office group are changing state.

darko-mijic, 2022-11-24:
I have fixed the issue by adding devices triggered by error to the group they were not part of and removing them immediately after that.

closed 2023-01-01 by github-actions[bot]