Koenkk/zigbee2mqtt #15032: [New device support]:  Model ZY-M100-S manufactured by Tuya Zigbee ID: TS0601 | _TZE204_ztc6ggyl

ashwin2101, 2022-11-16:
### Link

https://nl.aliexpress.com/item/1005004705672138.html?aff_fcid=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&tt=CPS_NORMAL&aff_fsk=_9gTCE1&aff_platform=shareComponent-detail&sk=_9gTCE1&aff_trace_key=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&terminal_id=c60aa1c2bd3d4f80907b0cc2716fb935&gatewayAdapt=glo2nld

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xdc8e95fffe4dc425","nwkAddr":42257,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"b�\u0007+ic�\u0007+\u0013d�\u0007+i\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000e\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u00128�\u0007+i9�\u0007+\u0013��\u0007+e��\u0007+\u0012��\u0007+\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668623664419,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the new device supporting page but I got stuck
The device model is ZY-M100S 

according to this page the device is supported in Zigbee2MQT

https://zigbee.blakadder.com/Tuya_ZY-M100-S.html

mattkerrison, 2022-11-17:
+1 on this, I bought it under the (wrong) impression it was already supported

AndrewNovikov, 2022-11-17:
I vote for this one as well. Same story.
+1

Dbroek, 2022-11-17:
Device gets recognizes but throws a lot of error's:

fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,36]}}
fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,81]}}
fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,18]}}
fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,133]}}
fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 9 with data {"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,130]}}
[Z2M@ha.futurispueri.com](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/)
[Devices](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/)
[Dashboard](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/dashboard)
[Map](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/map)
[Settings](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/settings)
[Groups](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/groups)
[OTA](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/ota)
[Touchlink](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/touchlink)
[Logs](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/logs)
[Extensions](https://ha.futurispueri.com/api/hassio_ingress/gxaTXpaA4GSC2fsq5VYNm2iAqGakV61lcpmK263v-Ko/#/extensions)
English
Show only
all
Filter by text
Enter search criteria
Log level
debug
Logging level
 
Debug 2022-11-17 19:13:21Received MQTT message on 'homeassistant/sensor/0xdc8e95fffe4dc3dd/tumble_switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xdc8e95fffe4dc3dd"],"manufacturer":"TuYa","model":"Human presence sensor with fall function (MIR-HE200-TY)","name":"0xdc8e95fffe4dc3dd"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","name":"0xdc8e95fffe4dc3dd tumble switch","state_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","unique_id":"0xdc8e95fffe4dc3dd_tumble_switch_zigbee2mqtt","value_template":"{{ value_json.tumble_switch }}"}'
Debug 2022-11-17 19:13:21Received MQTT message on 'homeassistant/select/0xdc8e95fffe4dc3dd/tumble_switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/set/tumble_switch","device":{"identifiers":["zigbee2mqtt_0xdc8e95fffe4dc3dd"],"manufacturer":"TuYa","model":"Human presence sensor with fall function (MIR-HE200-TY)","name":"0xdc8e95fffe4dc3dd"},"json_attributes_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","name":"0xdc8e95fffe4dc3dd tumble switch","options":["ON","OFF"],"state_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","unique_id":"0xdc8e95fffe4dc3dd_tumble_switch_zigbee2mqtt","value_template":"{{ value_json.tumble_switch }}"}'
Debug 2022-11-17 19:13:21Received MQTT message on 'homeassistant/sensor/0xdc8e95fffe4dc3dd/fall_sensitivity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xdc8e95fffe4dc3dd"],"manufacturer":"TuYa","model":"Human presence sensor with fall function (MIR-HE200-TY)","name":"0xdc8e95fffe4dc3dd"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","name":"0xdc8e95fffe4dc3dd fall sensitivity","state_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","unique_id":"0xdc8e95fffe4dc3dd_fall_sensitivity_zigbee2mqtt","unit_of_measurement":"","value_template":"{{ value_json.fall_sensitivity }}"}'
Debug 2022-11-17 19:13:21Received MQTT message on 'homeassistant/number/0xdc8e95fffe4dc3dd/fall_sensitivity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/set/fall_sensitivity","device":{"identifiers":["zigbee2mqtt_0xdc8e95fffe4dc3dd"],"manufacturer":"TuYa","model":"Human presence sensor with fall function (MIR-HE200-TY)","name":"0xdc8e95fffe4dc3dd"},"json_attributes_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","max":10,"min":1,"name":"0xdc8e95fffe4dc3dd fall sensitivity","state_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd","step":1,"unique_id":"0xdc8e95fffe4dc3dd_fall_sensitivity_zigbee2mqtt","value_template":"{{ value_json.fall_sensitivity }}"}'
Debug 2022-11-17 19:13:21Received MQTT message on 'homeassistant/sensor/0xdc8e95fffe4dc3dd/tumble_alarm_time/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xdc8e95fffe4dc3dd"],"manufacturer":"TuYa","model":"Human presence sensor with fall function (MIR-HE200-TY)","name":"0xdc8e95fffe4dc3dd"},"enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0xdc8e95fffe4dc3dd", [... cut ...]
Warning 2022-11-17 19:13:25fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 6 with data {"dp":6,"datatype":4,"data":{"type":"Buffer","data":[1]}}
Debug 2022-11-17 19:13:25Received Zigbee message from '0xdc8e95fffe4dc3dd', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":101}],"seq":19456}' from endpoint 1 with groupID 0
Warning 2022-11-17 19:13:25fromZigbee.tuya_radar_sensor_fall: NOT RECOGNIZED DP 101 with data {"dp":101,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
Debug 2022-11-17 19:13:26

cloudharps, 2022-11-18:
Same Problem. Converter doesn't works.

bachoo786, 2022-11-18:
I have bought the device ZY-M100 but get the following errors when trying to connect it to zigbee2mqtt:

```
Warning 2022-11-18 13:35:18Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_ztc6ggyl'
Warning 2022-11-18 13:35:18Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning 2022-11-18 13:35:18Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_ztc6ggyl'
Warning 2022-11-18 13:35:18Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

I thought this device was supported. Has anyone managed to connect it to zigbee2mqtt yet? thanks

acuadulza, 2022-11-18:
Same issue.
Using  ZY-M100-S with Z2M (version 1.28.2-1).
Coordinator Sonoff Zigbee 3.0 USB Dongle revision 20211217
Warning messages:

> Warning 2022-11-18 18:04:33Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE204_ztc6ggyl'
> Warning 2022-11-18 18:04:33Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

bachoo786, 2022-11-18:
@Koenkk any idea mate?

ashwin2101, 2022-11-18:
Hi @Koenkk I also get NOT RECOGNIZED DP errors 

[image]

DoRFox, 2022-11-19:
Zigbee2MQTT version: 1.28.2-1
By default, my sensor **ZY-M100-S** (TS0601 | _TZE204_ztc6ggyl) also is not recognized in Z2M. Аlthough its ID is present in the  [converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js), line 3458.

 I copied the appropriate part of the converter code for _TS0601_smart_human_presence_sensor_ as an external converter as mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/15032#issuecomment-1318932826), and it solved my problem. The sensor is discovered and works correctly.

bachoo786, 2022-11-19:
> Zigbee2MQTT version: 1.28.2-1 By default, my sensor **ZY-M100-S** (TS0601 | _TZE204_ztc6ggyl) also is not recognized in Z2M. Аlthough its ID is present in the [converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L3453).
> 
> I copied the appropriate part of the converter code for _TS0601_smart_human_presence_sensor_ as an external converter as mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/15032#issuecomment-1318932826), and it solved my problem. The sensor is discovered and works correctly.

what part did you cocpy from the external converter code? I have been copying the whole code as advised by @Koenkk . Also the latest version of z2m is 1.28.2, how are you on 1.28.2-1?

acuadulza, 2022-11-19:
@DoRFox  how di you it ? 
I also added the external converter as suggested by @Koenkk, but I keep getting  error messages....

[image]

bachoo786, 2022-11-19:
> @DoRFox how di you it ? I also added the external converter as suggested by @Koenkk, but I keep getting error messages....
> 
> [image]

So I upgraded to the latest dev version and removed the external converter and it worked for me.

acuadulza, 2022-11-19:
> So I upgraded to the latest dev version and removed the external converter and it worked for me.

Do you mean v1.28.2-1  ?   I had already this version   :-(

cloudharps, 2022-11-19:
> > So I upgraded to the latest dev version and removed the external converter and it worked for me.
> 
> @bachoo786 Do you mean v1.28.2-1 ? I had already this version :-(

Did you install zigbee2mqtt edge?

bachoo786, 2022-11-19:
> > So I upgraded to the latest dev version and removed the external converter and it worked for me.
> 
> @bachoo786 Do you mean v1.28.2-1 ? I had already this version :-(

I am running z2m on docker so I used the zigbee2mqtt:latest-dev image

bachoo786, 2022-11-19:
> > > So I upgraded to the latest dev version and removed the external converter and it worked for me.
> > 
> > 
> > @bachoo786 Do you mean v1.28.2-1 ? I had already this version :-(
> 
> Did you install zigbee2mqtt edge?

Nope I am on docker

DoRFox, 2022-11-19:
> what part did you cocpy from the external converter code? I have been copying the whole code as advised by @Koenkk . Also the latest version of z2m is 1.28.2, how are you on 1.28.2-1?

Recommended code is incorrect for this sensor.
[Correct example](https://www.dropbox.com/s/k04ucghttfqgvh4/ext_converter.js?dl=0)

cloudharps, 2022-11-19:
> Recommended code is incorrect for this sensor. [Correct example](https://www.dropbox.com/s/k04ucghttfqgvh4/ext_converter.js?dl=0)

Answer. Perfectly works. Thanks for the link.

acuadulza, 2022-11-19:
> Recommended code is incorrect for this sensor. [Correct example](https://www.dropbox.com/s/k04ucghttfqgvh4/ext_converter.js?dl=0)

@DoRFox Thank you very much !

Koenkk, 2022-11-20:
@DoRFox can you provide the data/database.db entry where this external converter works for?

ashwin2101, 2022-11-20:
> @DoRFox Thank you very much !

Thank you, I can confirm that is working

ashwin2101, 2022-11-20:
> @DoRFox can you provide the data/database.db entry where this external converter works for?

@Koenkk This is my db entry based on the Converter from @DoRFox  https://www.dropbox.com/s/k04ucghttfqgvh4/ext_converter.js?dl=0

{"id":7,"type":"Router","ieeeAddr":"0xdc8e95fffe4dc425","nwkAddr":4911,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"Ǫ\n+iȪ\n+iȪ\n+\u0013\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000e\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000g\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668936551721,"defaultSendRequestWhen":"immediate"}

DoRFox, 2022-11-20:
> @DoRFox can you provide the data/database.db entry where this external converter works for?

Yes, here she is, [database.db](https://www.dropbox.com/s/44y53rpn4emhjxu/database.db?dl=0)

Koenkk, 2022-11-20:
This converter is also available via the dev branch now and will be included in the next release, assuming this can be closed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-20 by Koenkk

ashwin2101, 2022-11-20:
thanks for all the effort, you guys are great

yayitazale, 2022-12-13:
Just a little question, I tryed both the external converter and the dev branch but the device it doesn't even pair. I'm holding the reset button for 6 seconsd as it is explained in the paper instructions attached to the device. 

Is my sensor dead or I'm doing something wrong? It is powered with a 12v but I don't see any led indicator or nothing to know that it is at least powered up.

[image]

Thanks!

DoRFox, 2022-12-13:
> 

But the label on the back informs us that this device is powered by AC 80-230 ?

yayitazale, 2022-12-13:
> > 
> 
> But the label on the back informs us that this device is powered by AC 80-230, right?

You are totaly right, I dind't even notice that, I was sure it was 12v as the aliexpress seller sent me also a 12v converter. It works perfectly thank you so much!

yayitazale, 2022-12-29:
Debugging the console I have seen that this type of sensors sends a lot of messages even if the state of presence, target_distance, or illumination does not change.

Is there a way to limit the number of messages it sends?

I am planning to replace all my motion sensors by these ones and if I add one per each room it will increase a lot the total number of messages to be processed by the coordinator.

ikoz, 2023-01-06:
Does anyone know if it is possible to make lights turn on and off via *binding* using these devices - so they work if the coordinator is down?

yayitazale, 2023-01-06:
> Does anyone know if it is possible to make lights turn on and off via _binding_ using these devices - so they work if the coordinator is down?

Not sure but I think that the answer is no:

[image]

ikoz, 2023-01-06:
> > Does anyone know if it is possible to make lights turn on and off via _binding_ using these devices - so they work if the coordinator is down?
> 
> Not sure but I think that the answer is no:
> 
> [image]

I suppoce "cocina" in Destination is your coordinator? Looks it doesnt support adding more bindings beyond the default. Thanks for checking!

yayitazale, 2023-01-06:
> I suppoce "cocina" in Destination is your coordinator? Looks it doesnt support adding more bindings beyond the default. Thanks for checking!

"Cocina" destination is a lights group but I should show something like genONOFF o presence on clusters?

yayitazale, 2023-01-06:
[image]

ikoz, 2023-01-06:
> [image]

Thank you, that's what I suspected. Like most sensors it doesnt have output clusters so I assume it can't bind to devices directly. A switch would have a genOnOff output cluster.

marcog20, 2023-04-07:
> Correct example

Hi, at the end were you able to reduce the traffic generated by this device? (i.e. target destination expose)

yayitazale, 2023-04-07:
No, but I have 8 of them and the system runs well.

darkssside, 2023-08-01:
Good evening. Trying to connect ZY-M100-S..
Added ext_converter.js . The device has been detected, but there are errors... constantly. Like these:
```
fromZigbee.tuya_smart_human_presense_sensor: NOT RECOGNIZED DP 109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,60]}}
debug 2023-08-01 19:34:08: Received Zigbee message from '0xa4c1380d8b841c62', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,60],"type":"Buffer"},"datatype":2,"dp":109}],"seq":17920}' from endpoint 1 with groupID 0
```

Does anyone have any thoughts?