Koenkk/zigbee2mqtt #15050: XGG 52PZ2M Gateway firmware issue

kisekinopureya, 2022-11-17:
### What happened?

I recently bought XGG 52PZ2M Gateway to use it with zigbee2mqtt, Managed to set up zigbee2mqtt and connected gateway to it. But it was only pairing with sonoff sensors, not other few brands i've tested. Since their [website](https://z2m.wiki/) is down. I can't access to documents. I decided to reflash the firmware but none of the recommended software on zigbee2mqtt wiki's 'Flashing CC1352/CC2652/CC2538 based adapters' worked. Then i tried to flash 'CC1352P2_CC2652P_launchpad_coordinator_20220219.hex' via internal captive portal ([it looks exactly like this](https://user-images.githubusercontent.com/28802021/202458119-9ae42c60-5928-4714-96b2-de4351cbcede.png)) of the device but it soft bricked. After researching i managed to confirm that esp8266 chip is working after flashing different firmware via usb but couldn't find any firmware that can work with zigbee2mqtt again. Does anyone have firmware file for esp8266 that works with zigbee2mqtt ?

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

As far as i recall it was not the latest firmware linked on website

### Adapter

XGG 52PZ2MGateway

kisekinopureya, 2022-11-18:
Sorry if i didn't clariy my issue properly. The gateway i mentioned is listed under recommended network devices on zigbee2mqtt.io. z2m.wiki is gateway's manufacturer's website. [image] My issue is, i flashed CC2652P firmware to ESP8266 chip and i am trying to find ESP chip's firmware back. Tried few ESP firmware but couldn't add device to zigbee2mqtt.

danieledwardgeorgehitchcock, 2022-11-18:
There may be some information on the [WaybackMachine](https://web.archive.org/web/20220125224020/https://z2m.wiki/index.html) that may be of help (support channels and emails)

closed 2022-12-27 by github-actions[bot]