Koenkk/zigbee2mqtt #15054: [New device support]:

JonnaI89, 2022-11-17:
### Link

https://www.namron.com/no/produkter/namron-zigbee-termostatplugg-16a/

### Database entry

Device type: Router IEEE Address: 0x9035eafffe5fa013

### Comments

Hope this device will be supported.

slippern1, 2022-12-04:
This is my first attempt to add a new device to zigbee2mqtt.
The device is reporting power, temperature and state of the switch, but i cannot figure out any thermostat settings.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512749'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '4512749', // Vendor model number, look on the device for a model number
    vendor: 'Namron', // Vendor of the device (only used for documentation and startup logging)
    description: 'Namron - Zigbee Thermostat Outlet Socket', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.temperature, fz.metering, fz.electrical_measurement, fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [
        e.temperature(), e.power(), e.current(), e.voltage(), e.switch()
    ],

    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },

};

module.exports = definition;
`

slippern1, 2022-12-06:
Added som reporting stuff.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512749'],
    model: '4512749',
    vendor: 'Namron',
    description: 'Namron - Zigbee Thermostat Outlet Socket',
    extend: [ extend.switch() ],
    fromZigbee: [ fz.metering, fz.electrical_measurement, fz.on_off ],
    toZigbee: [tz.on_off ],
    exposes: [ e.temperature(), e.power(), e.current(), e.voltage(), e.switch(),
    ],

        //Report if on/off button is manually pressed on the device.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
        //How often the device should report on various readings.
        await reporting.temperature(endpoint, {min: 300, change: 50});
        await reporting.rmsCurrent(endpoint, {min: 60, change: 10});
        await reporting.rmsVoltage(endpoint, {min: 60, change: 10});
        await reporting.activePower(endpoint, {min: 60, change: 10});
   }
};

module.exports = definition;
```

slippern1, 2023-01-06:
I have sent an email to Namron asking for documentation on the device.

slippern1, 2023-01-08:
Fixed an temperature reporting typo

```const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512749'],
    model: '4512749',
    vendor: 'Namron',
    description: 'Namron - Zigbee Thermostat Outlet Socket',
    extend: [ extend.switch() ],
    fromZigbee: [ fz.metering, fz.electrical_measurement, fz.on_off, fz.temperature],
    toZigbee: [tz.on_off ],
    exposes: [ e.temperature(), e.power(), e.current(), e.voltage(), e.switch(),
    ],

        //Report if on/off button is manually pressed on the device.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
        //How often the device should report on various readings.
        await reporting.temperature(endpoint, {min: 60, change: 10});
        await reporting.rmsCurrent(endpoint, {min: 60, change: 10});
        await reporting.rmsVoltage(endpoint, {min: 60, change: 10});
        await reporting.activePower(endpoint, {min: 60, change: 10});
   }
};

module.exports = definition;
```

kolaf, 2023-01-09:
The latest version works for me. Latest version is missing `const fz = require('zigbee-herdsman-converters/converters/fromZigbee');` at the top.

slippern1, 2023-01-09:
Added 'power_on_behavior' and code clean up.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512749'],
    model: '4512749',
    vendor: 'Namron',
    description: 'Namron - Zigbee Thermostat Outlet Socket',
    fromZigbee: [ fz.metering, fz.electrical_measurement, fz.on_off, fz.temperature, fz.power_on_behavior ],
    toZigbee: [tz.on_off, tz.power_on_behavior ],
    exposes: [ e.temperature(), e.power(), e.current(), e.voltage(), e.switch(),
    exposes.enum('power_on_behavior', ea.ALL, ['on', 'off', 'restore']).withDescription('Recover state after power outage'),
    ],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'msTemperatureMeasurement']);
            await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await endpoint.read('haElectricalMeasurement', ['acCurrentMultiplier', 'acCurrentDivisor']);
            
            await reporting.onOff(endpoint);
            await reporting.temperature(endpoint);
            await reporting.rmsVoltage(endpoint);
            await reporting.rmsCurrent(endpoint);
            await reporting.activePower(endpoint);
            
   }
};

module.exports = definition;
```

slippern1, 2023-01-09:
After an Google Image search i found out that this Namron Termostatplugg seems to be the exact same device as this one; https://www.sunricher.com/zigbee-smart-plug-sr-zg9020a.html

slippern1, 2023-01-10:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5303

closed 2023-02-18 by github-actions[bot]