Koenkk/zigbee2mqtt #15056: Zigbee2mqtt causes high CPU usage

geoffreydemaagd, 2022-11-17:
### What happened?

When starting zigbee  (systemctl start zigbee2mqtt) the CPU usage increases to 100% and more
Stopping Zigbee and all is ok, So i has something to do when Herdsmann or Zigbee2mqtt.

I updates zigbee2mqtt to the latest version.

Started zigbee in example below with -test to show zigbee was the creator( 147% cpu)

[image]

node -v
v16.15.1

npm -v
8.11.0

### Zigbee2MQTT version

1.28.2 commit: e66fe853

### Adapter firmware version

20211217

### Adapter

zStack3x0

aostanin, 2022-11-18:
I'm having the same issue with the 1.28.2 Docker image. Downgrading to 1.28.0 seems to have solved it.

geoffreydemaagd, 2022-11-29:
After digging for a while I see in the V8 CPU profiler that after starting Z2M the garbage collector is constanly running. This is causing the high CPU load. Tried to install different version but the problem stays. Unfortunatly I still have a RPI 3b+ so the available memory isn't so great. Does anybody have an idea why this problem suddenly pops up? Memory usage is quite the same as when the problem wasn't there.

Koenkk, 2022-11-29:
What device do you have in your network?

geoffreydemaagd, 2022-11-29:
Koen,

I have only 3 devices
WSDCGQ11LM	
KEYZB-110
TS011F_plug_3

SONOFF ZigBee 3.0 USB Dongle Plus, TI CC2652P

Even without the USB sonofff connected, the garbage collector initiated after starting Z2M is running constantly with high CPU as a result.

Koenkk, 2022-11-29:
Can you provide the herdsman debug log when starting z2m + 5 minutes after this? (after 5 minutes the cpu usage is still 100%, right?)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

geoffreydemaagd, 2022-11-29:
[Logging.txt](https://github.com/Koenkk/zigbee2mqtt/files/10116201/Logging.txt)

Koen, 

Tried to capture the output but there are so many messages I can't copy them all. In attached to log I could capture

Many thanks for your help

Koenkk, 2022-11-30:
In the logs I only see the starting phase of Zigbee2MQTT. In this phase a lot of commands are fired at the adapter so 100% CPU usage would not be strange at this phase. 

Can you provide me the logs when Zigbee2MQTT has started (wait 60 seconds after start) and the CPU usage is 100%?

geoffreydemaagd, 2022-11-30:
Installed different version op NPM and node (14.x,16.x 18.x) just to be sure the problem is not related with those version.

The log what I send was already the logging when the cpu was going crazy.

Normally my CPU is around

[image]

From the moment the parser of Herdsman starts the CPU is increasing

[image]

And after a few sec it's like

[image]

Stopping zigbee and the cpu goes back to normal

Koenkk, 2022-11-30:
You can paste the log file here.

geoffreydemaagd, 2022-11-30:
[putty.log](https://github.com/Koenkk/zigbee2mqtt/files/10126898/putty.log)
 Thank you for your help

Koenkk, 2022-11-30:
Can you try removing the `TS011F_plug_3` from your network and see if this issue disappears?

geoffreydemaagd, 2022-11-30:
I removed all devices (tried first the TS011F but was not the cause) but the problem stays the same.

Koenkk, 2022-12-01:
Can you provide the logging when having no devices paired (and also all removed from power) from starting z2m + 5 minutes after this when the cpu usage is 100%?

geoffreydemaagd, 2022-12-02:
Removed all devices=> staring zigbee2mqtt => cpu stays stable

I tried to join only 1 device (remove the rest to test each device)
But from the moment I try to join the problem occures immedialtly

[logging021222.txt](https://github.com/Koenkk/zigbee2mqtt/files/10139673/logging021222.txt)

I was trying to join and approx from this line the cpu went crazy
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1026,"srcaddr":51165,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":21,"securityuse":0,"timestamp":7793532,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,0,10,0,0,41,92,8]}} +56s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":2140}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":51165,"endpoint":1,"linkquality":21,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +56s
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '51165', skipping... +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,5,4,221,199,1,1,0,21,0,134,236,118,0,0,8,24,1,10,0,0,33,224,27,221,199] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [29,13,254,37,68,129,0,0,3,4,221,199,1,1,0,18,0,145,237,118,0,0,17,24,2,10,0,0,41,248,3,20,0,40,255,16,0,41,177,39,221,199,29,93] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,5,4,221,199,1,1,0,21,0,134,236,118,0,0,8,24,1,10,0,0,33,224,27,221,199,29,13,254,37,68,129,0,0,3,4,221,199,1,1,0,18,0,145,237,118,0,0,17,24,2,10,0,0,41,248,3,20,0,40,255,16,0,41,177,39,221,199,29,93] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,5,4,221,199,1,1,0,21,0,134,236,118,0,0,8,24,1,10,0,0,33,224,27,221,199,29] - 

Maybe I should try an older version of the coordinator software?

Koenkk, 2022-12-02:
- What device is this?
- Can you provide the logging again but then running with `DEBUG=*` instead of  `DEBUG=zigbee-herdsman*`

geoffreydemaagd, 2022-12-04:
I have 3 devices but with each device I have the same problem
KEYZB-110
WSDCGQ11LM
TS011F_plug_3
[Logging041222.txt](https://github.com/Koenkk/zigbee2mqtt/files/10148130/Logging041222.txt)

From the moment those lines appears the cpu starting to reach high and never stops getting high

32;1mserialport/bindings/unixRead [0mStarting read [32m+3ms[0m

Koenkk, 2022-12-04:
Seems it is caused by the serialport library. I just updated it to 10.5.0, (from 9.2.8) can you check if this fixes the issue?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

geoffreydemaagd, 2022-12-05:
Tried with 1.28.4-dev but the same behavior is happening. Still lots of serialport/bindings/unixRead �[0mStarting read �[32m+3ms�[0m 
and of course high cpu usage

Koenkk, 2022-12-06:
Do you have any other USB devices attached? Somehow the serialport library is not working correctly on your system.

geoffreydemaagd, 2022-12-06:
Koen, I have 2 RPI3b+ and have already tested it on both RPI's. Both have the problem, even with only 1 USB (the dongle) attached to it.

Koenkk, 2022-12-07:
More things you could try (but I'm not sure if this fixes the problem)
- Run `npm rebuild --build-from-source` in the Zigbee2MQTT directory
- Upgrade your adapter firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

geoffreydemaagd, 2022-12-17:
I just installed a full new OS image with zigbee2mqtt on a brand new raspberry 4 8GB.
I still find the load of zigbee2mqtt much too high.
[image]

Is this also the case on your device?

geoffreydemaagd, 2022-12-17:
Wasn't working -  I'm currenlty installing a fresh new image on a RPI4 8GB. Will keep you informed as soon as its finished

geoffreydemaagd, 2022-12-18:
Koen,

I've found the reason of the high cpu usage and zigbee was not the cause. 
I have also OLAD installed on my RPI to control my DMX lighting. In one of those config files there was only a prefix mentionned of which usb device it could use for example ttyUSB

Both programs worked and without seeing it in the debug and logs it (not of ola neither in zigbee) was hard to pinpoint the actual problem.

I was settting up a brand new RPI 4 with 8 GB (new os etc) and the problem arised also on that device.

Changed the OLA to the actual serial/by-id and the problem was gone.

In the past I had a symlink configured for zigbee2mqtt, in that setup I had no problem it all but started since using the serial/by-id in the config yaml of zigbee2mqtt (now that I know the problem, it all became clear) 

Many thanks for all your cooperation and good advice. That's what I like here.

closed 2022-12-18 by geoffreydemaagd

Koenkk, 2022-12-18:
Good to hear!