Koenkk/zigbee2mqtt #15057: Error: No datapoint defined for 'transition'

MarcPot, 2022-11-17:
### What happened?

I'm using the :latest docker version of z2m and one of my dimmers does not work anymore because it doesn't understand the 'transition' parameter.
Reverted to 1.27.0 that doesn't have the problem. (Just did a leap back, not a specific version.)

Using Home Assistant 2022.11.3
Frontend 20221108.0 - latest

Docker version latest (not working)
```
Received MQTT message on 'zigbee2mqtt/Prikkabel/set' with data '{"state":"ON","transition":0.0}'
Debug 2022-11-17 19:21:05Publishing 'set' 'state' to 'Prikkabel'
Debug 2022-11-17 19:21:05Received Zigbee message from 'Prikkabel', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,87],"type":"Buffer"},"datatype":2,"dp":2}],"seq":13121}' from endpoint 1 with groupID 0
Info 2022-11-17 19:21:05MQTT publish: topic 'zigbee2mqtt/Prikkabel', payload '{"brightness":87,"countdown":null,"light_type":"led","linkquality":76,"power_on_behavior":"previous","state":"ON"}'
Info 2022-11-17 19:21:05MQTT publish: topic 'zigbee2mqtt/Prikkabel/type', payload '{"light":"led"}'
Error 2022-11-17 19:21:05Publish 'set' 'state' to 'Prikkabel' failed: 'Error: No datapoint defined for 'transition''
Debug 2022-11-17 19:21:05Error: No datapoint defined for 'transition' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/lib/tuya.js:1391:23) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Info 2022-11-17 19:21:05MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Prikkabel' failed: 'Error: No datapoint defined for 'transition''","meta":{"friendly_name":"Prikkabel"},"type":"zigbee_publish_error"}'
Error 2022-11-17 19:21:05No converter available for 'transition' (0)
Debug 2022-11-17 19:21:05Received MQTT message on 'zigbee2mqtt/Prikkabel/set' with data '{"state":"ON","transition":0.0,"brightness":86}'
Debug 2022-11-17 19:21:05Publishing 'set' 'brightness' to 'Prikkabel'
Error 2022-11-17 19:21:05Publish 'set' 'brightness' to 'Prikkabel' failed: 'Error: No datapoint defined for 'transition''
Debug 2022-11-17 19:21:05Error: No datapoint defined for 'transition' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/lib/tuya.js:1391:23) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at MqttClient.emit (node:events:513:28) at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1547:12) at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:535:12) at work (/app/node_modules/mqtt/lib/client.js:438:12) at processTicksAndRejections (node:internal/process/task_queues:78:11)
```

Docker version 1.27.0 (working)

```
Debug 2022-11-17 18:45:13Received MQTT message on 'zigbee2mqtt/Prikkabel/set' with data '{"state":"ON","transition":0.0,"brightness":95}'
Debug 2022-11-17 18:45:13Publishing 'set' 'brightness' to 'Prikkabel'
Debug 2022-11-17 18:45:13to moes_105_dimmer key=[brightness], value=[95]
Error 2022-11-17 18:45:13No converter available for 'transition' (0)
Debug 2022-11-17 18:45:13Received Zigbee message from 'Prikkabel', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,75],"type":"Buffer"},"datatype":2,"dp":2}],"seq":58944}' from endpoint 1 with groupID 0
Debug 2022-11-17 18:45:13from moes_105_dimmer, dp=[2], datatype=[2], value=[331]
Info 2022-11-17 18:45:13MQTT publish: topic 'zigbee2mqtt/Prikkabel', payload '{"brightness":84,"light_type":"led","linkquality":72,"power_on_behavior":"previous","state":"ON"}'
Error 2022-11-17 18:45:13No converter available for 'transition' (0)
Debug 2022-11-17 18:45:13Received Zigbee message from 'Prikkabel', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":59200}' from endpoint 1 with groupID 0
Debug 2022-11-17 18:45:13from moes_105_dimmer, dp=[1], datatype=[1], value=[true]
Info 2022-11-17 18:45:13MQTT publish: topic 'zigbee2mqtt/Prikkabel', payload '{"brightness":84,"light_type":"led","linkquality":76,"power_on_behavior":"previous","state":"ON"}'
Debug 2022-11-17 18:45:13Received Zigbee message from 'Prikkabel', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,118],"type":"Buffer"},"datatype":2,"dp":2}],"seq":59456}' from endpoint 1 with groupID 0
Debug 2022-11-17 18:45:13from moes_105_dimmer, dp=[2], datatype=[2], value=[374]
Info 2022-11-17 18:45:13MQTT publish: topic 'zigbee2mqtt/Prikkabel', payload '{"brightness":95,"light_type":"led","linkquality":76,"power_on_behavior":"previous","state":"ON"}
```

### Zigbee2MQTT version

 1.28.2 commit: 360a777

### Adapter firmware version

 20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

danieledwardgeorgehitchcock, 2022-11-18:
It looks like in both cases, the `transition` converter / datapoint isn't available.

Could you not just remove the `transition` key from the MQTT message that you are sending to Z2M and therefore the device?

MarcPot, 2022-11-18:
The problem is it’s multiple Home Assistant integrations that are doing it.
So it would mean I have to open an issue with all of them to fix it.

At that point I’d just rather keep z2m at a working previous version.

danieledwardgeorgehitchcock, 2022-11-18:
As the `transition` is an optional value in the HA service call, I do not think it is correct that an integration sets this anyway. Which integration(s) in particular are using the `transition` key in HA?

MarcPot, 2022-11-18:
Currently don’t have the option to check all of them.

Adaptive Lighting/Circadian Lighting I am sure of.

danieledwardgeorgehitchcock, 2022-11-18:
I'd raise a bug with them

danieledwardgeorgehitchcock, 2022-11-18:
Looks like this was fixed in Z2M yesterday:

Koenkk/zigbee-herdsman-converters@a04849930bd84455209eff815aa594b4a35c3a05

closed 2022-12-26 by github-actions[bot]