Koenkk/zigbee2mqtt #15068: TuYa TS0012_switch_module changing states of both switches even if only one was triggered

dtrochow, 2022-11-18:
### What happened?

I paired my "GIRIER Tuya ZigBee 3.0 Smart Light Switch Module No Neutral Wire Required" module ([link](https://pl.aliexpress.com/item/1005003439384086.html?spm=a2g0o.order_list.0.0.21ef1c24RAkgWI&gatewayAdapt=glo2pol)), which was detected in the GUI as TuYa TS0012_switch_module.
 
 After triggering one of the switches in GUI, the both of them are changing the state (either to on or to off state). The second one is changing after about a second or half of a second.
  
[image]
[image]

 
 Interesting thing is, that when I use the physical module inputs to change the state of the switch, it affects only one of them.
 
 

### What did you expect to happen?

The software switches in the device card change the states of the each switch (relay) independently.

### How to reproduce it (minimal and precise)

1. Pair the device
2. Go to the device card in Zigbee2MQTT GUI -> then Exposes tab
3. Try to switch left or right switch (the both of them will change the state, even if only one was triggered)

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

 20220219

### Adapter

SONOFF ZBDongle-P/ ZBDongle-E

dtrochow, 2022-11-19:
Replicates the [#13656](https://github.com/Koenkk/zigbee2mqtt/issues/13656)

closed 2022-11-19 by dtrochow

lvs-vladimir, 2022-11-20:
Hello, I have the same problem. The latest version of z2m is installed.

omolko, 2023-04-22:
same problem with ZHA or Mqtt. Did someone get a soluce?