Koenkk/zigbee2mqtt #15070: Can`t calibrate current TS011F_plug_3

mefisto22, 2022-11-18:
### What happened?

I purasched a  TS011F_plug_3 plug. I now trying to calibrate. I can calibrate power and voltage, but current i can`t. I trying with an 60W bulb. It is constantly show 0,46A. Only when i set -200%, then the value is jump to -0,54A. When i set value between 0% and -200% nothing happens.  I tried with latest, and latest-dev version too.

### What did you expect to happen?

Calibrate current and it will be show the true value

### How to reproduce it (minimal and precise)

Connect the device to the zigbee network, and try calibrate

### Zigbee2MQTT version

1.28.2-dev commit: 8510e99

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10044087/log.txt)

mefisto22, 2022-12-21:
still not work

Koenkk, 2022-12-22:
this should work in the next z2m version (released on 1 January)

mefisto22, 2023-01-06:
still not work

mefisto22, 2023-02-11:
It will be work anytime?

mefisto22, 2023-03-20:
still not work

closed 2023-07-02 by github-actions[bot]

mattg2, 2023-07-18:
for me it is still not working

mefisto22, 2023-07-18:
it doesn't work for me either