Koenkk/zigbee2mqtt #15075: [Feature request]: Add support for filtered_attributes on device creation

gwww, 2022-11-19:
### Is your feature request related to a problem? Please describe

When a new device is added all of the device's attributes are passed to Home Assistant, even when `device_options`, `filtered_attributes` is set. The new Inovelli Blue device (model VZM31-SN) has around 100 attributes. Each of those attributes end up as an entity in Home Assistant. It would be nice if Home Assistant did not know these attributes exist, and hence would not create an entity for each attribute.

`filtered_attributes` works well once the device and entities are created in Home Assistant in that the filter list is respected for published messages. This is useful in that each attribute that is published as a part of a message causes HA to generate an event. Without filtering a simple button press causes hundreds of events.

The next step is that the entities are not created in the first place. The existing `filtered_attributes` list seems like a good way to do that.

### Describe the solution you'd like

Use `filtered_attributes` list on all messages published.

### Describe alternatives you've considered

Manually (and painfully) deleting each unwanted entity in HA.

### Additional context

Here is a snipped of the `MQTT publish: topic 'zigbee2mqtt/bridge/event', payload` message sent when the new device is added to the network (I have the full formatted message if you would like it; it is ~900 lines):

```
                {
                    "access":7,
                    "description":"0-254:This is the color of the LED strip in a hex representation. 255:Synchronization with default all LED strip color parameter.",
                    "name":"defaultLed1ColorWhenOn",
                    "property":"defaultLed1ColorWhenOn",
                    "type":"numeric",
                    "value_max":255,
                    "value_min":0
                },
                {
                    "access":7,
                    "description":"0-254:This is the color of the LED strip in a hex representation. 255:Synchronization with default all LED strip color parameter.",
                    "name":"defaultLed1ColorWhenOff",
                    "property":"defaultLed1ColorWhenOff",
                    "type":"numeric",
                    "value_max":255,
                    "value_min":0
                },
                {
                    "access":7,
                    "description":"Intesity of LED strip when on. 101 = Syncronized with default all LED strip intensity parameter.",
                    "name":"defaultLed1IntensityWhenOn",
                    "property":"defaultLed1IntensityWhenOn",
                    "type":"numeric",
                    "value_max":101,
                    "value_min":0
                },
                {
                    "access":7,
                    "description":"Intesity of LED strip when off. 101 = Syncronized with default all LED strip intensity parameter.",
                    "name":"defaultLed1IntensityWhenOff",
                    "property":"defaultLed1IntensityWhenOff",
                    "type":"numeric",
                    "value_max":101,
                    "value_min":0
                },
                {
                    "access":7,
                    "description":"0-254:This is the color of the LED strip in a hex representation. 255:Synchronization with default all LED strip color parameter.",
                    "name":"defaultLed2ColorWhenOn",
                    "property":"defaultLed2ColorWhenOn",
                    "type":"numeric",
                    "value_max":255,
                    "value_min":0
                },
```

Daniel15, 2023-02-13:
I'd love to see this implemented. @Koenkk  could you please reopen this?

sjorge, 2023-02-14:
I think the underlying issue here is probably two fold:
1. unless you know the device ieee before hand, you can't add the filtered_attributes  entries before joining the device (using global device_options can be used here as a work around)
2. (the main issue) filtered_attributes does not remove them from the exposes data, which the home assistant extension uses (and other things like the frontend for that matter)

gwww, 2023-02-14:
I know very little about the internals, so from a "this feature would be nice" perspective... I care about filtering attributes that are published and those attributes being associated with a specific device type. In my particular case it's the Inovelli Blue Dimmer switch.

As to the front end, I do not want anything filtered from there. To me, that is "command central" where everything has to be visible. In particular, all the filtered attributes I want to access them so that I can configure the device. I just don't want all those attributes in HA as I will never try and configure the device there. All I want in HA is "control" attributes.

ZetaWaves, 2023-03-06:
Having more than 10+ entities for each device in HA will cause HA to be unstable. It will also cause MQTT to be unstable. You will have long start times (sometimes up to 1 hour to start MQTT on boot if network is large enough).This needs addressed ASAP. It is NOT normal to have this many entities in HA for each device. There is no reason to log every change to every configuration parameter on every device. This will cause extremely large recorder database sizes as well which will make the problem even worse overall. Please address this issue somehow. These exposes need to be configurable in Z2M, but they should not be entities in HA that are constantly monitored.

gwww, 2023-03-07:
As a work around what I'm doing is disabling entities on HA that I don't need. On the other hand, I have not installed an Inovelli switch in my live network due to waiting for a few more bug fixes coming in firmware updates. That said in my testing I wrote a Python script to disable HA entities. It's here: https://gist.github.com/gwww/1c959a00ec2bcf59f7041fe731b33a3c. Disabling entities gets around the recorder database size.

closed 2023-04-15 by github-actions[bot]

studioIngrid, 2023-06-19:
is it possible to reopen this request?

openr0ad, 2024-06-27:
Devices like this may become popular very quickly, especially considering they are coming out with a version of this device with mmWave technology implemented. Other large manufacturers may follow suit with their own implementation of these devices. Having an option in z2m to select specific controls to be propagated to HA will be useful.

karaminder, 2024-07-07:
I desperately need this with 120+ zigbee devices in one network with 32+ inovelli dimmers.