Koenkk/zigbee2mqtt #15076: [New device support]: Mercator SPP02

Schmee323, 2022-11-19:
### Link

https://www.ikuu.com.au/product/double-power-point/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x588e81fffed899e9","nwkAddr":64386,"manufId":4098,"manufName":"_TZ3210_raqjcxo5","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,4,5,6,1794,2820,61184,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3210_raqjcxo5","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"inClusterList":[4,5,6,61184,3],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668839023849,"defaultSendRequestWhen":"immediate"}

### Comments

These work with ZHA but not with zigbee2mqtt which is a bit weird. Looks like it runs on Tuya TS011F

### External converter

```shell
Tried a few things but really not much good instruction on how to do this via home assistant. Not much that is up do date at least.
```

ph0nebooth, 2022-12-05:
I second this. The SPPUSB02 is supported but not the non-USB variant.

artys29, 2022-12-11:
Hi I'm having a similar issue. My device is recognised but as TS011f_plug_1. Unable to control second power point. 
Copied the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/13190, which seemed to work, but then it caused issues with my other zigbee power points

Koenkk, 2022-12-11:
@Schmee323 fixed the issue for yours

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@artys29 can you provide the data/database.db entry of your device?

Schmee323, 2022-12-11:
Thanks so much!

 

Is there a way to move to the dev branch without re-pairing all devices? There doesn’t seem to be any info on this that I can find. 

 

Obviously, repairing the entire network is a huge pain, so a step by step guide for HA would be great if it is possible.

 

Thanks for all your hard work!

 

Thanks,

Simeon Larrea

artys29, 2022-12-12:
> @Schmee323 fixed the issue for yours
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
> 
> @artys29 can you provide the data/database.db entry of your device?

{"id":14,"type":"Router","ieeeAddr":"0x588e81fffed39e76","nwkAddr":34391,"manufId":4098,"manufName":"_TZ3210_raqjcxo5","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820,61184,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\f (+\u0013\u0018 (+\u0016","65506":31,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3210_raqjcxo5","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":74,"powerSource":1}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"offWaitTime":0,"onTime":0,"moesStartUpOnOff":0}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"activePower":0,"rmsVoltage":235,"rmsCurrent":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,173]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173a9","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173a9","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173a9","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[4,5,6,61184,3],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":0,"offWaitTime":0,"onTime":0,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173a9","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-892627439},"lastSeen":1670831743306,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-12-12:
@Schmee323 you don't need to re-pair when switching to edge. Just migrate your config from the HA addon page.

@artys29 then it should also be fixed in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-12 by Koenkk