Koenkk/zigbee2mqtt #15077: [New device support]: Mercator Plugbase SPBS01G

Schmee323, 2022-11-19:
### Link

https://www.ikuu.com.au/product/plugbase/

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xec1bbdfffe686371","nwkAddr":8140,"manufId":4098,"manufName":"_TZ3210_tfxwxklq","powerSource":"Mains (single phase)","modelId":"TS0101","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,4,5,6,61184,3],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0101","manufacturerName":"_TZ3210_tfxwxklq","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1668752475391,"defaultSendRequestWhen":"immediate"}

### Comments

These work with ZHA but not with zigbee2mqtt which is a bit weird. Looks like it runs on Tuya TS0101

### External converter

```shell
Tried a few things but really not much good instruction on how to do this via home assistant. Not much that is up do date at least. Would be great if there was a guide on how to do step by step for HA via GUI. Seems all instructions are for linux console which can lead to a broken HA if you are not 100% sure what u are doing.
```

Koenkk, 2022-11-20:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0101', ['_TYZB01_ijihzffk', '_TZ3210_tfxwxklq']),
    model: 'TS0101',
    vendor: 'TuYa',
    description: 'Zigbee Socket',
    whiteLabel: [{vendor: 'Larkkey', model: 'PS080'}],
    extend: extend.switch(),
    meta: {disableDefaultResponse: true},
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Schmee323, 2022-11-20:
Awesome. Worked perfectly.

Thanks so much!!!!

Koenkk, 2022-11-20:
great, added it for out-of-the-box support.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-20 by Koenkk