Koenkk/zigbee2mqtt #15079: Curtain switch positioning doesn't work (TS130F).

vindolin, 2022-11-19:
### What happened?

I have a bunch of these curtain switch modules:  https://www.aliexpress.com/item/4001190275372.html

They successfully bind as:
  Zigbee Model:     TS130F
  Zigbee Manufacturer: _TZ3000_qa8s8vca

(The specification page says SC400ZB-EU)

Everything works except moving the blinds to specific position.

I successfully calibrated my blinds via the UI using the calibrate switch and the OPEN/CLOSE/STOP buttons.

The motor cuts off at the right time.

When I move the blinds with {"state": "DOWN"} and stop them with {"state": "STOP"} somewhere on the way down, the stop and the position value reports the correct value.

Setting the position with the physical buttons also works and the position attribute gets set to the correct value.

**But when I try to set them to e.g. 50% with {"position": "50"} they randomly open or close 100%.**

If someone wants to look into the issue, I'm happy to ship one of the curtain switch devices to you.

### Zigbee2MQTT version

1.28.2-dev

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Ultrawipf, 2022-12-30:
Got the same issue. The module does not stop at the right time when using the position setting. 
It opens and closes fully but setting a position does not work and because closing is faster than opening it will drift a lot with every cycle as well.

closed 2023-02-06 by github-actions[bot]

piogundel, 2023-05-01:
Hi guys! I have the same issue. 7 curtains and two of them seem to not receive/accept set cover position message. Up down stop commands and physical buttons work well. Did you manage to overcome this?

Pedder007, 2023-09-18:
Hi, the same here with these.
If got a lot of others of _TZ3000_ltiqubue and them I can send to whatever percentage without any problem.
In addition I have the issue, that the backlight is completely OFF and cannot be switched on.

piogundel, 2024-01-22:
Automations in home assistant that run set cover position command on multiple covers at same time are causing network congestion - that is why some of cover did not respond correctly. You need to use mqtt publish as a service in ha automations in zigbee to mqtt and use groups of covers but created in z2m, not in ha. You can also use it for separate covers but most likely you would need to Set a delay between commands. It is all described in z2m documentation.

kevincaradant, 2024-07-16:
Hi everyone

Did you solve the issue ? It's really Just this device because others versions works fine.
Indeed, as soon as I click on a spexcifc position, the roller will fully close or open.
Maybe it can  be an issue of z2m because when the device down and up using arrow even on z2m, the roller knows the position.

mwitkow, 2024-07-30:
I have exactly the same problem 

I took can't set position via command (everything else works) on both my TS130F_TZ3000_qa8s8vca. Does anyone know if there has been any firmware upgrades? 

I saw there is a firmware for TS130F but a slightly different version (_TZ3000_1dd0d5yi)
https://github.com/Koenkk/zigbee-OTA/pull/325/files

are these compatible and can be flashed?

Coollision, 2024-08-13:
Hi all, I also have a _TZ3000_qa8s8vca model, and I'm seeing the same things, no position setting possible, and also non-functional backlight

daniel-sc, 2024-09-05:
I observe the same for some of my tf130f. Note that I do _not_ use z2m but ZHA - maybe this is indeed rather a firmware issue?