Koenkk/zigbee2mqtt #15086: Zigbee2Mqtt dead after some hours

sl4m01, 2022-11-19:
### What happened?

Zigbee2Mqtt seems to be dead after some hours.
Device status are not updated anymore:

- I do have a tradfri motion sensor https://www.zigbee2mqtt.io/devices/E1525_E1745.html#ikea-e1525%252Fe1745 which is just not sending detection anymore. 
- Or a tradfri OnOff switch where the actions are not detected anymore https://www.zigbee2mqtt.io/devices/E1743.html
- Blitzwolf Plug just stays at the same measured power level for hours and doesnt react anymore https://www.zigbee2mqtt.io/devices/TS0121_plug.html#tuya-ts0121_plug

However the bridge MQTT state is still "online" and i am getting a {"healthy":true},"status":"ok"} after requesting zigbee2mqtt/bridge/request/health_check

Restart Zigbee2Mqtt does not help zigbee2mqtt/bridge/request/restart .. bridge state is going offline and online in ~ 3 seconds.
Only a restart of the raspberry pi help sometime partly:
- sometimes the motion sensor is alive again
- sometimes the tradfri on / off is alive again

I am runing openhabian with openhab on a raspberry pi 3b+. The problems are new ... i am using zigbee2mqtt some years already and the issue occurred some weeks ago. 

To solve the issue i updated from 1.22 to 1.28.2, but no improvement. 

### How to reproduce it (minimal and precise)

Just wait for some hours

### Zigbee2MQTT version

1.2.8.2

### Adapter firmware version

20201127

### Adapter

CC2531 USB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10047317/log.txt)

uweb-503, 2022-11-19:
I seem to have the same problem. Until noon today everything was working, about 2 hours before I had changed the status of a device, everything was fine. The attempt to change the status again failed later. Changes to other devices do not work either.

Running Home Assistant with Z2M integration (1.28.2-1) and mosquitto broker (6.1.3).

uweb-503, 2022-11-19:
My problem is solved after restarting the zigbee2mqtt add-on.

lux73, 2022-11-20:
CC2531 Zigbee Coordinator considered as weak - please buy a newer powerful CC2652x Coordinator

**CC2531 are well-known as problematic - if you like a strong reliable Zigbee Network please switch!**

you could use the old CC2531 as Router (works without glitches) but please not more as Coordinator

t0bse, 2022-11-20:
i had a problem like this for two days now, after i upgraded my rpi 3 and zigbee2mqtt. 
i saw that power management for wifi on my pi was enabled (don't know what is the default) 
with "iwconfig"
i disabled it now and it seems like it's running again, like the months before. 
maybe this helps, you can disable it with "sudo iwconfig wlan0 power off"

i don't know if this was accidentaly enabled after upgrading lots of pi system services?

closed 2023-01-13 by github-actions[bot]