Koenkk/zigbee2mqtt #15087: IKEA Styrbar remote E2001/E2002 doesnn't pair with Sonoff Zigbee3.0 USB Dongle

kovaccs, 2022-11-19:
### What happened?

IKEA STYRBAR E2001 (https://www.zigbee2mqtt.io/devices/E2001_E2002.html) couldn't pair to the coordinator (Sonoff Dongle).
The IKEA remote device does nothing, but I tried the pairing very close to the coordinator (its LED remains dark even after four button presses, and after holding it down for 10 seconds.
When I try to pair it with the IKEA repeatther, it pretends to succeed, but then the router (https://www.zigbee2mqtt.io/devices/E1746.html) disappears from the Zigbee network. The other Zigbee devices connected to the corrdinator are working properly. 
The pressing the button four times in a row is not suitable for pairing, nothing happens on the device.  :-O
Multiple attempts give the same result. 

[
Sonoff Zigbee3.0 USB Dongle and IKEA Repeather runing factory firmwares.
Zigbee2MQTT running as a Home Assiatant ad-on under HassOS ver. 9.2.
Home Assiatant Core ver. 2022.11.3
Home Assistant Supervisor ver. 2022.10.2
Zigbe2MQTT ver. 1.28.2-1
]
 

### What did you expect to happen?

Can't pair with coordinator. Pair with IKEA router, but after that the router leave the network.

### How to reproduce it (minimal and precise)

After many attempts, the result is the same

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

mvklingeren, 2022-11-20:
Same problem here, and i can't pair the remote anymore either.

I had a spare STYRBAR remote here that i was able to replace it with. 

I guess it has to do with the latest firmware of these remotes.

GSzabados, 2022-11-27:
There has been no new firmware released for the STYBAR remotes since release.

https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

kovaccs, 2022-11-28:
> There has been no new firmware released for the STYBAR remotes since release.
> 
> https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

Ok, but this is not a user-friendly situation! :-(

GSzabados, 2022-11-28:
Csaba,

I am just guessing from your description, but here are my thoughts.

1. The LED cannot stay dark, it should light up to show that the device entered into pairing mode. Either you are not pressing quickly enough the button on the back or the batteries are dead, but as you can use the hold down method, I guess the first one happens.
2. Pairing by the 10 second hold down option is utilizing the LightLink/TouchLink (I think that is how it called) method. To pair a device that way to Z2M you need to use the coordinator. Move the device close to the coordinator's antenna and in the Z2M menu use the Touchlink option, then the scan button. The device would pair this way to the network.
3. I don't know what repeater you are using, is it the IKEA repeater with the USB port or just an IKEA socket or bulb. As far as I know from some Reddit posts, the socket has a maximum 6 end device limit. Maybe the repeater has similar, not sure. But what I can tell, bulbs used to behave weirdly to TouchLink pairing. If the remote was not paired to the network and a TouchLink pairing was invoked between the remote and a bulb, the remote just took the bulb off from the network, and paired it directly to the remote. But if the remote was paired to the same network as the bulb, then the TouchLink pairing to the bulb just initiated a group binding, so both stayed on the same network, but the remote was directly controlling the bulb. I have not tested this with the STYRBAR remotes, but it used to work this way with the 5 button remotes and bulbs. And probably this is what you can see by the device (repeater) leaving the network. (If I understand correctly what you meant in the OP.)

The only things what I can suggest to try is maybe try option 2 to pair the remote with TouchLink directly the coordinator. Or try to press more times (6 or 8) the button quickly, so you will reach the 4 press in 5 seconds somehow. With the presses it is really important that you have to press quickly, as you have 5 seconds for the 4 presses. 

I hope it helps!

If you need some other help, feel free to drop me a private message on the HA community forum, even in Hungarian. :wink:

Gábor

kovaccs, 2022-11-28:
> Csaba,
> 
> I am just guessing from your description, but here are my thoughts.
> 
> 1. The LED cannot stay dark, it should light up to show that the device entered into pairing mode. Either you are not pressing quickly enough the button on the back or the batteries are dead, but as you can use the hold down method, I guess the first one happens.
> 2. Pairing by the 10 second hold down option is utilizing the LightLink/TouchLink (I think that is how it called) method. To pair a device that way to Z2M you need to use the coordinator. Move the device close to the coordinator's antenna and in the Z2M menu use the Touchlink option, then the scan button. The device would pair this way to the network.
> 3. I don't know what repeater you are using, is it the IKEA repeater with the USB port or just an IKEA socket or bulb. As far as I know from some Reddit posts, the socket has a maximum 6 end device limit. Maybe the repeater has similar, not sure. But what I can tell, bulbs used to behave weirdly to TouchLink pairing. If the remote was not paired to the network and a TouchLink pairing was invoked between the remote and a bulb, the remote just took the bulb off from the network, and paired it directly to the remote. But if the remote was paired to the same network as the bulb, then the TouchLink pairing to the bulb just initiated a group binding, so both stayed on the same network, but the remote was directly controlling the bulb. I have not tested this with the STYRBAR remotes, but it used to work this way with the 5 button remotes and bulbs. And probably this is what you can see by the device (repeater) leaving the network. (If I understand correctly what you meant in the OP.)
> 
> The only things what I can suggest to try is maybe try option 2 to pair the remote with TouchLink directly the coordinator. Or try to press more times (6 or 8) the button quickly, so you will reach the 4 press in 5 seconds somehow. With the presses it is really important that you have to press quickly, as you have 5 seconds for the 4 presses.
> 
> I hope it helps!
> 
> If you need some other help, feel free to drop me a private message on the HA community forum, even in Hungarian. 😉
> 
> Gábor

Köszönöm Gábor a részletes foglalkozást, meg is oldódott! Üdv. Csaba.

Thank you Gabor for detailed instructions!
The mistake I made was that I didn't press fast enough four times to pair with the coordinator.
Based on your suggestion, I did this and the Zonoff Zigbes dongle added the STYRBAR to the network without any problems with the factory firmware. I think the IKEA router (not light bulbs and other equipment) would also have been paired with this technique.

The issue can be closed.

GSzabados, 2022-11-28:
@mvklingeren, do you have any success as well? 

@kovaccs, as you opened the issue, you can close it as well.

kovaccs, 2022-11-28:
The problem raised is not a Zigbe2mqtt error that was solved by user intervention.

closed 2022-11-28 by kovaccs

jperquin, 2024-06-10:
FWIW - these devices are very finicky with batteries. After a FW update (which takes a good 90 mins.) all 3 of my E2002's would not come back on, even though the (retained) battery state in Z2M did not indicate any issues. A couple of fully charged LADDA AAA's fixed the it though.