Koenkk/zigbee2mqtt #15092: Support for Tuya _TZ3000_3a9beq8a

vk496, 2022-11-19:
### Link

https://a.aliexpress.com/_Ew8CLqN

### Database entry

Idk where should get this line, sorry. _TZ3000_3a9beq8a

### Comments

Hello,

This device pairs correctly as simple switch TS0001. Adding it to the Tuya WDH02 exposes new useful options, like switch_type or power on behavior, which seems to work properly.

HOWEVER, what is not working is the report of the state. It changes from the Z2M, but manual switch is not reported back to Z2M. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        fingerprint: [{modelID: 'TS0001', manufacturerName: '_TZ3000_hktqahrq'}, {manufacturerName: '_TZ3000_hktqahrq'},
            {manufacturerName: '_TZ3000_q6a3tepg'}, {modelID: 'TS000F', manufacturerName: '_TZ3000_m9af2l6g'},
            {modelID: 'TS000F', manufacturerName: '_TZ3000_mx3vgyea'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_npzfdcof'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_5ng23zjs'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_rmjr4ufz'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_v7gnj3ad'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_ark8nv4y'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_mx3vgyea'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_qsp2pwtf'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_3a9beq8a'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_46t1rvdu'}],
        model: 'WHD02',
        vendor: 'TuYa',
        whiteLabel: [{vendor: 'TuYa', model: 'iHSW02'}],
        description: 'Wall switch module',
        toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type]),
        fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type]),
        exposes: extend.switch().exposes.concat([e.power_on_behavior(), e.switch_type_2()]),
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
    }
]
```

Koenkk, 2022-11-20:
Added!
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-20 by Koenkk

vk496, 2022-11-20:
@Koenkk thanks for putting it in dev, but it still have the bug of not reporting properly the state to Z2M. Any idea?

Koenkk, 2022-11-20:
What if you enable if via the reporting tab of the device (cluster genOnOff, attribute `onOff`)

vk496, 2022-11-21:
Hi,

I think it doesn't work. I don't have that exposed by default, but what I do is to repair (so it's done from Z2M side) and I see the Coordinator binded to it. But is still the same behavior. 

The status is reflected if I manually refresh the state. I attach the clusters, in the case you may find them useful.

[image]

Koenkk, 2022-11-21:
In the configure, below `await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);` can you add:

```js
await reporting.onOff(endpoint);
```

vk496, 2022-11-24:
Hello @Koenkk and sorry for late response.

It seems that device is not answering for that binding. I played with the manual switch in order to generate some logs:

```
Failed to configure 'zs_luz_salon_2', attempt 4 (ReferenceError: endpoint is not defined at Object.configure (/app/dist/util/externally-loaded.js:34:35) at Configure.configure (/app/lib/extension/configure.ts:115:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21))

Error 2022-11-24 23:40:36Publish 'get' 'state' to 'zs_luz_salon_2' failed: 'Error: Read 0xcc86ecfffe0d91cb/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

Info 2022-11-24 23:40:36MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'state' to 'zs_luz_salon_2' failed: 'Error: Read 0xcc86ecfffe0d91cb/1 genOnOff([\"onOff\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"zs_luz_salon_2"},"type":"zigbee_publish_error"}'

Debug 2022-11-24 23:40:54Received MQTT message on 'zigbee2mqtt/zs_luz_salon_2/get' with data '{"state":""}'

Debug 2022-11-24 23:40:54Publishing get 'get' 'state' to 'zs_luz_salon_2'

Debug 2022-11-24 23:40:54Received Zigbee message from 'zs_luz_salon_2', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null

Info 2022-11-24 23:40:54MQTT publish: topic 'zigbee2mqtt/zs_luz_salon_2', payload '{"linkquality":255,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'

Debug 2022-11-24 23:40:58Received MQTT message on 'zigbee2mqtt/zs_luz_salon_2/get' with data '{"state":""}'

Debug 2022-11-24 23:40:58Publishing get 'get' 'state' to 'zs_luz_salon_2'

Debug 2022-11-24 23:40:58Received Zigbee message from 'zs_luz_salon_2', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null

Info 2022-11-24 23:40:58MQTT publish: topic 'zigbee2mqtt/zs_luz_salon_2', payload '{"linkquality":255,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'

Debug 2022-11-24 23:41:01Received MQTT message on 'zigbee2mqtt/zs_luz_salon_2/get' with data '{"state":""}'

Debug 2022-11-24 23:41:01Publishing get 'get' 'state' to 'zs_luz_salon_2'

Debug 2022-11-24 23:41:01Received Zigbee message from 'zs_luz_salon_2', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null

Info 2022-11-24 23:41:01MQTT publish: topic 'zigbee2mqtt/zs_luz_salon_2', payload '{"linkquality":255,"power_on_behavior":"previous","state":"ON","switch_type":"toggle"}'

Debug 2022-11-24 23:41:06Received MQTT message on 'zigbee2mqtt/zs_luz_salon_2/get' with data '{"state":""}'

Debug 2022-11-24 23:41:06Publishing get 'get' 'state' to 'zs_luz_salon_2'

Debug 2022-11-24 23:41:06Received Zigbee message from 'zs_luz_salon_2', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null

Info 2022-11-24 23:41:06MQTT publish: topic 'zigbee2mqtt/zs_luz_salon_2', payload '{"linkquality":255,"power_on_behavior":"previous","state":"OFF","switch_type":"toggle"}'

```

vk496, 2022-11-24:
Nvm. `endpoint` was not defined before, so it was failing. After your fix, works perfectly!

This is the final configure part that works properly:

```
configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await reporting.onOff(endpoint);
        },
```

Koenkk, 2022-11-25:
Great, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)