Koenkk/zigbee2mqtt #15101: Legrand netatmo 600086 ( 067776 on Z2M) reports 'OPEN" when closed and vice versa

novisys, 2022-11-20:
### What happened?

What happened?
Legrand netatmo Dooxie 600086 (show as 067776 on Z2M) shutter reports 'OPEN" when closed and vice versa
invert_cover : true or false, the STATE of the shutter roller is always the wrong state.
invert_cover only related to the percentage the roller shutter is opened.
Also when using STOP button: this one change the state to OPEN by default, on my use i need to STOP the shutter after is FULLY CLOSED because it's continue to try open it, and with this option STOP my shutter is showing as OPEN when it's CLOSED !
Very annoying as lot of checks are done on the state of the roller shutter and not on the percentage.
Maybe add possibility to invert the state too?
same as issue: [Legrand netatmo Valena Next roller shutter reports 'OPEN" when closed and vice versa](https://github.com/Koenkk/zigbee2mqtt/issues/14425) (#14425)

but it's not corrected on my side !

### What did you expect to happen?

When the roller shutter is opened, the state will be "OPEN" and vice versa.

### How to reproduce it (minimal and precise)

Simple to reproduce with this device: the state is always wrong on my 5 products[Legrand netatmo Valena Next roller shutter reports 'OPEN" when closed and vice versa](https://github.com/Koenkk/zigbee2mqtt/issues/14425) (#14425)

### Zigbee2MQTT version

 1.28.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Koenkk, 2022-11-20:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    zigbeeModel: [' Shutter switch with neutral\u0000\u0000\u0000'],
    model: '067776',
    vendor: 'Legrand',
    description: 'Netatmo wired shutter switch',
    // the physical LED will be green when permit join is true, off otherwise and red when not linked
    fromZigbee: [
        // Devices can send an identify message when the configuration button is pressed
        // (behind the physical buttons)
        // Used on the official gateway to send to every devices an identify command (green)
        fz.identify, fz.ignore_basic_report,
        // support binary report on moving state (supposed)
        fz.legrand_binary_input_moving, fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.legrand_identify, tz.legrand_settingEnableLedInDark],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

novisys, 2022-11-20:
i already tested with another external convert and it;s worked !
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [' Shutter switch with neutral\u0000\u0000\u0000'],
    model: '600086',
    vendor: 'Legrand',
    description: 'Netatmo wired shutter switch CUSTOM',
    // the physical LED will be green when permit join is true, off otherwise and red when not linked
    fromZigbee: [
        // Devices can send an identify message when the configuration button is pressed
        // (behind the physical buttons)
        // Used on the official gateway to send to every devices an identify command (green)
        fz.identify, fz.ignore_basic_report,
        // support binary report on moving state (supposed)
        fz.legrand_binary_input_moving, fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt, tz.legrand_identify],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;
```

with yours it's work when i change this line:
`    meta: {coverInverted: true},
`
to 
`    meta: {coverInverted: false},
`

Koenkk, 2022-11-20:
I just checked and the `converInverted: true` was added by @McGiverGim in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4770.

I guess the devices are slightly different, can you both post the data/database.db entry of this device?

novisys, 2022-11-20:
> converInverted

Thanks a lot for your time and your Help 👍 
here is the data:
{"id":71,"type":"Router","ieeeAddr":"0x0004740000a3bcfd","nwkAddr":63332,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Shutter switch with neutral\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,258,4,5,15,64513],"outClusterList":[258,0,64513,5,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"zclVersion":2}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":100}}},"binds":[{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00212effff07347b","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00212effff07347b","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff07347b","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":8,"dateCode":" \u000020220513\u0000\u0000\u0000\u0000\u0000","swBuildId":"002f","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-904845555},"lastSeen":1668967419076,"defaultSendRequestWhen":"immediate"}

novisys, 2022-11-21:
any update to integrate it without External_converter ?

Koenkk, 2022-11-22:
Waiting for the data/database.db entry from @McGiverGim

Koenkk, 2022-11-22:
@McGiverGim i see

Included the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SirJohnDoe, 2022-12-11:
I'm not sure exactly what changed, but ever since this update the shade cover reports the wrong state. Regardless of the 'Invert_cover' setting, the switch reports 'CLOSE' as state when it's up (opened). Personally I think the naming is quite confusing, as you could describe it as open/close for either position: up (open) and down (closed) would make more sense to me. In any case, in HA it shows as closed (pic from [this](https://www.home-assistant.io/integrations/cover/) page):

[image]

Inverting (=true) works, as the buttons will show in HA as open, but the toggle service in HA won't work due to the state still being 'CLOSE' and the history in HA will also show closed. I would say that the state should match the position as described at the 'invert_cover' toggle, this is currently not the case. Invert = false shows position as 0, state is CLOSE (is in fact open). Invert = true shows position is 100, state is CLOSE (is in fact open). The wiring of the switch is correct, as the physical buttons work as expected.

I run Z2M 1.28.4, and the firmware/builddate of the Legrand is 001a/20200605. My adapter is the Sonoff Zigbee 3.0 Dongle Plus. The switch is a [Legrand 741807](https://www.elektrobode.nl/products/legrand-valena-next-with-netatmo-valena-next-wn-geconnecteerde-rolluikschakelaar-wit-741807) that shows up as Legrand 067776.

SirJohnDoe, 2022-12-18:
Sure thing:

`{"id":14,"type":"Router","ieeeAddr":"0x00047400008cfefd","nwkAddr":6064,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Shutter switch with neutral\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,258,4,5,15,64513],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"modelId":" Shutter switch with neutral\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020200605\u0000\u0000\u0000\u0000\u0000","swBuildId":"001a"}},"genIdentify":{"attributes":{"identifyTime":60}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":100}},"genBinaryInput":{"attributes":{"presentValue":0}}},"binds":[{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad88d","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad88d","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b00194ad88d","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1}],"meta":{}}},"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020200605\u0000\u0000\u0000\u0000\u0000","swBuildId":"001a","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-904845555},"lastSeen":1671365379828,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2022-12-18:
Seems yours has different (older) software on it. I pushed a fix, let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

SirJohnDoe, 2022-12-18:
Thank you! It works. State is now 'OPEN' when up, and the toggle service in HA works correctly.

Koenkk, 2022-12-19:
Great!

closed 2022-12-19 by Koenkk

stefanuytterhoeven, 2023-01-01:
1.29.0-1 reintroduced the error again. Please correct?

SirJohnDoe, 2023-01-01:
I just switched from edge to 1.29.0 and it still works fine, as expected.

stefanuytterhoeven, 2023-01-02:
How is this possible? Like with previous update, the state was open, when the cover was closed. 
And the four days later, it's ok again. Without an update?? So strange. And yes, after the update, I restarted ha several times, and also started z2m several times. ?'?

icagn, 2023-01-04:
Le problème est résolu pour 067776_001a, mais pas pour 067776

Minique85, 2023-02-11:
Leaving a comment on my experience here, it may help someone. 

I am using Zigbee2MQTT on SONOFF Dongle Plus integrated in Home Assistant.
I had two main issues:
1. The reported position was incorrect, and changing the setting invert_cover on the device did not help (the state was opposite, I could then not open a close cover from HA)
2. I did not get any reports from the button after controlling remotely, the position was unknown. On manual button press the state was immediate - but then I had the problem 1 above.

I am using Roller Shutter buttons that date from 2019. I decided to upgrade them all to firmware version 47 before posting here - I did this using the Legrand Gateway (pack 199109). It took about 4 hours to upgrade all my Legrand devices.

After upgrade, the 1st problem was solved. But the reporting issue remained (nb 2).

I then changed under each device the reporting settings "max rep interval" to 60 (seconds). And that solved my second issue. The button does report its state after 60s. By that time my roller has reached its position. Default value was 62000!

Now Home Assistant cards are working fine and all behave as expected.

stefanuytterhoeven, 2023-08-02:
I just did an update of Zigbee2MQTT in Home assistant.
and this problem came back.
So, physically, the cover is OPEN (UP)
Now, Zigbee2MQTT has current_position: 0, and status is CLOSED.
invert_cover: FALSE

Setting invert_cover:TRUE doesn't solve it.

I've had this error before, and it was solved. But, with this new release, it came back. 
Can you have a look at it?

AmauryFages, 2024-05-21:
Same in may 2024 with some EVENO cover reporting 100 when actually it's opened ...
that's kind of bullshit problem appearing here and there according to covers vendor.