Koenkk/zigbee2mqtt #15102: Zigbee2mqtt fail to start

kiker666, 2022-11-20:
### What happened?

Hello, i am stuck with this issue. Could anyone help me a bit pleaseee?
NAS synology dsm7.1 version. 
HA running in a docker. Mqtt addon installed and running.
I bought sonoff dongle plus p, flashed zigbee2mqtt coordinator firmware.
I add drivers and controllers, and NAS detect usb as /dev/ttyUSB0:
[image]
Then i created a docker container zigbee2mqtt ( yaml pic attached)
[image]
But then, when i try to run it, says:
[image]

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20220219

### Adapter

Dongle Plus "ZBDongle-P" (based on CC2652P)

Brix1330, 2022-11-20:
Got this problem as well after updating HA to 2022.11.3, somehow after updating the dongle is no longer detected on _/dev/ttyUSB0_ but moved to _/dev/ttyUSB1_. 

Have you checked on _HA > Settings > System > Hardware > All Hardware_ (under three dots; right corner)?

thomashogema, 2022-11-21:
Had a similar issue after updating to 2022.11.3 while running Home Assistant. The dongle location in the hardware tab did not change, but Z2M could not find it.
For me, the solution was to restart the host machine.

closed 2022-12-29 by github-actions[bot]