Koenkk/zigbee2mqtt #15108: [New device support]: Bseed Zigbee Glass Touch 2gang

Hausy1986, 2022-11-20:
### Link

https://de.aliexpress.com/item/1005002842345807.html?spm=a2g0o.productlist.0.0.69645327XLPIBu&algo_pvid=f16dfabe-2215-46ed-b221-079285dfd9cf&aem_p4p_detail=202211201008345513992527262440005301746&algo_exp_id=f16dfabe-2215-46ed-b221-079285dfd9cf-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028845537231%22%7D&pdp_npi=2%40dis%21EUR%2138.1%2111.0%21%21%21%21%21%402100bde116689677146098317e47fb%2112000028845537231%21sea&curPageLogUid=2vZwyV9pVa2G&ad_pvid=202211201008345513992527262440005301746_1&ad_pvid=202211201008345513992527262440005301746_1

### Database entry

from zigbee-adapter in iobroker: {"type":"attributeReport","data":{"onOff":0},"linkquality":255,"groupID":null,"cluster":"genOnOff","meta":{"zclTransactionSequenceNumber":93,"manufacturerCode":null,"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0}},"endpoint_id":2}

### Comments

Hallo, hoffe mir kann hier einer helfen und dieses Gerät implementieren. Ich weiß das leider nicht mehr weiter.
[image]

Und zwar folgendes: Der Lichtschalter von Bseed mit den 2 Touch-Relais wird zwar erkannt, aber als ein anderes Gerät = [https://www.zigbee2mqtt.io/devices/TS0012.html](url)
[image]

Eigentlich wäre es kein Problem, weil bei dem Lichtschalter mit einem touch funktioniert es auch, nur hier schalten sich immer **beide Relais** mit, egal ob ich left oder right state anwähle. Auch das löschen eines der state's bringt keine Veränderung. 

Ich würde mich freuen, wenn sich einer findet und mir hilft bzw. das Gerät mit aufnimmt, da ich leider mehrere davon nutzen wollte/muss :D. Ich würde auch gerne noch mehr Daten bereitstellen nur habe ich das mit Zigbee2mqtt und mosquito nicht hinbekommen, musste sogar ein komplettes Backup auf meinem Raspberry neu aufspielen.

Gruß 
Hausy

danieledwardgeorgehitchcock, 2022-11-21:
It is common for Zigbee2MQTT to support generic white labelled devices using a single converter as you see above.

I believe that there is already a bug report (will update my comment when I find it) regarding TS0012 device states working incorrectly

Hausy1986, 2022-11-22:
Erstmal vielen Dank für die Hilfestellung. Ich habe versucht Zigbee2MQTT auf meinen RaspberryPi zu installieren, leider kriege ich es nicht hin und musste schon wieder für das ganze System ein Backup einspielen, da auch der Zigbee Adapter im Iobroker, womit ich die Geräte hinzufüge, nicht mehr funktionierte.

Gibt es eine Möglichkeit über den Adapter im Iobroker an die Daten zu kommen? 

_Google Translate:_
Many thanks for your help. I tried to install Zigbee2MQTT on my RaspberryPi, unfortunately I can't do it and had to import a backup for the whole system again, because the Zigbee adapter in Iobroker, with which I add the devices, no longer worked.

Is there a way to get the data via the adapter in the iobroker?

closed 2023-01-25 by github-actions[bot]