Koenkk/zigbee2mqtt #15111: Can't togle options/configuration (power_outage_memory) Aqara Smart Plug ZNCZ12LM

horspiste, 2022-11-21:
### What happened?

The Aqara has multiple configuration options (led_disabled_night,power_outage_memory,auto_off,overload_protection) and they are exposed properly in the "Expose" tab for the device. However when I try to configure them using the Zigbee2mqtt interface I get the following error: (for power_outage_memory). 

Publish 'set' 'power_outage_memory' to 'Prise Chauffe Eau' failed: 'Error: Write 0x00158d0008833818/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### How to reproduce it (minimal and precise)

Just click on the true/false switch on the expose tab for the device for any of the configurable options.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

tube_zb_gw_cc2652p2_version_2

### Debug log

 2022-11-20 22:09:12Received MQTT message on 'zigbee2mqtt/Prise Chauffe Eau/set' with data '{"power_outage_memory":true}'
Debug 2022-11-20 22:09:12Publishing 'set' 'power_outage_memory' to 'Prise Chauffe Eau'
Error 2022-11-20 22:09:12Publish 'set' 'power_outage_memory' to 'Prise Chauffe Eau' failed: 'Error: Write 0x00158d0008833818/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Debug 2022-11-20 22:09:12Error: Write 0x00158d0008833818/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:2214:17) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)

horspiste, 2022-12-22:
I haven’t found a solution yet.

ricerocketjockey, 2022-12-22:
I am also having this problem with the same plug.

`Zigbee2MQTT:error 2022-12-22 13:52:04: Publish 'set' 'power_outage_memory' to 'Power' failed: 'Error: Write 0x00158d0008833891/1 aqaraOpple({"513":{"value":1,"type":16}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

DannyFeliz, 2023-01-07:
I have the same issue I also tried with the [mqtt explorer]([MQTT Explorer Addon](https://community.home-assistant.io/t/mqtt-explorer-addon/465758)) and got the same results.

thekev, 2023-01-23:
Same issue here. I have a zigbee network capture of the error, but it may not be insightful. I have a sonoff hub coming today, and I'll work on capturing the vendor-specific interaction for the feature.

thekev, 2023-01-27:
I was able to grab a capture. It looks more like the payload of the ZNCZ11LM. I was able to set it by moving `ZNCZ12LM` to line [2158](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L2158) and the switch behaves as expected after power loss. Moving the getter still threw an unsupported attribute error. I didn't capture that transaction from the aqara hub though. I'm not able to put more time into it right now, but I may go for a pull request that covers the full coverage of options.

victorstein, 2023-01-27:
@thekev is the hero we need not the one we deserve.

smos08, 2023-02-19:
> I was able to grab a capture. It looks more like the payload of the ZNCZ11LM. I was able to set it by moving `ZNCZ12LM` to line [...]

that would be amazing, thank you

handcoding, 2023-03-27:
@thekev Can anyone walk me through how to apply the fix that @thekev mentioned? I’m cursed with this same smart plug, and if I could just get the settings to stick, I think I’d be able to use this.

hornet-dev, 2023-04-13:
I have the same issue, and am willing to help test any changes or provide additional info. Just let me know.

DannyFeliz, 2023-05-14:
Bot, this issue still exists.

handcoding, 2023-06-15:
Still present.

asapkenzy, 2023-07-12:
this issue still exists.

ravichowdhury88, 2023-07-13:
Is there a plan on getting this resolved ?

Koenkk, 2023-07-13:
I've added the fix from @thekev , so this should be fixed now

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

thekev, 2023-07-13:
Thanks @Koenkk for patching that. I haven't had the will or free time to do a PR.

smos08, 2023-07-13:
thank you @Koenkk from all of us users

DannyFeliz, 2023-07-13:
@Koenkk thank you for fixing this 🙏🏻

Koenkk, 2023-07-15:
Credits go the @thekev , I just integrated the change 😄 

I assume this issue can be closed now.

closed 2023-07-15 by Koenkk