Koenkk/zigbee2mqtt #15112: Moes EDM-1ZBB-EU, Scenes not working

Mikescotland, 2022-11-21:
### What happened?

Moes dimmer TS0601_dimmer (tuya) has ability to change brightness without switching on the light.
1. Manually adjusting brightness in the exposes tab works, light is next time switched on to the required brightness.
2. Scenes do not work, saving different brightness doesn't change any value on the device, more, sometimes scene where the switch ON/OFF is off, recalls with switch toggled to ON.

In the log first change is via brightness slider, second via scene, I can see they have different parameters (setpoint etc) 

### What did you expect to happen?

Scenes would save properly without a glitch turning sometimes light on and scenes would affect only brightness level, exactly like manually can be done from "exposes" tab. 

### How to reproduce it (minimal and precise)

1. Turn off the light, change brightness manually, turn on - brightness remembered.
2. Create 2 scenes with different brightness values and save - recall doesn't work. 

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

Sonoff plus

### Debug log

 2022-11-21 04:02:42Received MQTT message on 'zigbee2mqtt/Main Light LR/set' with data '{"brightness":254}'
Debug 2022-11-21 04:02:42Publishing 'set' 'brightness' to 'Main Light LR'
Debug 2022-11-21 04:02:42Received Zigbee message from 'Main Light LR', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1}' from endpoint 1 with groupID 0
Info 2022-11-21 04:02:42MQTT publish: topic 'zigbee2mqtt/Main Light LR', payload '{"brightness":254,"color_mode":"xy","linkquality":98,"min_brightness":4,"state":"OFF"}'
Debug 2022-11-21 04:02:47Received MQTT message on 'zigbee2mqtt/Main Light LR/set' with data '{"scene_recall":2}'
Debug 2022-11-21 04:02:47Publishing 'set' 'scene_recall' to 'Main Light LR'
Info 2022-11-21 04:02:47MQTT publish: topic 'zigbee2mqtt/Main Light LR', payload '{"brightness":9,"color_mode":"xy","linkquality":98,"min_brightness":4,"state":"OFF"}'
Debug 2022-11-21 04:02:47Received Zigbee message from 'Main Light LR', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":256}' from endpoint 1 with groupID 0
Info 2022-11-21 04:02:47MQTT publish: topic 'zigbee2mqtt/Main Light LR', payload '{"brightness":9,"color_mode":"xy","linkquality":98,"min_brightness":4,"state":"OFF"}'

Koenkk, 2022-11-21:
Since this a TuYa device (which uses a custom api) these standard Zigbee features don't seem to be supported.

Mikescotland, 2022-11-21:
> features don't seem to be supported. Nothing we can do about it from the
> z2m side.

I understand, I'm just pointing that brightness works with off status from
exposes tab, Scenes seem to be not working. Looks like something in the
scene saving / recalling doesn't properly work.
It's i believe, completely not related to tuya protocol, as standard
brightness change without switching on the bulb from the exposes tab works
fine.

Mikescotland, 2022-11-27:
Solution for those who own similar Moes dimmers (also the modular ones), you can change the brightness using zigbee2mqtt using HA automations, where action is as follows:

service: mqtt.publish
data:
  topic: zigbee2mqtt/YOUR_DEVICE/set
  payload: "{\"brightness\":254}"

This changes the brightness without physically switching on the lights.

closed 2023-01-04 by github-actions[bot]