Koenkk/zigbee2mqtt #15118: EndDevice connects to coordinator but leaves after configuration when connected to router

LukevdPalen, 2022-11-21:
### What happened?

In y situation I have some Zigbee[ temperature sensors](https://nl.aliexpress.com/item/1005003518393383.html?spm=a2g0o.order_list.0.0.700779d2dWQkM8&gatewayAdapt=glo2nld) in my house that were previously directly connected to my coordinator (Sonoff ZB Bridge) flashed with [Tasmota](https://zigbee.blakadder.com/Sonoff_ZBBridge.html). This was all working perfectly until I decided to add some [Ikea Tradfri Power Sockets](https://www.ikea.com/nl/en/p/tradfri-wireless-control-outlet-smart-90356166/) to extend the range of my network. I use Zigbee2Mqtt and allowed the device to connect via the socket. But when I put the temperature sensor into configuration mode, it connects to the router and shows up in Zigbee2Mqtt but immediately leaves the network a few times. So a few questions that are crossing my mind..

1. Is the router responsible to allow devices to acces to the network, and if so, why does it successfully complete the interview and leaves the network? Or is the router just passing the signal to the coordinator?
2. I only have devices right now from Ikea that acts as routers, so it's hard for me to test if this also happens with routers of other brands. Are Ikea Zigbee product known for being incompatible with other Zigbee devices?
3. Or could it also be related to the Coordinator or EndDevice?

So to sum up:
Coordinator <> EndDeviced = working perfectly
Coordinator <> Router <> EndDevice = Configures devices, but EndDeviced leaves network immediately.

[image]

### What did you expect to happen?

Devices not leaving the network after succesfull interview

### How to reproduce it (minimal and precise)

Pairing is enabled on Ikea Socket (router)
Device enters pairing mode
Pairing is successful, device is fully functional
After a few seconds device leaves the network with message "Device xxx left the network"

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Zigbee Bridge EFR32MG21

### Debug log

```
Debug 2022-11-21 12:21:07Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Socket TV","time":254,"transaction":"dymq8-1","value":true}'
Info 2022-11-21 12:21:07Zigbee: allowing new devices to join via Socket TV.
Info 2022-11-21 12:21:08MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Socket TV","time":254,"value":true},"status":"ok","transaction":"dymq8-1"}'
Info 2022-11-21 12:21:18Device 'Living' joined
Info 2022-11-21 12:21:18MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living","ieee_address":"0xa4c1380e24690dcf"},"type":"device_joined"}'
Info 2022-11-21 12:21:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Living"},"type":"device_connected"}'
Info 2022-11-21 12:21:18MQTT publish: topic 'zigbee2mqtt/Living/availability', payload 'online'
Info 2022-11-21 12:21:18Starting interview of 'Living'
Info 2022-11-21 12:21:18MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living","ieee_address":"0xa4c1380e24690dcf","status":"started"},"type":"device_interview"}'
Info 2022-11-21 12:21:18MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Living"},"type":"pairing"}'
Debug 2022-11-21 12:21:19Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0201"}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:19Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:19Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_8ybe88nf"}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:19Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:20Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:20Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:21Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:21Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:21Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"appVersion":69}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:21Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:22Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:22Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:22Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:22Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:22Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:22Skipping message, definition is undefined and still interviewing
Debug 2022-11-21 12:21:23Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:23Skipping message, definition is undefined and still interviewing
Info 2022-11-21 12:21:23Successfully interviewed 'Living', device has successfully been paired
Info 2022-11-21 12:21:23Device 'Living' is supported, identified as: TuYa Temperature & humidity sensor (IH-K009)
[... 20 lines cut ...]
Info 2022-11-21 12:21:40MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1380e24690dcf"},"type":"device_removed"}'
Info 2022-11-21 12:21:43Device 'Living' joined
Info 2022-11-21 12:21:43MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living","ieee_address":"0xa4c1380e24690dcf"},"type":"device_joined"}'
Info 2022-11-21 12:21:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Living"},"type":"device_connected"}'
Info 2022-11-21 12:21:43Starting interview of 'Living'
Info 2022-11-21 12:21:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living","ieee_address":"0xa4c1380e24690dcf","status":"started"},"type":"device_interview"}'
Info 2022-11-21 12:21:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Living"},"type":"pairing"}'
Debug 2022-11-21 12:21:45Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:45Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Debug 2022-11-21 12:21:45Received Zigbee message from 'Living', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info 2022-11-21 12:21:45Successfully interviewed 'Living', device has successfully been paired
Info 2022-11-21 12:21:45Device 'Living' is supported, identified as: TuYa Temperature & humidity sensor (IH-K009)
Info 2022-11-21 12:21:45MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Voltage of the battery in millivolts","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"IH-K009","options":[{"access":2,"description":"Number of digits after decimal point for temperature, takes into effect on next report of device.","name":"temperature [... cut ...]
Info 2022-11-21 12:21:48MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1380e24690dcf"},"type":"device_removed"}'
Warning 2022-11-21 12:21:48Device 'Living' left the network
Info 2022-11-21 12:21:48MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Living","ieee_address":"0xa4c1380e24690dcf"},"type":"device_leave"}'
Info 2022-11-21 12:21:48MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xa4c1380e24690dcf"},"type":"device_removed"}'

```

closed 2022-11-23 by LukevdPalen

mebezac, 2022-11-28:
@LukevdPalen how did you solve this?

droneman83, 2022-12-23:
i also have same isus with this model and have had it for some time now :/