Koenkk/zigbee2mqtt #15120: TS110E _TZ3210_3mpwqzuu broken power outage memory

ster1um, 2022-11-21:
### What happened?

When change the power outage memory from restore to on or off is not possible to return to restore setting, the device remember last selection of on or off. 
Whit tuya gateway work well.
Only one way for work is remove and repair and unselected power outage because the default preset is restore.  

### What did you expect to happen?

Fix the broken selection

### How to reproduce it (minimal and precise)

change power outage memory from restore to on or off 

### Zigbee2MQTT version

1.28.2-dev and latest stable

### Adapter firmware version

20220928

### Adapter

sonoff dongle plus cc2652p

Koenkk, 2022-11-21:
Can you sniff the traffic when setting this with the TuYa gateway?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

ster1um, 2022-11-22:
> Can you sniff the traffic when setting this with the TuYa gateway?
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

yes i can, but from TI download is broken... You have the zip swrc045z.zip?

ster1um, 2022-11-22:
I find this

```
///// POWER ON BAV. OFF

Frame 2638: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 210
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 43
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 43
    Command: Write Attributes (0x02)
    Attribute Field, Enum8: 0
        Attribute: Unknown (0x8002)
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 0 (0x00)

///// POWER ON BAV. ON

Frame 3660: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 44
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 1
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 1
    Command: Write Attributes (0x02)
    Attribute Field, Enum8: 1
        Attribute: Unknown (0x8002)
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 1 (0x01)

///// POWER ON BAV. MEMORY

Frame 4064: 52 bytes on wire (416 bits), 52 bytes captured (416 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 81
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 34
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 34
    Command: Write Attributes (0x02)
    Attribute Field, Enum8: 2
        Attribute: Unknown (0x8002)
        Data Type: 8-Bit Enumeration (0x30)
        Uint8: 2 (0x02)

```

ster1um, 2022-11-22:
Add others function: 

MIN. BRIGHTNESS ENDPOINT 2

```
MIN. BEFORE  SELECTION THE MIN. BRIGHTNESS 

Frame 8792: 53 bytes on wire (424 bits), 53 bytes captured (424 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 123
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 10
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 10
    Command: Write Attributes (0x02)
    Attribute Field, Uint16: 12
        Attribute: Unknown (0xfc03)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 12 (0x000c)  >>>  1%
 

MIN. AFTER SELECTION THE MIN. BRIGHTNESS 

Frame 8798: 53 bytes on wire (424 bits), 53 bytes captured (424 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 125
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 12
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 12
    Command: Write Attributes (0x02)
    Attribute Field, Uint16: 284
        Attribute: Unknown (0xfc03)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 284 (0x011c)  >>>  28%

```

MAX. BRIGHTNESS ENDPOINT 2

```
MAX. BEFORE  SELECTION THE MAX. BRIGHTNESS 

Frame 9678: 53 bytes on wire (424 bits), 53 bytes captured (424 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 207
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 90
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 90
    Command: Write Attributes (0x02)
    Attribute Field, Uint16: 998
        Attribute: Unknown (0xfc04)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 998 (0x03e6)  >>> 100%

MAX .AFTER SELECTION THE MAX. BRIGHTNESS
 Frame 9684: 53 bytes on wire (424 bits), 53 bytes captured (424 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0xa887, Src: 0x0000
ZigBee Network Layer Data, Dst: 0xa887, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 209
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 92
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 92
    Command: Write Attributes (0x02)
    Attribute Field, Uint16: 789
        Attribute: Unknown (0xfc04)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 789 (0x0315)
```

Koenkk, 2022-11-22:
I've checked the power on behaviour but this should be the same as Z2M sends, can you also sniff the messages when setting it via z2m so we can compare them?

Koenkk, 2022-11-23:
Can you explain what you mean with:

> MAX. BEFORE  SELECTION THE MAX. BRIGHTNESS 

>  MAX .AFTER SELECTION THE MAX. BRIGHTNESS

ster1um, 2022-11-25:
> Can you explain what you mean with:
> 
> > MAX. BEFORE  SELECTION THE MAX. BRIGHTNESS
> 
> > MAX .AFTER SELECTION THE MAX. BRIGHTNESS
hello,
after send command for min or max brightness the tuya gateway send 2 command, first after selection the level and before the new level.
I don't know the motivation for this.

I have a question for send the payload via dev console but i don't understand the format use...

```
Frame 1753: 53 bytes on wire (424 bits), 53 bytes captured (424 bits) on interface -, id 0
IEEE 802.15.4 Data, Dst: 0x8198, Src: 0x0000
ZigBee Network Layer Data, Dst: 0x8198, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 2
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 103
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 18
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 18
    Command: Write Attributes (0x02)
    Attribute Field, Uint16: 358
        Attribute: Unknown (0xfc03)
        Data Type: 16-Bit Unsigned Integer (0x21)
        Uint16: 358 (0x0166)
```
This is the command send for setting the min brightness, the attribute field I think is the payload,correct?

how should the code formed?

thanks

Koenkk, 2022-11-26:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c
- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

ster1um, 2022-11-26:
Thanks for works but work partial
It doesn't work on the z2m frontend page, work only from physical button and not report the right level of brightness it stuck on 255.
I fixed the min. brightness to e.g. 100 from button go to 100 but z2m report 255.

if you need the tuya code
```
min bri l1 	code 3  
min bri l2 	code 9 
max bri l1 	code 5
max bri l2 	code 11
cuntdown l1 	code 6 
cuntdown l2 	code 12 
state l1	code 1
state l2	code 7  
type_of_switch l1    code 4   
type_of_switch l2    code 10   
power_on_status		code 14
level bri l1	code 2  
level bri l2	code 8 
```

ster1um, 2022-11-26:
With your external convert, the selection of switch type doesn't work

Koenkk, 2022-11-27:
I didn't know this device also supports the TuYa api, that simplifies things.

It seems it is similar to the TS0601, can you try if the whole first gang (gang 2 won't work) with this converter: https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c

ster1um, 2022-11-27:
> https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c

this doesn't work 😥
I read the code from tuya devoloper board.

ster1um, 2022-11-27:
> https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c

this device not support the light type
```

            } else if (key === 'switch_type') {
                const lookup = {led: 0, incandescent: 1, halogen: 2};
                payload = {64514: {value: lookup[value], type: 0x20}};
```

Koenkk, 2022-11-27:
Can you sniff the traffic when just turning this device ON via the TuYa gateway?

ster1um, 2022-11-28:
> Can you sniff the traffic when just turning this device ON via the TuYa gateway?

see this attachment
1 of 2
[https://pastebin.com/X9KyW3c7](url)

2 of 2
[https://pastebin.com/XrMh95tA](url)

Koenkk, 2022-11-28:
Can you provide just the command that was send when turning on the device?

ster1um, 2022-11-28:
> Can you provide just the command that was send when turning on the device?

Sorry, I realized you need another thing...

```
Frame 67: 48 bytes on wire (384 bits), 48 bytes captured (384 bits) on interface -, id 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Nov 28, 2022 22:01:18.323113000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1669669278.323113000 seconds
    [Time delta from previous captured frame: 0.351520000 seconds]
    [Time delta from previous displayed frame: 0.351520000 seconds]
    [Time since reference or first frame: 13.323113000 seconds]
    Frame Number: 67
    Frame Length: 48 bytes (384 bits)
    Capture Length: 48 bytes (384 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:zbee_aps:zbee_zcl:zbee_zcl_general.onoff]
IEEE 802.15.4 Data, Dst: 0x5105, Src: 0x0000
    Frame Control Field: 0x8861, Frame Type: Data, Acknowledge Request, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..1. .... = Acknowledge Request: True
        .... .... .1.. .... = PAN ID Compression: True
        .... .... 0... .... = Reserved: False
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        10.. .... .... .... = Source Addressing Mode: Short/16-bit (0x2)
    Sequence Number: 140
    Destination PAN: 0x498b
    Destination: 0x5105
    Source: 0x0000
    [Extended Source: cc:86:ec:ff:fe:a9:a9:18 (cc:86:ec:ff:fe:a9:a9:18)]
    [Origin: 5]
    FCS: 0x353c (Correct)
ZigBee Network Layer Data, Dst: 0x5105, Src: 0x0000
    Frame Control Field: 0x0208, Frame Type: Data, Discover Route: Suppress, Security Data
        .... .... .... ..00 = Frame Type: Data (0x0)
        .... .... ..00 10.. = Protocol Version: 2
        .... .... 00.. .... = Discover Route: Suppress (0x0)
        .... ...0 .... .... = Multicast: False
        .... ..1. .... .... = Security: True
        .... .0.. .... .... = Source Route: False
        .... 0... .... .... = Destination: False
        ...0 .... .... .... = Extended Source: False
        ..0. .... .... .... = End Device Initiator: False
    Destination: 0x5105
    Source: 0x0000
    Radius: 30
    Sequence Number: 15
    [Extended Source: cc:86:ec:ff:fe:a9:a9:18 (cc:86:ec:ff:fe:a9:a9:18)]
    [Origin: 5]
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 187960
        Extended Source: cc:86:ec:ff:fe:a9:a9:18 (cc:86:ec:ff:fe:a9:a9:18)
        Key Sequence Number: 0
        Message Integrity Code: 9e519cd8
        [Key: KEY]
        [Key Label: ]
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
[... 78 lines cut ...]
        Key Sequence Number: 0
        Message Integrity Code: ab964413
        [Key: KEY]
        [Key Label: ]
ZigBee Application Support Layer Data, Dst Endpt: 2, Src Endpt: 1
    Frame Control Field: Data (0x00)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .0.. .... = Acknowledgement Request: False
        0... .... = Extended Header: False
    Destination Endpoint: 2
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 56
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 123
    Command: On (0x01)
```

Koenkk, 2022-11-29:
So it doesn't seem to use the TuYa api. I reverted https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c , 

Can you provide the single packet send when changing the switch type to halogen with the original TuYa gateway?

ster1um, 2022-11-29:
> So it doesn't seem to use the TuYa api. I reverted https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c ,
> 
> Can you provide the single packet send when changing the switch type to halogen with the original TuYa gateway?

This device **not support** the light type.

This device support this feature:
- Onoff
- Brightness
- Minimum brightness
- Maximum brightness
- Power on behaviour 
- Countdown

- type of switch (momentary toggle state)
Only this it's broken.

Koenkk, 2022-11-30:
Found the issue with the switch type, updated https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c

ster1um, 2022-11-30:
> Found the issue with the switch type, updated https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c
Ok work well...
Last question, is there a possibility to remember the selection of min and max brightness?
now after selection return to 0

Koenkk, 2022-11-30:
Yes, updated https://gist.github.com/Koenkk/0316c10873dcc48905c6e8f63fa4b09c

ster1um, 2022-11-30:
is it normal that selectors min and max brightness the scale is 1-255 and the brightness is 0-254
[url=https://postimg.cc/FfZ6Z0M5][img]https://i.postimg.cc/FfZ6Z0M5/Immagine.png[/img][/url]

Koenkk, 2022-11-30:
yes!

ster1um, 2022-11-30:
yes! all work well

Koenkk, 2022-12-01:
Great integrated, please do a final test without external converter.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-01 by Koenkk

ster1um, 2022-12-02:
I try the latest-dev, get an error to select the min or max brightness to 0. the scale it's wrong. please fix to 1-255.

` Publish 'set' 'min_brightness' to 'luce' failed: 'RangeError [ERR_OUT_OF_RANGE]: Write 0xa4c13830306c281a/1 genLevelCtrl({"64515":{"value":-4,"type":33}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received -4)'`

Koenkk, 2022-12-02:
Setting min_brightness  0 is not allowed, the minimum value is 1.

ster1um, 2022-12-02:
> Setting min_brightness 0 is not allowed, the minimum value is 1.

yes I understand but in your last converter the scale for min and max brightness is 1 to 255, now 0 to 255.

Koenkk, 2022-12-03:
Can you clarify where the range is now 0 to 255?

ster1um, 2022-12-05:
sorry, I force remove the device and repair it's all good!

gras55, 2023-01-02:
There is some issue with ts110e since 1.29 it no longer reports on and off position in front end

TKasperczyk, 2024-02-26:
> There is some issue with ts110e since 1.29 it no longer reports on and off position in front end

I can confirm. The value of the 0x0000 attribute (on_off) in the 0x0006 cluster (OnOff) is always false, even if the light is actually turned on.