Koenkk/zigbee2mqtt #15121: Add an EXPOSE instruction in the Dockerfile

redshift2, 2022-11-21:
### What happened?

with the kind of new zigbee2mqtt web interface the docker file is missing the expose instruction for the port
perhaps add
EXPOSE 8080 
to the docker file

### Zigbee2MQTT version

1.22.4

### Adapter firmware version

n/a

### Adapter

n/a

danieledwardgeorgehitchcock, 2022-11-21:
This could add to confusion as the port is configurable in configuration.yaml too - what happens if the two get mismatched?

Koenkk, 2022-11-21:
The port can indeed by customised, so not sure if this is a good thing to do.

redshift2, 2022-11-21:
I don't believe using the network_mode host is the correct way to solve this. Using net mode would be for something like a VOIP container (asterisk for example). It also kind of creates other issues with larger docker deployments.
However I understand your concern about the port mis-match. Perhaps in the in the configuration.yaml file it defaulted to 8080 with a comment stating to for docker installations use port 8080 or whatever. Its then possible to do a port redirect from the host to the container (-p 8102:8080 for a command line example). Remapping host ports to a container is a common docker thing

danieledwardgeorgehitchcock, 2022-11-21:
The trouble with the above approach, is what happens to all of the users that are using a non-standard port already? If it is locked down, there may be a lot of backlash and a peak of issue reporting

redshift2, 2022-11-21:
I'm not really sure how many people are using a non-standard port with this.  The default config file does not have a web port call out in it (it probably would be good to add it) , and the default instructions on the ZIGBEE2MQTT page has port 8080 listed. This is kind of a new feature, I can't believe that there will be much if any backlash or issue reporting and the system security improvements is worth it.

closed 2022-12-29 by github-actions[bot]