Koenkk/zigbee2mqtt #15122: Popp 701721 TRV, firmware offered via OTA "not officially released" by vendor, problems

ernstkl, 2022-11-21:
### What happened?

Ca. beginning of 2022-Nov there was a firmware update "01.18.0008 01.18" offered for OTA.
I did the update. 

Later the device showed problematic behavior: pi_heating_demand often gets stuck at 1%, even when setpoint is 20 deg below current temperature for hours, or when window_open_external is True. This did not change by re-pairing the device. 

When I contacted the Vendor (Popp) about this behavior they told me "this firmware is a Danfoss firmware and was never approved by us".

### What did you expect to happen?

Maybe a Danfoss firmware should not be offered for Popp devices? Except the devices are equal in every respect which I cannot judge.

### Zigbee2MQTT version

1.28.2 

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (ZBDongle-P)

rsporsche, 2022-11-22:
"pi_heating_demand often gets stuck at 1%, even when setpoint is 20 deg below current temperature for hours"

I have this same problem, I thought that the POPP was essentially identical to the Danfoss though so I can't see why it would behave differently. If POPP aren't supporting their devices it might be worth paying the little extra for Danfoss.

ernstkl, 2022-11-22:
Meanwhile I experienced the "stuck at 1%" problem also with the new device on which no OTA has been performed. So this problem is not related to the "01.18.0008 01.18" Danfoss firmware.

ernstkl, 2022-11-24:
> Meanwhile I experienced the "stuck at 1%" problem also with the new device on which no OTA has been performed. So this problem is not related to the "01.18.0008 01.18" Danfoss firmware.

So as the "stuck at 1%" issue is not related to the "wrong firmware installed" issue, it is off topic.
For that reason I am now going to edit the issue title and description to make it clean and to the point.

Continuing the "off topic" discussion shortly for the possible benefit of readers: I got a reply by Popp today reg. the 1%-issue. They tell me (summary): 
> "A valve that is OK closes already at about 10% actuator position. The remaining travel down to zero normally just results in compressing the rubber seal. Closing down to 1% is what the TRV does for a valve that is detected to be problematic." 

Apparently, in my case this was still not enough to stop the heating water, so I will have to fix that.
So "stuck at 1%" is not an error but by design.

gebauer, 2022-12-04:
Hi,

this is an eye-opener for me.

I have 5 POP devices, three of them I bought only recently. They are on the 01.12.0008 01.12 firmware from 20220929 (which I didn't notice). 
I now tried for several hours to update them to 01.18.0008 01.18, which seems to be an Danfoss firmware, which is older.
Luckily (?) the devices always aborted the update after 1-2h.

It is very confusing the Danfoss and POP have different firmwares with the same firmware stye - but with different meanings.

The 01.12 from Danfoss is pretty old and does not support externals temeperature sensors very good. The 0.12 from POP seems to very modern and exposes everything the 1.18 from Danfos does...
Somehow Zigbee2MQTT has to handle with this 🤷‍♂️

Best
Jan

ernstkl, 2022-12-19:
I do not have any firmware file.

wintersteiger, 2023-01-13:
I arrived here because I'm looking for updated firmware for a Popp valve (currently at 01.12.0008 01.12 dated 20220927, but not currently using zigbee2mqtt). I noticed that my valve reports a manufacturer ID of 1246, which is Danfoss' ID, i.e., Popp really should be using their own ID if they don't "approve" of Danfoss' firmware.

closed 2023-02-21 by github-actions[bot]

B0rax, 2023-04-02:
Was there any solution? Is there a way to update the Popp TRVs through Z2M or not?

k3a, 2023-04-02:
I updated my Pop using Z2M to the Denfoss 01.18.0008 01.18 firmware, and it appears to be working properly.
Initially, I noticed a strange behavior in which TRV would freeze.
I assumed it was because of the unofficial firmware, but it also happened to one TRV that I did not upgrade.

However, all TRVs have been working perfectly since then.

So upgrading appears to be safe, but there is no way to downgrade. I recommend not upgrading unless you have a specific problem that you know should be fixed in a new firmware.

ernstkl, 2023-04-02:
one reason to actually do the firmware upgrade is when you want to use the "external_measured_temperature" related features, see the [device page](https://www.zigbee2mqtt.io/devices/701721.html)