Koenkk/zigbee2mqtt #15128: [New device support]: unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_9yapgbuv'

antonio203, 2022-11-22:
### Link

https://fr.aliexpress.com/item/1005004838346776.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138c3c72004e1","nwkAddr":50735,"manufId":4417,"manufName":"_TZE200_9yapgbuv","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u000f+\u0019�\u000f+\u0007\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_9yapgbuv","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669122722508,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
I have a problem to control a simple Temperature & humidity sensor identified by Zigbee2MQTT as: model 'TS0601' and manufacturer name '_TZE200_9yapgbuv'

I tried several external_converter.js (copied from devices/tuya.js), but in all cases with negative results: the temperature and humidity are not displayed.

In ZHA project there is a similar issue with this device here (https://github.com/zigpy/zha-device-handlers/issues/1918) and the solution seems to use the configuration of device: ("_TZE200_bjawzodf", "TS0601")], which I think is the configuration that I am using in the external_converter.js.

Regards

### External converter

```shell
const definition = {
	fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_9yapgbuv'}],
	model: 'my_model',
	vendor: 'TuYa',
	description: 'Temperature & humidity sensor',
	fromZigbee: [fz.tuya_temperature_humidity_sensor], 
	toZigbee: [], 
	exposes: [e.temperature(), e.humidity(), e.battery()], 
};

module.exports = definition;
```

mikolaj-kow, 2022-12-02:
I can confirm, my WSD500A (_TZE200_9yapgbuv) pairs successfully with zigbee2mqtt 1.28.4 and produces following output:

```
{
    "last_seen": "2022-12-02T16:34:25+01:00",
    "linkquality": 78,
    "battery": null,
    "humidity": null,
    "temperature": null,
    "voltage": null
}
```

fredgazeaud, 2022-12-03:
Hi,
Same behavior for me. Pairing ok with modified tuya.js config. I do not get anythink better than following trace in debug mode:

`2022-11-29 12:44:31Received Zigbee message from '0xa4c1386baedc57cf', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
2022-11-29 12:44:32Received Zigbee message from '0xa4c1386baedc57cf', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
2022-11-29 12:45:45Received Zigbee message from '0xa4c1386baedc57cf', type 'attributeReport', cluster 'genBasic', data '{"65503":"{\u0000\u0000\u0000\u0019{\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011{\u0000\u0000\u0000\u0019{\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011F�\u0018+\u0019F�\u0018+\u0007\u0000\u0000\u0000\u0000\u0011{\u0000\u0000\u0000\u0019{\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
2022-11-29 12:45:51Received Zigbee message from '0xa4c1386baedc57cf', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
2022-11-29 12:46:10Received MQTT message on 'zigbee2mqtt/0xa4c1386baedc57cf/1/set' with data '{"read":{"attributes":["localTime"],"cluster":"genTime","options":{}}}'`

Koenkk, 2022-12-03:
`_TZE200_9yapgbuv` is supported now in z2m 1.28.4

closed 2022-12-03 by Koenkk

antonio203, 2022-12-03:
Yes, I can confirm that is working with version: 1.28.4-dev

antonio203, 2022-12-04:
Although initially it seemed that the device was charging, after a night of operation I see that no temperature or humidity value has been updated.

It only works for me using the 'TS0601_temperature_humidity_sensor' model as a base.

This is my code that I use in an external_converter:

const fzLocal = {
    my_temperature_humidity_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        options: [exposes.options.precision('temperature'), exposes.options.calibration('temperature'),
            exposes.options.precision('humidity'), exposes.options.calibration('humidity')],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.tthTemperature:
                    result.temperature = calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature');
                    break;
                case tuya.dataPoints.tthHumidity:
                    // result.humidity = calibrateAndPrecisionRoundOptions(value, options, 'humidity');
                    result.humidity = calibrateAndPrecisionRoundOptions(
						(value / (meta.device.manufacturerName === '_TZE200_bjawzodf' ? 10 : 1)),
						options, 'humidity');					
                    break;
                case tuya.dataPoints.tthBatteryLevel:
                    // result.battery = value;
					result.battery_level = {0: 'low', 1: 'middle', 2: 'high'}[value];
					result.battery_low = value === 0 ? true : false;
					result.battery = {0: 25, 1: 50, 2: 100}[value];                           // <- Improvemente: numeric battery level
                    break;
                case tuya.dataPoints.nousTempUnitConvert:
                    result.temperature_unit_convert = {0x00: 'celsius', 0x01: 'fahrenheit'}[value];
                    break;
                default:
                    meta.logger.warn(`zigbee-herdsman-converters:nous_lcd_temperature_humidity_sensor: NOT RECOGNIZED ` +
                        `DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

const my_definition = 
{
	fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_9yapgbuv'}],
	model: 'TS0601_temperature_humidity_sensor',
	vendor: 'TuYa',
	description: 'Temperature & humidity sensor',
    fromZigbee: [ fzLocal.my_temperature_humidity_sensor ],
    toZigbee: [], 
	exposes: (device, options) => 
	{
       const exps = [e.temperature(), e.humidity(), e.battery()];
       if (!device ||
     	   device.manufacturerName === '_TZE200_qyflbnbj' || 
		   device.manufacturerName === '_TZE200_9yapgbuv') 
	   {
          exps.push(e.battery_low());
          exps.push(exposes.enum('battery_level', ea.STATE, ['low', 'middle', 'high']).withDescription('Battery level state'));
       }
	   
	   exps.push(e.linkquality());
	   return exps;
    },
};

Koenkk, 2022-12-04:
Can you update to the latest dev and try the following converter? https://gist.github.com/Koenkk/af1f538fddb6ab7f26931ef6eb9a6070 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

fredgazeaud, 2022-12-08:
Hi,
Tried https://gist.github.com/Koenkk/af1f538fddb6ab7f26931ef6eb9a6070 yesterday but same result. I had to remove `tuya.exposes.batteryState()` and `[3, 'battery_state', tuya.valueConverter.batteryState]` which prevented z2m to start (do I missed another change?). No temp neither humidity or battery info decoded.
```
info 2022-12-08 11:11:28Successfully interviewed '0xa4c1386baedc57cf', device has successfully been paired
info 2022-12-08 11:11:28Device '0xa4c1386baedc57cf' is supported, identified as: TuYa Temperature & humidity sensor (TS0601_temperature_humidity_sensor)
info 2022-12-08 11:11:28MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_temperature_humidity_sensor","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xa4c1386baedc57cf","ieee_address":"0xa4c1386baedc57cf","status":"successful","supported":true},"type":"device_interview"}'
debug 2022-12-08 11:12:04Received Zigbee message from '0xa4c1386baedc57cf', type 'attributeReport', cluster 'genBasic', data '{"65503":"��$+\u0019��$+\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
info 2022-12-08 11:12:04MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:01+01:00","linkquality":134,"temperature":null}'
info 2022-12-08 11:12:54MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:51+01:00","linkquality":134,"temperature":null}'
debug 2022-12-08 11:12:54Device '0xa4c1386baedc57cf' announced itself
info 2022-12-08 11:12:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1386baedc57cf","ieee_address":"0xa4c1386baedc57cf"},"type":"device_announce"}'
info 2022-12-08 11:12:55MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:52+01:00","linkquality":134,"temperature":null}'
debug 2022-12-08 11:12:56Retrieving state of '0xa4c1386baedc57cf' after reconnect
debug 2022-12-08 11:12:58Received Zigbee message from '0xa4c1386baedc57cf', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
info 2022-12-08 11:12:58MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:55+01:00","linkquality":156,"temperature":null}'
debug 2022-12-08 11:12:59Received Zigbee message from '0xa4c1386baedc57cf', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
info 2022-12-08 11:12:59MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:56+01:00","linkquality":153,"temperature":null}'
info 2022-12-08 11:13:00MQTT publish: topic 'zigbee2mqtt/0xa4c1386baedc57cf', payload '{"battery":null,"humidity":null,"last_seen":"2022-12-08T11:12:57+01:00","linkquality":153,"temperature":null}'
```
Regards

antonio203, 2022-12-08:
Hello everyone,

> Can you update to the latest dev and try the following converter? https://gist.github.com/Koenkk/af1f538fddb6ab7f26931ef6eb9a6070 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hello everyone,

A few days ago I tried the suggested source code along with the latest version from the repository, but the zigbee2mqtt docker didn't start and restarted.

Today I tried again, and the docker starts, but the temperature and humidity values are not updated.

For me, the only code that works is the one I mentioned above, which was based on the TS0601_temperature_humidity_sensor' model

Regards

antonio203, 2022-12-08:
I have just seen where the problem was:

    fromZigbee: [tuya.fz.datapoints],                                     // Missed the call to tuya.fz.datapoints
    toZigbee: [tuya.tz.datapoints],                                       // Missed the call to tuya.fz.datapoints

And also, I have seen that the device only sends the datapoints: 1 (temperature), 2 (humidity), 3 (battery_state) and 9 (temperature_units). I have tried the following code to convert battery_state to battery (level) but without success, could you tell me how to do it?

        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [3, 'battery_state', tuya.valueConverter.batteryState],
            // [4, 'battery', tuya.valueConverter.raw],
            [3, 'battery', {0: 25, 1: 50, 2: 100}[tuya.valueConverter.raw] ],    // this does not work
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnit],			

Regards

Koenkk, 2022-12-08:
I updated https://gist.github.com/Koenkk/af1f538fddb6ab7f26931ef6eb9a6070

You cannot convert a datapoint (3) twice. This also doesn't make sense since it is duplicate information. I suggest to expose just battery and map battery state 0 to 0%.

antonio203, 2022-12-08:
Hello again,

I have just tested the new proposal, and I have 2 new questions:

1) With this code the device lost the calibration capability of the parameters: temperature_calibration and humidity_calibration, and in this case the error is around the 1 degree in regards to a Xiaomi device.

2) This approach seems to work, but I removed one of the temperature devices, and when try to reconnect it again, it seems that the code enters in a device loop with the following sequence, and never reads the values of temperature and humdity from this device:

	Zigbee2MQTT:debug 2022-12-08 20:31:25: Retrieving state of 'T5Z' after reconnect
	Zigbee2MQTT:debug 2022-12-08 20:31:28: Received Zigbee message from 'T5Z', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
	Zigbee2MQTT:debug 2022-12-08 20:31:28: Received Zigbee message from 'T5Z', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
	Zigbee2MQTT:debug 2022-12-08 20:32:42: Received Zigbee message from 'E1Z', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":77}' from endpoint 1 with groupID 0
	Zigbee2MQTT:info  2022-12-08 20:33:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"T5Z","ieee_address":"0xa4c13875f084e13e"},"type":"device_announce"}'
	Zigbee2MQTT:debug 2022-12-08 20:33:27: Device 'T5Z' announced itself
	Zigbee2MQTT:debug 2022-12-08 20:33:29: Retrieving state of 'T5Z' after reconnect
	Zigbee2MQTT:debug 2022-12-08 20:33:32: Received Zigbee message from 'T5Z', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":72}' from endpoint 1 with groupID 0
	Zigbee2MQTT:debug 2022-12-08 20:33:31: Received Zigbee message from 'T5Z', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
	Zigbee2MQTT:debug 2022-12-08 20:31:23: Device 'T5Z' announced itself
	Zigbee2MQTT:info  2022-12-08 20:31:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"T5Z","ieee_address":"0xa4c13875f084e13e"},"type":"device_announce"}'
	Zigbee2MQTT:debug 2022-12-08 20:31:03: Received Zigbee message from 'T5Z', type 'attributeReport', cluster 'genBasic', data '{"65503":"�\u0005%+\u0019�\u0005%+\u0007\u0000\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0

Koenkk, 2022-12-09:
> With this code the device lost the calibration capability of the parameters: temperature_calibration and humidity_calibration, and in this case the error is around 1 degree in regards to a Xiaomi device.

Added this in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5113 . Upgrade to the latest dev branch to test this.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

> This approach seems to work, but I removed one of the temperature devices, and when try to reconnect it again, it seems that the code enters in a device loop with the following sequence, and never reads the values of temperature and humdity from this device:

This is because TuYa devices use a custom API (and don't allow reading of values).

rolsch, 2023-02-13:
Hmm,
latest https://github.com/ioBroker/ioBroker.zigbee/commit/19d8438fc68c3166e73a0d25e368e20594257cee
comes with     "zigbee-herdsman": "0.14.91" and "zigbee-herdsman": "0.14.93" but don't read the temp and hum values.

Koenkk, 2023-03-22:
@ClementMarrast try to:

- force remove the device through z2m frontend
- restart z2m
- re-pair device