Koenkk/zigbee2mqtt #15134: [Feature request]: Expose externally measured temperature instead of internally measured temperature for Danfoss Ally TRV

rsporsche, 2022-11-22:
### Is your feature request related to a problem? Please describe

I use ZigBee2MQTT to integrate my ZigBee devices, including a set of Danfoss Ally TRVs, with Home Assistant.

Overall it's great but the temperature reported by the associated climate entities in Home Assistant is the 'internal' temperature measured by the TRV (local_temperature) even though I am using external temperature sensors by setting external_measured_room_sensor.

It would be preferable for the climate entity to report the actual temperature that's being regulated.

### Describe the solution you'd like

If external_measured_room_sensor is set then report that instead of local_temperature.

### Describe alternatives you've considered

I thought this could be solved in Home Assistant by creating a 'template' climate entity except that the functionality doesn't seem to exist in Home Assistant and I think it's the wrong way to solve the problem.

### Additional context

Nothing to add

rsporsche, 2022-11-22:
Reading the docs.. https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html#advanced-configuration

closed 2022-11-22 by rsporsche

rubenkr, 2023-01-10:
Could you please share your config as the link above with the #advanced configuration does in my view has no option listed to replace temperature of the TRV temperature data with external sensor.