Koenkk/zigbee2mqtt #15140: Moes double light switch not recognized correctly

bogdanm, 2022-11-23:
### What happened?

My newly purchased Moes Zigbee dual light switch (https://www.aliexpress.com/item/1005002061439926.html?spm=a2g0o.order_list.0.0.21ef1802SIuMuh) is recognized as [TS0012](https://www.zigbee2mqtt.io/devices/TS0012.html#tuya-ts0012), which prevents me from using some of the more advanced features of the switch (control of the switch's background light and power on behavior).

[image]

### What did you expect to happen?

I expected the switch to be recognized as [this model](https://www.zigbee2mqtt.io/devices/ZTS-EU_2gang.html). The single switch version (bought from the same Aliexpress seller as above) is recognized as [ZS-EUB_1gang](https://www.zigbee2mqtt.io/devices/ZS-EUB_1gang.html), which is still not correct (since my switch is a touch, not mechanical), but at least I can control power on behavior and the switch's background light using that model.

### How to reproduce it (minimal and precise)

- Install switch
- Put it in pairing mode (long press the left touch)
- Allow devices to join in Z2M
- Observe model ID

### Zigbee2MQTT version

1.28.2 commit: 360a777

### Adapter firmware version

 20200925

### Adapter

Electrolama zig-a-zig-ah!

danieledwardgeorgehitchcock, 2022-11-23:
Could you post up the database.db entry for this device so that I can check a few things?

bogdanm, 2022-11-23:
@danieledwardgeorgehitchcock :

```
{"id":7,"type":"EndDevice","ieeeAddr":"0xa4c13863d0536a19","nwkAddr":50434,"manufId":4417,"manufName":"_TZ3000_18ejxno0","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0010+\u0011","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_18ejxno0","zclVersion":3,"appVersion":80,"modelId":"TS0012","powerSource":3}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002160064d","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002160064d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1957453523},"lastSeen":1669209977385,"defaultSendRequestWhen":"immediate"
```

danieledwardgeorgehitchcock, 2022-11-23:
Could you try the below external converter (you'll have to remove the device and restart Z2M first):

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_g1ib5ldv'},
        {modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZTS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
            .withDescription('Indicator light status'),
        exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
            .withDescription('Controls the behavior when the device is powered on')],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch, fz.moes_switch],
    toZigbee: [tz.tuya_switch_state, tz.moes_switch],
    onEvent: tuya.onEventSetLocalTime,
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```

bogdanm, 2022-11-23:
With the converter in place, Z2M recognizes the model [ZTS-EU_2gang](https://www.zigbee2mqtt.io/devices/ZTS-EU_2gang.html#moes-zts-eu_2gang), but nothing works when I try to control the switch from the UI (not even on/off switching). The logs show this error:

```
zigbee2mqtt    | Zigbee2MQTT:info  2022-11-23 19:55:44: Configuring '0xa4c13863d0536a19'
zigbee2mqtt    | Zigbee2MQTT:info  2022-11-23 19:55:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Wall touch light switch (2 gang)","friendly_name":"0xa4c13863d0536a19","model":"ZTS-EU_2gang","supported":true,"vendor":"Moes"},"type":"pairing"}'
zigbee2mqtt    | Zigbee2MQTT:error 2022-11-23 19:55:54: Failed to configure '0xa4c13863d0536a19', attempt 1 (Error: Bind 0xa4c13863d0536a19/2 genOnOff from '0x00124b002160064d/1' failed (AREQ - ZDO - bindRsp after 10000ms)
zigbee2mqtt    |     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
zigbee2mqtt    |     at listOnTimeout (node:internal/timers:559:17)
zigbee2mqtt    |     at processTimers (node:internal/timers:502:7))
```

I tried to remove the switch and added it again, same result.

danieledwardgeorgehitchcock, 2022-11-24:
Try this:

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type, tz.tuya_backlight_mode]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type, fz.tuya_backlight_mode]),
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.power_on_behavior().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.power_on_behavior().withEndpoint('l2'),
 //       e.switch_type_2(),
        exposes.enum('backlight_mode', ea.ALL, ['LOW', 'MEDIUM', 'HIGH'])
            .withDescription('Indicator light status: LOW: Off | MEDIUM: On| HIGH: Inverted'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;
```

bogdanm, 2022-11-24:
@danieledwardgeorgehitchcock Thank you (sorry, forgot to say that earlier :) ). The new converter returns a JS error:

```
zigbee2mqtt    |     toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type, tz.tuya_backlight_mode]),
zigbee2mqtt    |               ^
zigbee2mqtt    |
zigbee2mqtt    | ReferenceError: extend is not defined
zigbee2mqtt    |     at /app/dist/util/externally-loaded.js:13:15
zigbee2mqtt    |     at Script.runInContext (node:vm:141:12)
zigbee2mqtt    |     at Script.runInNewContext (node:vm:146:17)
zigbee2mqtt    |     at Object.runInNewContext (node:vm:306:38)
zigbee2mqtt    |     at loadModuleFromText (/app/lib/util/utils.ts:148:8)
zigbee2mqtt    |     at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
zigbee2mqtt    |     at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
zigbee2mqtt    |     at getExternalConvertersDefinitions.next (<anonymous>)
zigbee2mqtt    |     at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
zigbee2mqtt    |     at new Controller (/app/lib/controller.ts:84:58)
```

Sorry, I am somewhat familiar with JS, but not with this source tree.

danieledwardgeorgehitchcock, 2022-11-24:
Just tweaked the code in the above converter. Could you give that a whirl?

bogdanm, 2022-11-24:
@danieledwardgeorgehitchcock Better this time. The interface looks like this now:

[image]

(previously I had a single setting for `power_on_behavior`, which was probably correct).

What works:
- Switching on/off the switches manually updates the state in the UI correctly.
- Although not correctly labeled, `backlight_mode` seems to work fine. I say "not correctly labeled" because I think that it should look more like `indicate_light` in https://www.zigbee2mqtt.io/devices/ZTS-EU_2gang.html#moes-zts-eu_2gang. Right now, its real behavior is switching between "no backlight at all", "backlight on only when lights are on" and "backlight on only when lights are off".

What doesn't work:
- if I try to set `l1_state` to `on`, the UI also switches `l2_state` to on immediately, and this is reflected in the light state. The same thing happens if I switch `l1_state` to `off` or if I switch `l2_state` to `on` or `off`. Effectively, even if the UI shows two switches, they behave as a single switch when I control them from the UI (but not when updating the remote switch state, see above).
- I don't think `power_on_behavior` works properly yet (I need to do more testing though).

danieledwardgeorgehitchcock, 2022-11-25:
Ok,

I feel that the above is good progress - I'm coding this blind using some existing converters as templates so, there may be a few things that need some fettling :-)

Try this one:

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type, tz.tuya_backlight_mode]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type, fz.tuya_backlight_mode]),
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.power_on_behavior().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.power_on_behavior().withEndpoint('l2'),
        exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
                .withDescription('Indicator light status'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await device.getEndpoint(1).read('genBasic',['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
```

This should (hopefully) fix the endpoint switching issue and change the backlight / indicator expose section - we may still need to work on the correct power on behaviour

bogdanm, 2022-11-25:
> I feel that the above is good progress - I'm coding this blind using some existing converters as templates so, there may be a few things that need some fettling :-)

You, sir, are awesome :) The interface now:

[image]

What works:

- Switch states work perfectly now, individually and in both directions.

What doesn't work:

- Compared to the last converter, `indicate_light` doesn't seem to do anything anymore. I'm not getting any errors when I press on any of the four buttons, but I'm also not seeing any change in the switch's backlight. 
- I don't think `switch_type` belongs here at all, since this is a touch switch and thus it's always momentary by nature.
- `power_on_behavior` is still weird. First, when I click on its "refresh state" icon, I get an error that says "No converter available for 'power_on_behavior' ()". I'm apparently able to set it (at least I don't get any error when I try to set it), but when I cycle the power the switch goes berserk and automatically enters pairing mode again. This might be a switch issue, but I don't think so, since my other (single) Moes switch that I mentioned in my first post (purchased from the same AliExpress seller at roughly the same time) doesn't do this. 

Can I do anything more to help with this? While I am not familiar with the project, I am a programmer, so I might be able to be a little more useful here :)

danieledwardgeorgehitchcock, 2022-11-25:
Ok,

It's good that the states are updating correctly so I'd still say we are moving in the right direction!

I stripped a lot of the converters out as I saw that they were part of the `moes_switch` converter.

The switch type may be important as the device I'd guess has relays, and this would control the behaviour of those?

Try this tweaked converter - if neither the power_on_behaviour nor the indicate_light settings work, then by power of testing and elimination, we can rule out the moes_switch converter and go back to some more generic converters:

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    toZigbee: extend.switch().toZigbee.concat([tz.tuya_switch_type, tz.moes_switch, tz.tuya_switch_state]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.tuya_switch_type, fz.moes_switch]),
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch_type_2(),
        exposes.enum('power_on_behaviour', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
                .withDescription('Power on behaviour'),
        exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
                .withDescription('Indicator light status'),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await device.getEndpoint(1).read('genBasic',['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
```

As for helping, just keep testing and follow along :-) hopefully, we will get there! I think we are close...

bogdanm, 2022-11-26:
@danieledwardgeorgehitchcock This version isn't much different from the previous one: the only thing that works is the swich state (in both directions). Setting `indicate_light` doesn't result in an error, but it doesn't do anything either (just like the last time). This time, when I try to set `power_on_behavior`, I get an error (`No converter available for 'power_on_behaviour' ("off")`, `No converter available for 'power_on_behaviour' ("on")` or `No converter available for 'power_on_behaviour' ("previous")`).
As for `switch_type`, I always assumed that this refers to the actual switch type, since mechanical switches can be momentary or they can toggle between on/off positions. I didn't consider think that this is actually related to the switch's relays, although I guess this is possible.

danieledwardgeorgehitchcock, 2022-11-27:
Yes, I knew that this wouldn't change much but just wanted to try a few things first. Try this one:

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    toZigbee: [tz.tuya_switch_type, tz.moes_switch, tz.tuya_switch_state],
    fromZigbee: [fz.tuya_switch_type, fz.moes_switch, fz.tuya_switch],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        e.switch().setAccess('state', ea.STATE_SET).withEndpoint('l1'),
        e.switch().setAccess('state', ea.STATE_SET).withEndpoint('l2'),
        e.switch_type_2(),
        exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
            .withDescription('Indicator light status'),
        exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
            .withDescription('Controls the behavior when the device is powered on')],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await device.getEndpoint(1).read('genBasic',['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
```

bogdanm, 2022-11-27:
In this version, `power_on_behavior` is the only thing that works :) Everything else is broken, including the states of the switces in any direction.

danieledwardgeorgehitchcock, 2022-11-28:
Ok,

Based on the above, this should hopefully give you everything in a working state:

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    moes_indicate_light: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('tuyaBacklightMode')) {
                const value = msg.data['tuyaBacklightMode'];
                const backlightLookup = {0: 'Off', 1: 'Switch', 2: 'Position', 3: 'Freeze'};
                return {indicate_light: backlightLookup[value]};
            }
        },
    }
};

const tzLocal = {
    moes_indicate_light: {
        key: ['indicate_light'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'Off': 0, 'Switch': 1, 'Position': 2, 'Freeze': 3};
            utils.validateValue(value, Object.keys(lookup));
            await entity.write('genOnOff', {tuyaBacklightMode: lookup[value]});
            return {state: {indicate_light: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genOnOff', ['tuyaBacklightMode']);
        },
    }
};

const definition = {
    fingerprint: [{modelID: 'TS0012', manufacturerName: '_TZ3000_18ejxno0'}],
    model: 'ZS-EU_2gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (2 gang)',
    toZigbee: extend.switch().toZigbee.concat([tz.tuya_switch_type, tz.moes_switch, tzLocal.moes_indicate_light]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.tuya_switch_type, fz.moes_switch, fzLocal.moes_indicate_light]),
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 2};
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch_type_2(),
        exposes.enum('indicate_light', ea.ALL, ['Off', 'Switch', 'Position', 'Freeze'])
            .withDescription('Controls the behaviour of the indicator light(s)'),
        exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
            .withDescription('Controls the behavior when the device is powered on')],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await device.getEndpoint(1).read('genBasic',['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
```

bogdanm, 2022-11-29:
I was wrong before. I thought that your previous converter made `power_on_behavior` work correctly, but it doesn't. It doesn't appear to change the mode at all, the power on behavior seems to be stuck on `previous` (I say `seems to` because I can't read its state). So it doesn't work in the latest converter either.

Other than that:

- Switches work in the new converter
- `indicate_light` still doesn't do anything and I can't read its state (`No converter available for 'get' 'indicate_light' ()`)
- `switch_type§ doesn't do anything, but it never did anything for this switch anyway, so I think we can leave this one out.

yomgui1, 2022-11-29:
@bogdanm  I've similar issues with similar products labelled from MOES, also orderer on AliExpress but on another "seller" : https://fr.aliexpress.com/item/1005002130544613.html

I've a couple of ZTS switches: 1-gang and 2-gang, and these last doesn't have the possiblity to choose power-off and led status. The 1-gang versions have the possiblity but they are not seen from the same manufacturer:

Long story: I've bought the same product (1-gang ZTS-EU) twice, few months of interval, both have exactly the same packaging, but...

* first one has no added white label on back of the package and is correctly recognized by Z2M as a MOES device (TS0601)  (that can work with or without neutral line)

`{"id":3,"type":"EndDevice","ieeeAddr":"0x0c4314fffe7ce79f","nwkAddr":40157,"manufId":4098,"manufName":"_TZE200_amp6tsvy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_amp6tsvy","stackVersion":0,"dateCode":"","zclVersion":3}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc22ab","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1482240604},"lastSeen":1669733879434,"defaultSendRequestWhen":"immediate"}`

* the second one has a white label added, with an "incorrect" name "ZS-EU1-BK-MS", but the product inside has the correct labelling on its back! This one is recognized by Z2M as Tuya device (TS0011) and is also work with or without neutral line

`{"id":25,"type":"EndDevice","ieeeAddr":"0xa4c138cc5f64c7f7","nwkAddr":54945,"manufId":4417,"manufName":"_TZ3000_hhiodade","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TZ3000_hhiodade","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificTuya_3":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc22ab","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-999208499},"lastSeen":1669733673235,"defaultSendRequestWhen":"immediate"}`

Both are plugged without neutral line and works nicely except for the Tuya one exposing a "switch_type" non-functional (I've got an error message if I change the state).

The main difference comes from their PCB:

* MOES 1-gang:
[image]

* TUYA 1-gang:
[image]

Exterior and back are almost identical on both (the wifi symbol is placed on same place).

Note: an example of db entry for the 2-gang Tuya version:

`{"id":24,"type":"EndDevice","ieeeAddr":"0xa4c138886a30782c","nwkAddr":31211,"manufId":4417,"manufName":"_TZ3000_18ejxno0","powerSource":"Mains (single phase)","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0011","65506":55,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_18ejxno0","zclVersion":3,"appVersion":80,"modelId":"TS0012","powerSource":3}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc22ab","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc22ab","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1957453523},"lastSeen":1669735188551,"defaultSendRequestWhen":"immediate"}`

bogdanm, 2022-11-30:
@yomgui1 Thanks a lot, that is very useful. As soon as I find some time I'll open one of them and see what I can find inside. It'd be a real shame if they didn't actually support power_on_behavior, that is a very useful feature for power outages.

bogdanm, 2022-11-30:
@danieledwardgeorgehitchcock 

> It may be that this device is using a different set of TuYa Data Points than the Moes converter that I am using above. This would explain why you aren't seeing any change when applying the indicate light and power on behaviour. If that is the case, these Data Points would be difficult to track down without linking the device to a TuYa Zigbee Gateway and doing some API probing.

Can you recommend a TuYa gateway suitable for API probing? (sorry if this a dumb question, I don't know much about TuYa at this point, but I'm willing to learn).

bogdanm, 2022-12-04:
@yomgui1 So it seems like I have the same version as you:

[image]

closed 2023-01-11 by github-actions[bot]