Koenkk/zigbee2mqtt #15144: Thermostat Using/Showing Temperature of Wrong Sensor

norsemanGrey, 2022-11-23:
### What happened?

I have paired a [Namron thermostat](https://www.zigbee2mqtt.io/devices/4512737_4512738.html#namron-4512737%252F4512738) with Z2M using the ZBDongle-E and integrated in HA.

The climate sensor is not showing temperature value of the sensor being used in the temperature control. There is an option to select between air temperature sensor (local_temperature) and floor temperature sensor (outdoor_temperature) for heating control. When the floor temperature sensor is selected the device is regulating by that temperature, however, the actual temperature shown on the Climate sensor in HA is still the air temperature.

[image]
[image]
[image]

[image]

### What did you expect to happen?

The value of the temperature sensor selected for temperature regulation should be the one used/shown in the HA climate sensor.

### How to reproduce it (minimal and precise)

Pair Namron (4512737/4512738) thermostat with coordinator and find in HA. Select floor sensor from HA/Z2M or on device. See that temperature shown on the Climate sensor in HA is the value of the air temperature sensor.

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (ZBDongle-E / EFR32MG21)

### Debug log

NA

danieledwardgeorgehitchcock, 2022-11-23:
You may need to create an MQTT Climate entity in Home Assistant for this rather than using the one derived from Zigbee2MQTT:

[Home Assistant - MQTT Climate](https://www.home-assistant.io/integrations/climate.mqtt/)

I think the default behaviour of Home Assistant is to take the `local_temperature` sensor entity for an automatically created climate entity..

atus, 2022-12-10:
+1
As a side effect, when looking at history this is what you get. My thermostat is outside the actual room.
[image]

I'm new to this space, but tried reading some docs. Any chance we could write an extension or converter to fix this temporarily?

atus, 2022-12-10:
Related issue on HA https://github.com/home-assistant/core/issues/78797

brainix112, 2022-12-30:
Hey I have just the same issue! 
Just got around to hook up the thermostat, changed from ZHA to z2m, and finally got everything over and noticed this.
With the thermostat on the outside of the room this is gonna create some issues..
As you can see, the thermostat entity is using the local_temperature
[image]
[image]

Should we open an issue at [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) as well about this?

Leissson, 2023-01-30:
Keep alive, please. Need this one too...

Leissson, 2023-01-30:
Think I finally find clean workaround for this.

In **z2m** `configuration.yaml` add `current_temperature_template` line under climate device:

```
devices:
  '0xxxxxxxxxxxxxxxxxx':
    friendly_name: Bathroom Thermostat
    homeassistant:
      name: Bathroom Thermostat
      current_temperature_template: '{{ value_json.outdoor_temperature }}'
    legacy: false
    optimistic: true
    thermostat_unit: celsius 
```

@norsemanGrey @brainix112 @atus Let me know what you think.

pittbull, 2023-02-21:
@Leissson Would you perhaps know how I can workaround this on the zwave version?

Leissson, 2023-02-22:
@pittbull Unofrtunately I'm not familiar with zwave. But you should somehow populate `current_temperature` with `outdoor_temperature`. It should be done before sending it to HA, so it thinks that is the correct value for using as `climate` temp.

Is @johanko in the right path?

https://github.com/zwave-js/zwave-js-ui/issues/661#issuecomment-782939932

pittbull, 2023-02-22:
Thanks for providing me with the insight to this other issue - it put me on a path to a solution.

I was unable to make customDevices.json work in websock mode, but when switching to mqtt and manipulating the discovery template for this specific device it actually worked.

Thank you so much (has bugged me for months)

Leissson, 2023-02-22:
Nice, I know the feeling!

closed 2023-04-05 by github-actions[bot]

atus, 2023-05-02:
Works like a charm. I consider this a fix so the issue might even be marked as fixed.
You won't be switching between floor and room sensor ever/often so it's a good enough fix.

Added this to documentation so it doesn't get lost.

joggs, 2024-11-30:
> homeassistant:
>       name: Bathroom Thermostat
>       current_temperature_template: '{{ value_json.outdoor_temperature }}'

Great job!
Had the same problem with Sunricher SR-ZG9092A

owangen, 2025-01-20:
> Anybody experiencing the same as me where the outdoor temperature is not longer exposed?

Yes. I¨m experiencing the same. Outdoor temperatur is still exposed, but the template "hack" does not work.

joggs, 2025-01-20:
Instead of outdoor temperature being a parameter, I see that is now is its own entity in home assistant

owangen, 2025-01-20:
> Instead of outdoor temperature being a parameter, I see that is now is its own entity in home assistant

Not sure what you mean. I have always had outdoor temperatur as an own attribute.

Leissson, 2025-01-21:
> Instead of outdoor temperature being a parameter, I see that is now is its own entity in home assistant

Entity for outdoor_temperature has been there all the time. With templating you force HA to use that value for heat control. 

I don't understand the issue either without details.

joggs, 2025-01-21:
> > Instead of outdoor temperature being a parameter, I see that is now is its own entity in home assistant
> 
> Not sure what you mean. I have always had outdoor temperatur as an own attribute.

I suppose you in the same way i did, used the attribute outdoor_temperature for the entity , but that attribute has disappeared for me lately but I then found out that there is a separate sensor for outdoor_temperature(e.g sensor.badrum_outdoor_teperature etc) so I now use that sensor instead of the attribute. Maybe it has been there all the time as Leisson says, but I did not know about it, so search for outdoor_temperature in developer tools -> states, and hopefully you find the sensor

owangen, 2025-03-30:
> I suppose you in the same way i did, used the attribute outdoor_temperature for the entity , but that attribute has disappeared for me lately but I then found out that there is a separate sensor for [...]

I don't quite understand. Have you got it working with changing the current_temperature_template? If so, could you share how your configuration looks?