Koenkk/zigbee2mqtt #15145: backlight_mode bug

StopDragon, 2022-11-23:
### What happened?

On some models the backlight_mode can be seen in the settings, 
on some models I have checked the backlight_mode cannot be set.

available: TS0013
not available: TS0011, TS0012

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

20220219

### Adapter

SONOFF ZB USB (version p)

danieledwardgeorgehitchcock, 2022-11-23:
This isn't a bug.

You are correct, support for the backlight settings have not been implemented in the device converter for the TS0011 and TS0012 devices but has been in the TS0013 devices.

Are you sure that the backlight can indeed be controlled on the other devices via the TuYa app? If so, it is more of a feature request and we would need to confirm the Data Point for that setting from TuYa

StopDragon, 2022-11-23:
Yes.
Then may I make a request?

danieledwardgeorgehitchcock, 2022-11-24:
Could you find the data point for the backlight setting?

StopDragon, 2022-11-24:
> Could you find the data points used by the different devices for the backlight setting?

I'm sorry but I didn't understand what you said. where can i see that?

danieledwardgeorgehitchcock, 2022-11-24:
[Find TuYa Data Points](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)

ajunio3, 2022-12-22:
backlight DP is 16

ajunio3, 2022-12-22:
really need this for tuya TS0013 3gang with neutral

ajunio3, 2022-12-22:
and yes i can control backlight and last_state in Smart life for this device

closed 2023-01-29 by github-actions[bot]