Koenkk/zigbee2mqtt #15148: Paired Schneider Wiser sockets can't be controlled

tkarlsson, 2022-11-23:
### What happened?

Pairing Schneider Wiser sockets works but are identified as ELKO equivalent.
1 socket (WDE002172) (https://github.com/Koenkk/zigbee2mqtt/issues/15146)
2 socket  (WDE002182) (https://github.com/Koenkk/zigbee2mqtt/issues/15147)

However, after pairing you can't control the sockets. The state object isn't built up complete initially. By adding a custom converter (copying the exact code from latest herdsman converters which should be used in this version) and restarting it will build up a complete state object (not sure if this is the issue):

**Logs:**
```
Zigbee2MQTT:debug 2022-11-23 11:51:58: Publishing 'set' 'power_on_behavior' to '0x50325ffffe4c2294'
Zigbee2MQTT:error 2022-11-23 11:52:00: Publish 'get' 'power_on_behavior' to '0x50325ffffe4c2294' failed: 'Error: Read 0x50325ffffe4c2294/242 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:debug 2022-11-23 11:52:00: Error: Read 0x50325ffffe4c2294/242 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:658:23)
    at runMicrotasks (<anonymous>)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at processTimers (node:internal/timers:499:9)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28)
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:109:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
```

**Custom converter:**
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['SOCKET/OUTLET/2'],
        model: 'WDE002182',
        vendor: 'Schneider Electric',
        description: 'Zigbee smart socket with power meter',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.EKO09738_metering, fz.power_on_behavior],
        toZigbee: [tz.on_off, tz.power_on_behavior],
        exposes: [e.switch(), e.power(), e.energy(), exposes.enum('power_on_behavior', ea.ALL, ['off', 'previous', 'on'])
            .withDescription('Controls the behaviour when the device is powered on'), e.current(), e.voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(6);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);
            // Unit supports acVoltage and acCurrent, but only acCurrent divisor/multiplier can be read
            await endpoint.read('haElectricalMeasurement', ['acCurrentDivisor', 'acCurrentMultiplier']);
            await reporting.readMeteringMultiplierDivisor(endpoint);
        },
};

module.exports = definition;
```

Once the restart has taken place and the object is built up it can be controlled and everything is working as expected even if I remove the custom converter.

This seems to be related to:
https://github.com/Koenkk/zigbee2mqtt/issues/14589
https://github.com/Koenkk/zigbee2mqtt/issues/15146
https://github.com/Koenkk/zigbee2mqtt/issues/15147

I have tried two different controllers with the same experience.

### How to reproduce it (minimal and precise)

Vanilla installation, discover device and then try to change state of one of the attributes

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

tkarlsson, 2022-11-23:
Also seems related to:
https://github.com/Koenkk/zigbee2mqtt/issues/12635

roysve, 2022-12-18:
Just want to say Schneider is the owner of Elko, so many of there products are same.

tkarlsson, 2022-12-18:
Yes. Not sure if there is a way to distinguish between the two brandings.

tkarlsson, 2022-12-18:
Also, as noted above. Restarting and refreshing the states of the exposed settings will push things to a state where it works perfectly. Have tried it on 10 different devices with the same result.

rindlerblabla, 2023-01-26:
Am experiencing the same. After a restart the power state is updated in z2m, but it is not possible to control from z2m. Haven't set up a custom converter.

tkarlsson, 2023-01-26:
As a note, the way I managed to get this sort of working without a custom inverter is to basically go into the exposes interface and refresh the individual attributes. After that it is working.

dennisgranasen, 2023-01-26:
> As a note, the way I managed to get this sort of working without a custom inverter is to basically go into the exposes interface and refresh the individual attributes. After that it is working.

That does not work for me. I've gotten it to work on all my devices, the only consistent way I've found is:
1. add it to z2m (works with or without the custom converter, it makes no difference aside from the model name and image). 
2. after pairing is complete, temporarily cut power to the outlet (e.g. via the fuse)
3. after restoring power the device works as expected.

I followed this method for my last four outlets that I added to my network over the last couple of weeks. They all work flawlessly after that, despite that I removed the custom converter.

rindlerblabla, 2023-01-26:
Power cycle doesn't change anything for me. Still can't control power on behavior and when trying to turn it on/off I get get the response:

```
 2023-01-26 21:16:17MQTT publish: topic 'zigbee2mqtt/Tvättmaskinen', payload '{"current":0,"energy":2.56,"last_seen":"2023-01-26T20:16:17.618Z","linkquality":40,"power":0,"power_on_behavior":null,"state":"ON","voltage":234}'

Error 2023-01-26 21:16:18Publish 'set' 'state' to 'Tvättmaskinen' failed: 'Error: Command 0x385b44fffe1dc292/242 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 56996 - 242 - 86 - 6 - 11 after 10000ms)'
```

However when turning on/off thr socket with the button it reports just fine in z2m.

closed 2023-03-05 by github-actions[bot]