Koenkk/zigbee2mqtt #15156: Update TS0601 from _TZE200_lsanae15

Auden69, 2022-11-23:
### What happened?

I can pairing the device but some informations are missing.

state (on/off) OK
ac_frequency NOK
energy OK
power NOK
power_factor NOK
voltage NOK
current NOK
produced_energy NOK
linkquality OK

During the pairing, I've error messages: 
Data 3 not defined for '_TZE200_lsanae15' with data {"dp":3,"datatype":0,"data":{"type":"Buffer","data";[22,11,22,11,0,0,22,208]}}
Data 4 not defined for '_TZE200_lsanae15' with data {"dp":4,"datatype":0,"data":{"type":"Buffer","data";[11,24,11,24,0,0,4,69]}}
Data 11 not defined for '_TZE200_lsanae15' with data {"dp":11,"datatype":0,"data":{"type":"Buffer","data";[1,0]}}

### What did you expect to happen?

To read all missing values...

### How to reproduce it (minimal and precise)

Just pairing the device.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Debug log

When i pair the device.... : 

022-11-23 23:32:02MQTT publish: topic 'homeassistant/switch/0xa4c138322e1741bc/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0xa4c138322e1741bc/set","device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/ac_frequency/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"frequency","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc ac frequency","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_ac_frequency_zigbee2mqtt","unit_of_measurement":"Hz","value_template":"{{ value_json.ac_frequency }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc power","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/power_factor/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"power_factor","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc power factor","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_power_factor_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.power_factor }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc current","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/produced_energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc produced energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_produced_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.produced_energy }}"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'homeassistant/sensor/0xa4c138322e1741bc/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138322e1741bc"],"manufacturer":"TuYa","model":"Zigbee DIN energy meter (TS0601_din_1)","name":"0xa4c138322e1741bc"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138322e1741bc","name":"0xa4c138322e1741bc linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138322e1741bc","unique_id":"0xa4c138322e1741bc_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2022-11-23 23:32:02Starting interview of '0xa4c138322e1741bc'
Info 2022-11-23 23:32:02MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138322e1741bc","ieee_address":"0xa4c138322e1741bc","status":"started"},"type":"device_interview"}'
Info 2022-11-23 23:32:02MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xa4c138322e1741bc"},"type":"pairing"}'
Info 2022-11-23 23:32:03Successfully interviewed '0xa4c138322e1741bc', device has successfully been paired
Info 2022-11-23 23:32:03Device '0xa4c138322e1741bc' is supported, identified as: TuYa Zigbee DIN energy meter (TS0601_din_1)
Info 2022-11-23 23:32:03MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee DIN energy meter","exposes":[{"features":[{"access":3,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Measured electrical AC frequency","name":"ac_frequency","property":"ac_frequency","type":"numeric","unit":"Hz"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":1,"description [... cut ...]
Info 2022-11-23 23:32:03MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee DIN energy meter","friendly_name":"0xa4c138322e1741bc","model":"TS0601_din_1","supported":true,"vendor":"TuYa"},"type":"pairing"}'
Info 2022-11-23 23:32:03Successfully configured '0xa4c138322e1741bc'
Info 2022-11-23 23:32:12MQTT publish: topic 'zigbee2mqtt/0xa4c138322e1741bc', payload '{"ac_frequency":null,"current":0,"energy":null,"linkquality":255,"power":null,"power_factor":null,"produced_energy":null,"state":null,"voltage":0}'

If you need more informations, just ask me.

Koenkk, 2022-11-24:
For this you will need to find the datapoints: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

Auden69, 2022-11-24:
Thanks for the link. I have a conbee 2, so i just ordered a Tuya Zigbee gateway to follow steps... I'll come back as soon as i'll find out data points. Thanks for your support.

Auden69, 2022-11-27:
Hello,

I'm back, this is datapoints i found following the guide  (no easy for the noob i am :-)  : 
I don't known if they are all usefull :

DP ID / code
forward energy total = 1
month energy = 3
daily energy = 4 
phase A = 6
Fault = 10
frozen set = 11
switch = 16
Alarm set = 17
Meter ID = 18
Clear event = 20
Forward Energy T1 = 21
Forward Energy T2 = 22
Forward Energy T3 = 23
Forward Energy T4 = 24

What's the next step ?
Thank's for your help...  :-)

Koenkk, 2022-11-27:
I should have checked this earlier, but we recently improved the support of `_TZE200_lsanae15`. These changes are not available in the latest release (will be on 1 December) only in the dev branch.

Can you check with the latest dev branch? (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Auden69, 2022-11-27:
Thanks, 
I tried last dev branch and it's ok ! ... i should try before posting here...
Do you know when i'll be integrated in the master branch ?

Thanks

Koenkk, 2022-11-27:
Next release (1 December)

closed 2022-11-27 by Koenkk

Hadatko, 2022-11-29:
Looking forward for it guys ;) Great work