Koenkk/zigbee2mqtt #15159: Bosch Radiator Thermostat II not pairing

Tempic, 2022-11-24:
### What happened?

I would like to add the Bosch Radiator Thermostat II to z2m.
The device support was [just added](https://github.com/Koenkk/zigbee2mqtt/issues/14926) by @niklasarnitz and from my understanding should be in the dev branch right now.

When trying to pair the device it gets picked up by z2m and the interview is started. After a while I just get an error message stating "Failed to interview". The thermostat shows up as unsupported in my list.

I am a fairly inexperienced user so I may be overlooking something here.

[image]

### What did you expect to happen?

The interview should be successful

### How to reproduce it (minimal and precise)

Allow devices to join via z2m frontend
Enable pairing mode on the thermostat

### Zigbee2MQTT version

1.28.2-dev commit: b15cd43

### Adapter firmware version

20220103

### Adapter

Zigbee 3.0 USB Dongle Plus, zStack3x0

### Debug log

2022-11-24 02:26:54Starting interview of '0x18fc2600000a0f85'
2022-11-24 02:26:54MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x18fc2600000a0f85","ieee_address":"0x18fc2600000a0f85","status":"started"},"type":"device_interview"}'
2022-11-24 02:26:54MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x18fc2600000a0f85"},"type":"pairing"}'
2022-11-24 02:29:16Failed to interview '0x18fc2600000a0f85', device has not successfully been paired

niklasarnitz, 2022-11-24:
Did you add the install code of your radiator thermostat to your zigbee dongle?
Without it, it will not pair.

You can find it as a QR code in the battery compartment.

niklasarnitz, 2022-11-24:
@Koenkk This should propably added to the documentation of the device.
Since I am far away from a computer for a while because I’m on vacation, could you add the note?

Tempic, 2022-11-24:
> Did you add the install code of your radiator thermostat to your zigbee dongle?

@niklasarnitz
That did it!
I haven't heard of install codes before, but now it got added successfully. 

Thank you for your very fast reply!

Tempic, 2022-11-24:
@niklasarnitz 
Sorry to bother you again in your vacation.

I was able to pair it yesterday and all the data it exposes show up in the z2m frontend.
I cannot make any changes to the device though. When I try to make a change I get an error a few seconds later.
In this example I was trying to set the temperature to 22°C.
Is there something else that needs to be changed or unlocked to be able to write to the thermostat? 
The thermostat was testes with manual and time setting on the device but no differences there.

Debug 2022-11-24 13:45:35Publishing 'set' 'occupied_heating_setpoint' to 'Wohnzimmer Thermostat Bosch'

Debug 2022-11-24 13:45:49Received Zigbee message from 'Wohnzimmer Thermostat Bosch', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2340}' from endpoint 1 with groupID 0

Info 2022-11-24 13:45:49MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Thermostat Bosch', payload '{"battery":100,"linkquality":105,"local_temperature":23.4,"local_temperature_calibration":null,"occupied_heating_setpoint":20.5}'

Debug 2022-11-24 13:46:19Received Zigbee message from 'Wohnzimmer Thermostat Bosch', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2350}' from endpoint 1 with groupID 0

Info 2022-11-24 13:46:19MQTT publish: topic 'zigbee2mqtt/Wohnzimmer Thermostat Bosch', payload '{"battery":100,"linkquality":112,"local_temperature":23.5,"local_temperature_calibration":null,"occupied_heating_setpoint":20.5}'

Error 2022-11-24 13:46:28Publish 'set' 'local_temperature_calibration' to 'Wohnzimmer Thermostat Bosch' failed: 'Error: Write 0x18fc2600000a0f85/1 hvacThermostat({"localTemperatureCalibration":-12}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

Info 2022-11-24 13:46:28MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'local_temperature_calibration' to 'Wohnzimmer Thermostat Bosch' failed: 'Error: Write 0x18fc2600000a0f85/1 hvacThermostat({\"localTemperatureCalibration\":-12}, {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"Wohnzimmer Thermostat Bosch"},"type":"zigbee_publish_error"}'

niklasarnitz, 2022-11-24:
I had that error once when adding the device first after fumbling around with the install code things.

The solution for me was to reset the device, force-remove the device from z2mqtt and then re pair the whole setup.

Tempic, 2022-11-24:
@niklasarnitz 
And you saved my day again :)
Deleting, resetting and re-pairing for 2 times did it and everything seems to be working now.

niklasarnitz, 2022-11-24:
No problem :)The Bosch thermostats sometimes remember a bit too much I guess.Sent from my iPhone

hannesfischer, 2023-02-11:
> Did you add the install code of your radiator thermostat to your zigbee dongle?
> Without it, it will not pair.
> 
> You can find it as a QR code in the battery compartment.

Hi,
Im kind off stuck here. Where do you add the Install Codes in zigbee2mqtt? I can't find it anywhere.

Tempic, 2023-02-11:
@hannesfischer Open the Zigbee2MQTT dashboard, then navigate to "Settings" at the top.
Under "Tools" you are able to add the install code from the thermostat.

closed 2023-02-11 by Tempic

hannesfischer, 2023-02-11:
@Tempic Thank you so much. I kind of missed that setting. It works now :D

tsloms, 2023-02-24:
Hey!
I have a similar problem. When I try to add the install Code z2mqtt states of would ne top short. The install Code printed on the thermostat is only 36 chars. Or 44 of you count the 'minus'. Any Suggestionen?

hannesfischer, 2023-02-24:
> Hey!
> I have a similar problem. When I try to add the install Code z2mqtt states of would be to short. The install Code printed on the thermostat is only 36 chars. Or 44 of you count the 'minus'. Any Suggestionen?
> 
> Edit. Z2mqtt expects 91 or 95...

I had the same issue at first. You have to scan the qr code next to the hex values. The text, the qr code returns, is the actual install code.

tsloms, 2023-02-24:
> I had the same issue at first. You have to scan the qr code next to the hex values. The text, the qr code returns, is the actual install code.

Thanks for your hint and the Fast reply!
Now z2mqtt says install Codes are not supported. I'm using a ConBee II stick, does this produce any issues?

hannesfischer, 2023-02-24:
I don't know if the conbee supports install codes. I'm using the sonoff zigbee 3.0 dongle.

sbley, 2023-10-22:
How do I get the thermostate into pairing mode? Just reinsert the batteries or do I have to hold the button for some seconds while inserting the batteries?

malicifus, 2024-01-11:
> How do I get the thermostate into pairing mode? Just reinsert the batteries or do I have to hold the button for some seconds while inserting the batteries?

As far as i know, keep holding the large button on the front while reinserting the batteries. The display will read "RES". Let go of the button and keep holding it again, until the device flashes green. It should restart and reset. At least thats whats in the manual

JGee1F, 2024-01-22:
> Thanks for your hint and the Fast reply! Now z2mqtt says install Codes are not supported. I'm using a ConBee II stick, does this produce any issues?

https://github.com/Koenkk/zigbee2mqtt/issues/17492