Koenkk/zigbee2mqtt #15161: [New device support]: TS0601 _TZE200_qtbrwrfv

EelcoS73, 2022-11-24:
### Link

https://alectohome.nl/collections/alecto_rookmelders/products/alecto-smart-smoke10-smart-zigbee-rookmelder-wit

### Database entry

{"id":13,"type":"EndDevice","ieeeAddr":"0x84fd27fffe8af2ea","nwkAddr":30502,"manufId":4098,"manufName":"_TZE200_qtbrwrfv","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��\u0011+\u0012��\u0011+\u0012","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_qtbrwrfv","stackVersion":0,"dateCode":"","appVersion":85}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1669280799032,"defaultSendRequestWhen":"immediate"}

### Comments

Recently I bought a second Alecto Smart-Smoke 10 smokedetector, but unlike the first one this one isn't supported.

The first one gave me this info:
Zigbee Model: tbrwrfv
Zigbee Manufacturer: _TYST11_qtbrwrfv
Description: Smoke detector
Support status: Supported
Firmware build date: 20180727
Manufacturer: Alecto
Model: SMART-SMOKE10

The second one:
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_qtbrwrfv
Support status: Not supported

The only visible difference is the production date, the one that is working is from 08/2020 and the second one that isn't working is from 03/2021

Additional info:
Zigbee2MQTT version: 1.28.2
Coordinator type: zStack3x0
Coordinator revision: 20220219

Thanks for your help!

Koenkk, 2022-11-24:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'tbrwrfv\u0000', manufacturerName: '_TYST11_qtbrwrfv'}, {modelID: 'TS0601', manufacturerName: '_TZE200_qtbrwrfv'}],
    model: 'SMART-SMOKE10',
    vendor: 'Alecto',
    description: 'Smoke detector',
    fromZigbee: [fzLocal.tuya_alecto_smoke],
    toZigbee: [tzLocal.tuya_alecto_smoke],
    meta: {},
    exposes: [exposes.text('smoke_state', ea.STATE, ['alarm', 'normal']),
        exposes.text('battery_state', ea.STATE, ['low', 'middle', 'high']),
        exposes.text('checking_result', ea.STATE, ['checking', 'check_success', 'check_failure', 'others']),
        exposes.numeric('smoke_value', ea.STATE),
        exposes.numeric('battery', ea.STATE),
        exposes.binary('lifecycle', ea.STATE, true, false),
        exposes.binary('self_checking', ea.STATE_SET, true, false),
        exposes.binary('silence', ea.STATE_SET, true, false)],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

EelcoS73, 2022-11-24:
When I do this Zigbee2MQTT is not starting anymore. Am I doing something wrong?

Koenkk, 2022-11-25:
Please try with https://gist.github.com/Koenkk/2c37552a3d16fc7d29c1838a99ee20f0

EelcoS73, 2022-11-28:
Thanks Koen,

With the link provided in your answer I succeded in getting the smoke detector to work!
I guess the housing is the same but what's inside is completely different. Do you include this converter in the next upgrade of Zigbee2MQTT? I think that anyone who buys this smokedetector will encounter the same problem.

Koenkk, 2022-11-28:
Added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

blakadder, 2022-12-05:
Wasn't this supposed to go under SMART-SMOKE10 model?