Koenkk/zigbee2mqtt #15164: Heating is completely broken right now (At least for TV02 thermostats)

timderspieler, 2022-11-24:
### What happened?

After the most recent changes to the tuya converter (https://github.com/Koenkk/zigbee-herdsman-converters/pull/4750) the heating with TV02 radiator valves is completely broken.

When there are manual changes made (e.g with better thermostat doing local temperature updates or scheduler component doing the scheduled temperature changes) the thermostat switches back to auto mode and completely ignores the adjusted attributes.

This currently leads to radiator valves (with the model TV02 at least, I dont have any other valves installed in my home) being completely open 24/7 -> Even if I turn the knob on the thermostat itself the changes will be discard and the heating continues.

Especially in this winter I wish that this issue gets resolved fast. I see no other way than remove them from Z2M right now and controlling them manually by hand (which sucks).

### What did you expect to happen?

I expected that the valve accepts the manual temperature changes and will stay at "manual" mode.

### How to reproduce it (minimal and precise)

1) Change the local_temperature_calibration or any other attribute of the valve in Z2M / HA / or by hand at the valve
2) The system mode will then switch from manual to auto

### Zigbee2MQTT version

1.28.2-dev commit: ca4badc

### Adapter firmware version

20210708

### Adapter

zStack3x0

Koenkk, 2022-11-24:
> When there are manual changes made (e.g with better thermostat doing local temperature updates or scheduler component doing the scheduled temperature changes) the thermostat switches back to auto mode and completely ignores the adjusted attributes.

If I understand correctly, when setting the temperature of this thermostat via MQTT, the thermostat kicks itself out of manual mode and switches back to auto mode? (and therefore ignore the set temperature)

Hessenpower01, 2022-11-24:
> If I understand correctly, when setting the temperature of this thermostat via MQTT, the thermostat kicks itself out of manual mode and switches back to auto mode? (and therefore ignore the set [...]

Dear @Koenkk, that's exactly what happens. Also when when switching it from "off" to "on" it changes to "auto" mode.

I don't see a usecase where people using them with Z2M (HA or whatever in the background) will ever have any use of the auto mode, as automation is done in HA.

As always, thanks a lot for your work!

Meik

Hessenpower01, 2022-11-24:
> So when after setting the temperature you also set the mode again, it works correctly?

Yes and no. Because when switching between the modes it also changes the temperature that is assigned from the mode. So when switching just back to the manual mode, it stays in the temperature that came from auto mode.

holantomas, 2022-11-24:
@Koenkk yes. If you have automation for getting actual temperature(no automatic report, need to set `online` attr to `ON` in some time periods you want) it even ignores what you set directly on TVR by buttons.

Koenkk, 2022-11-24:
@Hessenpower01 so what do you need to publish in order to keep it in manual and set the temperature?

Hessenpower01, 2022-11-24:
> @Hessenpower01 so what do you need to publish in order to keep it in manual and set the temperature?

... I don't know ... sorry. Which info may I offer?

holantomas, 2022-11-24:
@Koenkk 
@vladi1234 said in prev topic that he working on patch for this. I think he knows what to do because he already did that once.

Koenkk, 2022-11-24:
@vladi1234 can you answer this?

> so what do you need to publish in order to keep it in manual and set the temperature?

rendo0m, 2022-11-24:
Hello, 
got my TV02 since two week and wondered about this problem [found here](https://github.com/Koenkk/zigbee2mqtt/issues/14867). 

I tried to "betray" the scheduler. but that's that same.. so i went to Scenes. Default is 0. 
I changed it, without filling anything else, to manual and i think it solved the problem.

[image]

Maybe it's the way, to get it working, without switching back to auto after setting the temperature via "currentTemperature".
Let me know if this helped you too.
Cheers

Hessenpower01, 2022-11-24:
> Hello, got my TV02 since two weeks and wondered about this problem [found here](https://github.com/Koenkk/zigbee2mqtt/issues/14867).
> 
> I tried to "betray" the scheduler. but that's that same.. so i went to Scenes. Default is 0. I changed it, without filling anything else, to **manual** and i think it solved my problem.
> 
> [image]
> 
> Maybe it's the way, to get it working, without switching back to auto after setting the temperature via "currentTemperature". Let me know if this helped you too. Cheers

@rendo0m the setting as you showed ist standard, already like this in my installation

rendo0m, 2022-11-24:
sad. mine was "auto".

vladi1234, 2022-11-24:
Hello everyone,

I am planning to reanimate and integrate my extended code which is now bug free and includes new features.

Has a working `system_mode` for `Homeassistant`.
Similarly, it has a functional `schedule`.
Some other small corrections.

**All known problems are eliminated.**

[image]

Hessenpower01, 2022-11-24:
> Hello everyone,
> 
> i tried to remove errors from the user code "vladyspavlov", but I also found that this does not make sense, since in total it has more than 10 errors embedded in the code, which makes our thermostat stupid.
> 
> I am planning to reanimate and integrate my extended code which is now bug free and includes new features.
> 
> Has a working `system_mode` for `Homeassistant`. Similarly, it has a functional `schedule`. Some other small corrections.
> 
> **All known problems are eliminated.**
> 
> [image]

Happy to read that :-)

Wouldn't mind if OTA would also work ... thank you so much for your efforts

tlanfer, 2022-11-25:
Interestingly i seem to be having this exact issue, as far as i can tell. I try and change settings, but it doesnt react at all in the z2m gui, and if i try changing it from HA it swaps back after a second or two. I can change temperature on the physical device, and i see the change reflected in z2m.

Whats weird: i'm running it in docker on my homeserver, and i told docker-compose to pull and run the :1.28.0 tag to see if that fixes it and it looks like it doesnt seem to make a difference.

vladi1234, 2022-11-26:
Pull request https://github.com/Koenkk/zigbee-herdsman-converters/pull/5022

station7, 2022-11-27:
@vladi1234: Does your PR fix the set-window_open-issue ([#14163](https://github.com/Koenkk/zigbee2mqtt/issues/14163) and [#15181](https://github.com/Koenkk/zigbee2mqtt/issues/15181)) as well? I already had the set window_open-error before and not only in the latest version based on vladyspavlov's recent commits...

vladi1234, 2022-11-27:
> @vladi1234: Does your PR fix the set-window_open-issue ([#14163](https://github.com/Koenkk/zigbee2mqtt/issues/14163) and [#15181](https://github.com/Koenkk/zigbee2mqtt/issues/15181)) as well? I [...]

Can you perform TVR reset and then tell if it helped?

station7, 2022-11-28:
I set up a test environment with latest z2m-dev and applied your patch. At first it seemed to work, but if I now set "open_window" to ON via Webfrontend, it switches back to "OFF" after a second, **unless I set "system_mode" to "off"** - then it stays at "open_window = ON".

Please let me know, if you need more information or data - I am happy to assist.

Lipown, 2022-12-04:
Has the same issue, can you help me out how to apply the patch please?

vladi1234, 2022-12-04:
Our TV02 does not seem quite compatible with new Tuya format.

vladi1234, 2022-12-06:
Bug fix for known bugs TV02 which were installed in 0.28.1.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/5096

vladi1234, 2022-12-08:
This may take some time until the pull request comes in.

Hessenpower01, 2022-12-15:
**Workaround**

To be able to use them as they behave currently I leave them all the time in manual mode. I don't turn them off but reduce temperature to 5°C instead. Also created a script named "heating off" which in fact just set's temperature to 5°C.
So they remain in manual and temperature values coming from HA apply.

vladi1234, 2022-12-15:
The problem with the `auto` has already been fixed.

Wikibear, 2023-01-03:
If you want control everything, you don't need the auto mode or off or anything else. In my opinion you need this only if you don't want control it from outside. All other settings result in that what you setup into z2m. If you control it via Homeassist or Openhab iobroker etc. Then leave it on manual. Everything else you control it via Setpoint temperature. Nothing else will happen with auto, window open or off. That is how it is working for me and very fine. 5° are the lowest and the valve is closed. 8° is frost protection.
I didn't found any protection that the vent can't move over summer period if the vent is all the time closed and the winter season starts. Are you know about this kind of protection?
My old thermostat drive the vent fully each Thursday at 11 o'clock to prevent of stuck. If not, so I must develop a rule for it...

eknoes, 2023-01-04:
> I set up a test environment with latest z2m-dev, applied your patch, resetted TRV and paired it. At first it seemed to work, but if I now set "open_window" to ON via Webfrontend, it switches back to "OFF" after a second, **unless I set "system_mode" to "off"** - then I am able to toggle "open_window" as expected and if I switch "open_window" "ON" it stays "ON".
> 
> Please let me know, if you need more information or data - I am happy to assist.

I have currently the same issue. @vladi1234, afaik you fixed a similar problem in your PR for frost_protection and replaced the converter with `tuya.valueConverter.TV02FrostProtection`. Do you think a similar thing might fix the window_open problem? I have no idea of how all of this works yet, but might take a look into it if you think the cause is similar.

vladi1234, 2023-01-04:
I don't just have `frost_protection` `tuya.valueConverter.TV02FrostProtection` replaced, but also `heating_stop` and `system_mode` `tuya.valueConverter.TV02SystemMode` replaced. That should fix problem with switching into `auto` mode.

Your problem with PR does not have the slightest to do.

My question is, for what purpose do you want to use `open_window`?

eknoes, 2023-01-05:
I don't mean that my problem is a result of your PR :) As far as I understand it your PR solved a similar problem with frost_protection as I have with open_window. I am using it coupled with my window sensors to disable heating. That works for quite some time, but after a certain time it always resets itself from on to off immediately, as frost_protection did before your PR. My question is whether replacing open_windows method with your new method might solve that problem too?

vladi1234, 2023-01-05:
> My question is whether replacing open_windows method with your new method might solve that problem too?

Replacing 'open_windows' method can not fix these problems.

You must not connect a window sensor to `open_windows` it works **automatically**. See an excerpt from Manual for Thermostat:
```
8.4 Open Window Detection
The device automatically stop heating when it detects a sudden temperature drop(5°C in 5 minutes as default). This is usually caused by an opened window or door and the open window icon will display(open) on the device. The device will operate according to the preset window. Press the pair button to cancel.
App operation: Click the open window icon in APP to cancel the window opening function.
The open window detection only operates in automatic mode and manual mode.
```

For window sensor you use `heating_stop` then it works correctly.

eknoes, 2023-01-05:
Ah ok. Weird. Do you know why it then works after the reset for a couple of hours? See #14163

vladi1234, 2023-01-05:
I have already tried to reproduce the error, but I did not succeed. I don't think it will work properly for two hours after the reset. That does not really close TRV either.

In any case, **Open Window Detection** is not there for connecting to window sensor.

Function **heating_stop** works super well with window sensor and closes TRV properly.

eknoes, 2023-01-05:
> In any case, **Open Window Detection** is not there for connecting to window sensor.
> 
> Function **heating_stop** works super well with window sensor and closes TRV properly.

I just adapted my automations to use that instead.

> I have already tried to reproduce the error, but I did not succeed. I don't think it will work properly for two hours after the reset. That does not really close TRV either.

Oh I think it did. The TRV showed something that was not a temperature, when window_open was on. But if it does now work that way too, its fine for me.

DaN66O, 2023-01-08:
**Zigbee2MQTT version [1.29.1] commit: [7d67ffc]
Coordinator type
zStack30x
Coordinator revision
20211222**

I have this TV02 radiator valve
I set the target temperature, for example, at 25 degrees.
I adjust the local temperature to 26 degrees and the valve closes.
Everything is fine here.
Now I reduce the local temperature....
25 degrees - the valve is closed.
24 degrees - closed
23 degrees - closed **( not half closed or open, but completely closed! )**
...
21 degrees - the valve opens

**A difference of 4 degrees is unacceptable. :(**
What can be done about it ?
Or do I have a defective TV02 radiator valve ?

eknoes, 2023-01-09:
@DaN66O I had a similar problem after some time, but the boost still worked. Now I've resetted both and now it seems to work again.

station7, 2023-01-10:
> Function **heating_stop** works super well with window sensor and closes TRV properly.

Every time my window sensor sets **heating_stop="OFF"** when the window closes again, the [climate preset](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#climate) changes from "auto" to "manual". Does this happen on purpose? I would have expected that setting "heating_stop" to "OFF" does not have an impact on the chosen preset. Or am I missing something?

eknoes, 2023-01-10:
As far as I know this was fixed with the current version by @vladi1234

station7, 2023-01-10:
I am using the latest version [1.29.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.29.1)...

vladi1234, 2023-01-10:
> I set up a test environment with latest z2m-dev, applied your patch, resetted TRV and paired it. At first it seemed to work, but if I now set "open_window" to ON via Webfrontend, it switches back to [...]

> > Function **heating_stop** works super well with window sensor and closes TRV properly.
> 
> @vladi1234 Every time my window sensor sets **heating_stop="OFF"** when the window closes again, the [climate preset](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#climate) changes from "auto" to "manual". Does this happen on purpose? I would have expected that setting "heating_stop" to "OFF" does not have an impact on the chosen preset. Or am I missing something?

Do you want to use `window_open` or `heating_stop` in `auto` mode Schedule, whether I understand properly?

DaN66O, 2023-01-10:
> @DaN66O I did not understand how you set something, but no matter. I understood only that TRVhas 4 degrees difference.
> 
> 1. You can check TRV adapters.
> 2. Reset "of course important as well as says @eknoes"
> 3. Install new batteries and, of course, make a new installation.
> 
> If this does not help, then there is still `local_temperature_calibration`, since you can correct -5 to +5 degrees of deviation.

I did the installation from the beginning.
The batteries are new.
You misunderstood me. (English is not my native language )
I set **current_heating_setpoint** to 25 degrees.
Then using **local_temperature_calibration**, I change **local_temperature** to 26 and the valve closes immediately.
Everything is fine here.
Then, using **local_temperature_calibration**, I change **local_temperature** to 24 and the valve is still closed.
I thought the valve should be opening by now. !!
23 - the valve is closed....
So up to 21 degrees, only then the valve opens.
At the same time, let me remind you, **current_heating_setpoint** = 25

station7, 2023-01-10:
Hi @vladi1234, 

as I leard from your post, `window_open` should not be set based on an external window contact, so I adjusted my node-red-setup to set `heating_stop` to „ON“ instead when the window is open. And yes, I was thinking of using this in auto mode with the schedule I set over the webinterface (thanks for that as well).

vladi1234, 2023-01-10:
Hi @station7,

> Function heating_stop works super well with window sensor and closes TRV properly.

When I wrote this, I only meant the `manual` mode. But not `auto` = schedule.

> @vladi1234 Every time my window sensor sets heating_stop="OFF" when the window closes again, the [climate preset](https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#climate) changes from "auto" to [...]

This configuration as you use is not provided.

Yes, this is correct TRV behavior, that was also planned like that, because that must be according to logic.
In order to be able to control the temperature better via home automation, or to be able to set it according to daily requirements via voice control, you need `manual` mode.
Schedule `auto` in this case is not necessary.

I see only one way to use `auto` schedule. Easy to rely on thermostat's own `Open Window Detection` without using window sensor.

See an excerpt from Manual for Thermostat:
```
8.4 Open Window Detection
The device automatically stop heating when it detects a sudden temperature drop(5°C in 5 minutes as default). This is usually caused by an opened window or door and the open window icon will display(open) on the device. The device will operate according to the preset window. Press the pair button to cancel.
App operation: Click the open window icon in APP to cancel the window opening function.
The open window detection only operates in automatic mode and manual mode.
```

rotilho, 2023-02-16:
@DaN66O same issue I even contact Tuya but they said I should contact the seller.

Now I use the valve as on/off: I set it to 30 degrees or use the heat stop.

closed 2023-03-26 by github-actions[bot]