Koenkk/zigbee2mqtt #15170: [Feature request]: support for Bearer token

waclaw66, 2022-11-24:
### Is your feature request related to a problem? Please describe

My frontend runs on different machine in the same C network as the Nginx reverse proxy. I would like to secure that frontend with Bearer Authorization from unauthorized direct access to IP e.g. `http://192.168.1.100:8080`. Frontend would reject then all unauthorized access with `401` instead of that popup box with `Enter your z2m admin token` message.

### Describe the solution you'd like

Nginx reverse proxy would add a header with the authorization sent to frontend url...
`proxy_set_header Authorization "Bearer secret_token";`
User authorization is secured on side of reverse proxy.

### Describe alternatives you've considered

None

### Additional context

Nothing

Koenkk, 2022-11-25:
Please create issues related to the frontend here, note that you can already specify the token by setting the `token` query param: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2022-11-25 by Koenkk

waclaw66, 2022-11-28:
> note that you can already specify the token by setting the token query param

Yes, that's true, however zigbee2mqtt has `zigbee2mqtt-frontend` version `0.6.114` in dependencies, the `token` argument is supported since `0.6.115`.

Koenkk, 2022-11-28:
It will be supported in the next release (1 December)

waclaw66, 2022-11-29:
> It will be supported in the next release (1 December)

Thanks, I've updated `zigbee2mqtt-frontend` manually meanwhile, it works!