Koenkk/zigbee2mqtt #15171: Ubisys C4 - configuration of templates crash zigbee2mqtt and do not work correctly

pehon, 2022-11-24:
### What happened?

Applying a template to a Ubisys C4 does not work as documented (or I understand it wrong) and is also able to crash zigbee2mqtt.

Ubisys C4 Firmware 2.0.0 and 1.9.4

Reference:
[https://www.zigbee2mqtt.io/devices/C4.html](url)

### What did you expect to happen?

configure the Ubisys C4 buttons to dimmer_single

### How to reproduce it (minimal and precise)

Error 1:
```
mosquitto_pub -h mqtt -t 'zigbee2mqtt/Test_Taster_UbisysC4/set' -m '{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            }
        ]
    }
}'
```

Log:
```
Nov 24 19:20:17 rockpis npm[8826]: Zigbee2MQTT:warn  2022-11-24 19:20:17: ubisys: Using input(s) 0 and endpoint 1 for 'dimmer_single'.
Nov 24 19:20:17 rockpis npm[8826]: Zigbee2MQTT:warn  2022-11-24 19:20:17: ubisys: Using input(s) 1 and endpoint 2 for 'dimmer_single'.
Nov 24 19:20:17 rockpis npm[8826]: Zigbee2MQTT:warn  2022-11-24 19:20:17: ubisys: Using input(s) 2 and endpoint 3 for 'dimmer_single'.
Nov 24 19:20:17 rockpis npm[8826]: Zigbee2MQTT:warn  2022-11-24 19:20:17: ubisys: Using input(s) 3 and endpoint 4 for 'dimmer_single'.
Nov 24 19:21:18 rockpis npm[8826]: Zigbee2MQTT:error 2022-11-24 19:21:18: Publish 'set' 'configure_device_setup' to 'Test_Taster_UbisysC4' failed: 'Error: Write 0x001fee0000008bcc/232 manuSpecificUbisysDeviceSetup({"inputActions":{"elementType":"octetStr","elements":[[0,7,1,6,0,2],[0,134,1,8,0,5,0,50],[0,198,1,8,0,5,1,50],[0,11,1,8,0,3],[1,7,2,6,0,2],[1,134,2,8,0,5,0,50],[1,198,2,8,0,5,1,50],[1,11,2,8,0,3],[2,7,3,6,0,2],[2,134,3,8,0,5,0,50],[2,198,3,8,0,5,1,50],[2,11,3,8,0,3],[3,7,4,6,0,2],[3,134,4,8,0,5,0,50],[3,198,4,8,0,5,1,50],[3,11,4,8,0,3]]}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Nov 24 19:21:18 rockpis npm[8826]: Zigbee2MQTT:info  2022-11-24 19:21:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'configure_device_setup' to 'Test_Taster_UbisysC4' failed: 'Error: Write 0x001fee0000008bcc/232 manuSpecificUbisysDeviceSetup({\"inputActions\":{\"elementType\":\"octetStr\",\"elements\":[[0,7,1,6,0,2],[0,134,1,8,0,5,0,50],[0,198,1,8,0,5,1,50],[0,11,1,8,0,3],[1,7,2,6,0,2],[1,134,2,8,0,5,0,50],[1,198,2,8,0,5,1,50],[1,11,2,8,0,3],[2,7,3,6,0,2],[2,134,3,8,0,5,0,50],[2,198,3,8,0,5,1,50],[2,11,3,8,0,3],[3,7,4,6,0,2],[3,134,4,8,0,5,0,50],[3,198,4,8,0,5,1,50],[3,11,4,8,0,3]]}}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Test_Taster_UbisysC4"},"type":"zigbee_publish_error"}'
```

Press button1 on the device:
```
Nov 24 19:21:42 rockpis npm[8826]: Zigbee2MQTT:info  2022-11-24 19:21:42: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/update-state', payload 'idle'
Nov 24 19:21:42 rockpis npm[8826]: Zigbee2MQTT:info  2022-11-24 19:21:42: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/update_available', payload 'false'
Nov 24 19:21:42 rockpis npm[8826]: Zigbee2MQTT:info  2022-11-24 19:21:42: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/linkquality', payload '255'
Nov 24 19:21:42 rockpis npm[8826]: Zigbee2MQTT:info  2022-11-24 19:21:42: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/action', payload 'toggle'
```

shout be 1_toggle, 1_level_move_down or 1_level_move_up

Error 2:
```
mosquitto_pub -h mqtt -t 'zigbee2mqtt/Test_Taster_UbisysC4/set' -m '{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            }
        ]
    }
[... 89 lines cut ...]
Nov 24 19:24:41 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:24:41: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/configure_device_setup-input_actions', payload '0,13,1,6,0,1,1,13,2,6,0,1,2,13,3,6,0,1,3,13,4,6,0,1'
```

Press button1 on the device:
```
Nov 24 19:25:47 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:25:47: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/update-state', payload 'idle'
Nov 24 19:25:47 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:25:47: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/update_available', payload 'false'
Nov 24 19:25:47 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:25:47: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/linkquality', payload '255'
Nov 24 19:25:47 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:25:47: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/configure_device_setup-input_actions', payload '0,13,1,6,0,1,1,13,2,6,0,1,2,13,3,6,0,1,3,13,4,6,0,1'
Nov 24 19:25:47 rockpis npm[8910]: Zigbee2MQTT:info  2022-11-24 19:25:47: MQTT publish: topic 'zigbee2mqtt/Test_Taster_UbisysC4/action', payload 'on'
```

this is ok, but not what I actually need

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2

sjorge, 2022-11-25:
A big with this was recently fixed, it's not in the latest release yet, it should be available in the December release:
https://github.com/Koenkk/zigbee2mqtt/issues/14932

pehon, 2022-11-27:
Thanks for you answer. I was trying to confirm this issue with the dev branch, but I guess I'm not able to switch to the dev branch correctly:

```
git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git
git checkout -b dev
npm ci
npm start
```
But I'm still on version 1.28.2 (commit #e66fe85)

Guess I have to wait for the December release.

The steps are mentioned in [https://github.com/Koenkk/zigbee2mqtt/issues/1174](url) because the manual does not work for my setup [https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html](url)

pehon, 2022-12-05:
After upgrading to [1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [52e545f9](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f9) the same issue occurs. How do I reopen this issue?

ahemwe, 2022-12-10:
I confirm that the issue still occurs on commit [52e545f9](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f9).

Could you please re-open the issue @Koenkk

pehon, 2023-01-15:
With [1.29.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.29.2) commit: [bb3e8f6d](https://github.com/Koenkk/zigbee2mqtt/commit/bb3e8f6d) I was able to configure two inputs sucessfully
But with 4 i get

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'configure_device_setup' to 'Test_C4_Taster' failed: 'Error: Write 0x001fee0000008c04/232 manuSpecificUbisysDeviceSetup({\"inputActions\":{\"elementType\":\"octetStr\",\"elements\":[[0,7,1,6,0,2],[0,134,1,8,0,5,0,50],[0,198,1,8,0,5,1,50],[0,11,1,8,0,3],[1,7,2,6,0,2],[1,134,2,8,0,5,0,50],[1,198,2,8,0,5,1,50],[1,11,2,8,0,3],[2,7,3,6,0,2],[2,134,3,8,0,5,0,50],[2,198,3,8,0,5,1,50],[2,11,3,8,0,3],[3,7,4,6,0,2],[3,134,4,8,0,5,0,50],[3,198,4,8,0,5,1,50],[3,11,4,8,0,3]]}}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Test_C4_Taster"},"type":"zigbee_publish_error"}'
Publish 'set' 'configure_device_setup' to 'Test_C4_Taster' failed: 'Error: Write 0x001fee0000008c04/232 manuSpecificUbisysDeviceSetup({"inputActions":{"elementType":"octetStr","elements":[[0,7,1,6,0,2],[0,134,1,8,0,5,0,50],[0,198,1,8,0,5,1,50],[0,11,1,8,0,3],[1,7,2,6,0,2],[1,134,2,8,0,5,0,50],[1,198,2,8,0,5,1,50],[1,11,2,8,0,3],[2,7,3,6,0,2],[2,134,3,8,0,5,0,50],[2,198,3,8,0,5,1,50],[2,11,3,8,0,3],[3,7,4,6,0,2],[3,134,4,8,0,5,0,50],[3,198,4,8,0,5,1,50],[3,11,4,8,0,3]]}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'

```
mosquitto_pub -h mqtt -t 'zigbee2mqtt/Test_C4_Taster/set' -m '{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            },
            {
                "type": "dimmer_single"
            }
        ]
    }
}'
```

sjorge, 2023-01-17:
Yeah there is still something weird, I think it might be the device firmware on the 2.x+

I can't read my old (complex and thus big payloads) but the devices still work fine, I cannot write the same payload (had the mosquitto_pub commands saved) to a new device of the same kind. If I write a smaller payload I sometimes also can't read it back. 😞

sjorge, 2023-01-26:
Ubisys dropped an update that switches to the 4.x zstack from the 3.x zstack they use.

Nothing specific about the issues we're seeing but
```
* Ongoing maintenance of the code base: Restructuring, refactoring and consolidating
* Round-up of bug-fixes and performance improvements
```

So who knows if they fixed whatever is causing the issue. I'll try and update tonight and see if there is an improvement.

sjorge, 2023-01-26:
It is not fixed :(

pehon, 2023-01-26:
did a firmware upgrade on a C4 from 2.0.0 to 2.1.2
But its not fixed here as well :-(

ahemwe, 2023-01-27:
Out of curiosity: How did you do the update? #15194 shows that OTA with z2m is not working

sjorge, 2023-01-27:
Mine just showed it had an update, it installed without issue.On C4, S2, S1, J1, and H1 (don’t have any other types)~ sjorge

sjorge, 2023-02-27:
Sadly still an issue, z2m just no longer crashes.

I did verify with wireshark that it is the device sending the INSUFFICIENT_SPACE error back to z2m so I'm not sure there is anything we can do z2m side.

closed 2023-04-08 by github-actions[bot]