Koenkk/zigbee2mqtt #15172: NPM version 9.1.2.  not support

Dylantje, 2022-11-24:
### What happened?

Update my Odroid N2
Then Reboot
z2M not working

### What did you expect to happen?

Start z2m

### How to reproduce it (minimal and precise)

Login with putty and run with log

pi@odroidn2:~$ node --version
v16.18.1
pi@odroidn2:~$ npm --version
9.1.2
pi@odroidn2:~$ npm uninstall

### Zigbee2MQTT version

 1.28.2 commit: e66fe85

### Adapter firmware version

    20201026

### Adapter

    zStack3x0

Koenkk, 2022-12-03:
Can you provide the error log?

lux73, 2023-02-05:
i can actually confirm that **npm 9.4.1 works** without any hassle

have upgraded from 8.19.3 to 9.4.1 few minutes ago without any Problem

z2m v1.30.0 / Debian 11.6