Koenkk/zigbee2mqtt #15175: Error: Failed to connect to the adapter

brantmacga, 2022-11-25:
### What happened?

Cannot get Zigbee2MQTT to start. I have followed every step shown in the help section for "Error: SRSP - SYS - ping after 6000ms". 

I previously had ZHA running; I removed all zigbee devices from the network and deleted the ZHA integration. Could possibly be related to this, but I'm not sure where to look next. As far as I can see, ZHA no longer exists within my installation. 

### What did you expect to happen?

Expected Zigbee2MQTT to start correctly. 

### How to reproduce it (minimal and precise)

Start Zigbee2MQTT, receive error. 

### Zigbee2MQTT version

1.28.2-1

### Adapter firmware version

6.10.3

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E Gateway

### Debug log

[23:16:11] INFO: Preparing to start...
[23:16:11] INFO: Socat not enabled
[23:16:13] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2022-11-24 23:16:18: Logging to console and directory: '/config/zigbee2mqtt/log/2022-11-24.23-16-18' filename: log.txt
Zigbee2MQTT:info  2022-11-24 23:16:18: Starting Zigbee2MQTT version 1.28.2 (commit #unknown)
Zigbee2MQTT:info  2022-11-24 23:16:18: Starting zigbee-herdsman (0.14.68)
Zigbee2MQTT:error 2022-11-24 23:16:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-11-24 23:16:38: Failed to start zigbee
Zigbee2MQTT:error 2022-11-24 23:16:38: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-11-24 23:16:38: Exiting...
Zigbee2MQTT:error 2022-11-24 23:16:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)

kirovilya, 2022-11-25:
specify
`adapter: ezsp`
in the configuration, next to the port address

Sorin-1, 2022-11-25:
I have sorted it out for me. I also use "new" Sonoff USB "E" adapter (that is built on another chip then "P" adapter was).

1. Disable integeration "Zigbee Home Automation", which may have been automatically installed while plugging in "E"-type Sonoff adapter. Interation ZHA causes port malfunction when Zigbee2mqtt tries to use port.
2. Look for the port assigned to Sonoff "E" adapter, and don't forget "ezsp" parameter (see last line at the end). 
3. Try zigbee2mqtt configuration that is below (use Your IP and password, check other things, and after some (3) messages in the log, just try to open web-interface):

data_path: /config/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
base_topic: zigbee2mqtt
server: mqtt://192.168.1.10:1883
advanced:
log_level: error
pan_id: 6744
channel: 11
network_key:
- 1
- 3
- 5
- 7
- 9
- 11
- 13
- 15
- 0
- 2
- 4
- 6
- 8
- 10
- 12
- 13
availability_blocklist: []
availability_passlist: []
last_seen: ISO_8601
homeassistant_legacy_entity_attributes: false
legacy_api: false
legacy_availability_payload: false
device_options:
legacy: false
blocklist: []
passlist: []
queue: {}
frontend:
port: 8099
experimental: {}
socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: '-d -d'
log: false user: mqtt-user
password: devicepassword
serial:
port: /dev/ttyACM0
user: mqtt-user
password: devicepassword
server: mqtt://192.168.1.10:1883
adapter: ezsp

brantmacga, 2022-11-25:
> specify
> `adapter: ezsp`
> in the configuration, next to the port address

this was the answer! Thank you!

closed 2022-11-25 by brantmacga