Koenkk/zigbee2mqtt #15176: TS0044 / _TZ3000_ee8nrt2l bind not possibe

ra-kal, 2022-11-25:
### What happened?

I'm not able to bind the wireless 4 button switch with any LED lamp. For all source points in web UI it allows me to bind only Ota and genTime clusters.

### What did you expect to happen?

I'm a newbe, so maybe I'm understanding it wrong. But it makes no sense for me that a button only device should contain input clusters. Actually I would expect the genOnOff cluster should be in Output clusters.

In clusters it looks like this (sorry, I have not found a way to export the clusters)

- Endpoint 1
  - Output clusters
    - genOta
    - genTime
  - Input clusters
    - genOnOff
    - genPowerCfg
    - genOnOff
    - genBasic
 - Endpoint 2
   - Input clusters
     - genPowerCfg
     - genOnOff
     - 57344
- Enpoint 3 and 4 are same as Endpoint 2

### How to reproduce it (minimal and precise)

In Bind tab select "source endpoint 1". Only "Ota" and "genTime" clusters are available for binding.

### Zigbee2MQTT version

1.28.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

MattWestb, 2022-11-25:
All TS004X exept TS004F is not one light controller they is sending  tuya commands that light is not understanding and the On-Off cluster is as in = reviving commands and not sending (= wrong made for light controlling devices).

Then TS004F is have 2 modes one "tuya scene" = Event mode for sending clicks  and one command mode = sending light commands to groups or devices.

If you successful doing one group binding to one light group all lights in the group is not understanding the commands then they is not Zigbee standard and will being ignored of all devices.

So for binding controllers to light groups you shall using light  controllers and the controlling is working also then the host system is offline and not something else that is sending click clicks.

ra-kal, 2022-11-25:
Thanks for the reply. Unfortunately I do not understand exactly what you write. 
I understand, that TS004X are not fully Zigbee compatible. It is a pitty, that this is not mentioned in the Zigbee2Mqtt devices list.
https://www.zigbee2mqtt.io/devices/TS0044.html

> All TS004X exept TS004F is not one light controller they is sending tuya commands that light is not understanding and the On-Off cluster is as in = reviving commands and not sending (= wrong made for light controlling devices).

I do not understand. Is the TS0044 publishing wrong clusters, or is it a bug in Z2M? I do not know how exactly Zigbee network works. I understand I can link two devices to send the commands directly from one. It was my intention to do it. But I do not know, if the cluster tree is published by the devices itself or if it is created in the Z2M based on the device identification.

> If you successful doing one group binding to one light group all lights in the group is not understanding the commands then they is not Zigbee standard and will being ignored of all devices.

At the moment I do not want group control. I just wanted to link one button to one RGB LED controller to test, how fast the on/off switching can be. The reason was, that if I switch the LED controller on/off in Home Assistant or via Z2M, it takes pretty long (more than 1s). So my only intention was to test, how fast LED controller reacts on changes. It is my intention to control some lights from HA web UI only and I hate when it is not immediate.

> So for binding controllers to light groups you shall using light controllers and the controlling is working also then the host system is offline and not something else that is sending click clicks.

I understand, but how I find out, which are the supported one? It is even difficult to find the exact model just from picture. Actually the 4 button switch I have does not look like the one on the picture but like a remote control with 4 buttons in a row.

closed 2023-01-02 by github-actions[bot]