Koenkk/zigbee2mqtt #15179: Raspberry 3 model B, fresh install fails with gyp / @serialport/bindings@9.2.8 install error

rduivenvoorde, 2022-11-25:
### What happened?

Thanks for this great project!!!

I've run zigbee2mqtt succesfully on other boxes I have, but now want it to run/try on a Raspberry 3 model B:

```
cat /etc/debian_version 
11.5
uname -a
Linux kast 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux
node --version
v18.12.1
npm --version
9.1.2
```

I keep having the serial port install issues (I see a lot in older issues). But fail to get things working:

- I've update node/npm to a recent version
-  I've succesfully installed serial-port package as described here: https://serialport.io/docs/guide-installation
- AND enabled serial port as written there...
- Installed all the needed packages to compile (build essentials etc etc).

But I keep having the errors like:

```
942 timing command:ci Completed in 103972ms
943 verbose stack Error: command failed
943 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:53:27)
943 verbose stack     at ChildProcess.emit (node:events:513:28)
943 verbose stack     at maybeClose (node:internal/child_process:1091:16)
943 verbose stack     at ChildProcess._handle.onexit (node:internal/child_process:302:5)
944 verbose pkgid @serialport/bindings@9.2.8
945 verbose cwd /home/geo/git/zigbee2mqtt
946 verbose Linux 5.15.76-v7+
947 verbose node v18.12.1
948 verbose npm  v9.1.2
949 error code 1
950 error path /home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings
951 error command failed
952 error command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
953 error make: Entering directory '/home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings/build'
953 error   CXX(target) Release/obj.target/bindings/src/serialport.o
953 error make: Leaving directory '/home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings/build'
954 error prebuild-install WARN install No prebuilt binaries found (target=18.12.1 runtime=node arch=arm libc= platform=linux)
954 error gyp info it worked if it ends with ok
954 error gyp info using node-gyp@9.3.0
954 error gyp info using node@18.12.1 | linux | arm
954 error gyp info find Python using Python version 3.9.2 found at "/usr/bin/python3"
954 error gyp info spawn /usr/bin/python3
954 error gyp info spawn args [
954 error gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
954 error gyp info spawn args   'binding.gyp',
954 error gyp info spawn args   '-f',
954 error gyp info spawn args   'make',
954 error gyp info spawn args   '-I',
954 error gyp info spawn args   '/home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings/build/config.gypi',
954 error gyp info spawn args   '-I',
954 error gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
954 error gyp info spawn args   '-I',
954 error gyp info spawn args   '/home/geo/.cache/node-gyp/18.12.1/include/node/common.gypi',
954 error gyp info spawn args   '-Dlibrary=shared_library',
954 error gyp info spawn args   '-Dvisibility=default',
954 error gyp info spawn args   '-Dnode_root_dir=/home/geo/.cache/node-gyp/18.12.1',
954 error gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
954 error gyp info spawn args   '-Dnode_lib_file=/home/geo/.cache/node-gyp/18.12.1/<(target_arch)/node.lib',
954 error gyp info spawn args   '-Dmodule_root_dir=/home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings',
954 error gyp info spawn args   '-Dnode_engine=v8',
954 error gyp info spawn args   '--depth=.',
954 error gyp info spawn args   '--no-parallel',
954 error gyp info spawn args   '--generator-output',
954 error gyp info spawn args   'build',
954 error gyp info spawn args   '-Goutput_dir=.'
954 error gyp info spawn args ]
954 error gyp info spawn make
954 error gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
954 error ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
954 error ../src/serialport.cpp:78:51: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_wor>
954 error    78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
954 error       |                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
954 error ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
954 error ../src/serialport.cpp:135:53: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_wo>
954 error   135 |   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
954 error       |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
954 error In file included from ../../../nan/nan_callbacks.h:83,
954 error                  from ../../../nan/nan.h:180,
954 error                  from ../src/./serialport.h:13,
954 error                  from ../src/serialport.cpp:1:
954 error ../../../nan/nan_callbacks_12_inl.h: In function ‘void Nan::imp::FunctionCallbackWrapper(const v8::FunctionCallbackInfo<v8::Value>&)’:
954 error ../../../nan/nan_callbacks_12_inl.h:169:6: internal compiler error: Segmentation fault
954 error   169 | void FunctionCallbackWrapper(const v8::FunctionCallbackInfo<v8::Value> &info) {
954 error       |      ^~~~~~~~~~~~~~~~~~~~~~~
954 error 0x769f08ff ???
954 error _ ../sysdeps/unix/sysv/linux/arm/sigrestorer.S:64
954 error Please submit a full bug report,
954 error with preprocessed source if appropriate.
954 error Please include the complete backtrace with any bug report.
954 error See <file:///usr/share/doc/gcc-10/README.Bugs> for instructions.
954 error make: *** [bindings.target.mk:114: Release/obj.target/bindings/src/serialport.o] Error 1
954 error gyp ERR! build error·
954 error gyp ERR! stack Error: `make` failed with exit code: 2
954 error gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:203:23)
954 error gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
954 error gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
954 error gyp ERR! System Linux 5.15.76-v7+
954 error gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
954 error gyp ERR! cwd /home/geo/git/zigbee2mqtt/node_modules/@serialport/bindings
954 error gyp ERR! node -v v18.12.1
954 error gyp ERR! node-gyp -v v9.3.0
954 error gyp ERR! not ok
955 verbose exit 1
956 timing npm Completed in 104260ms
957 verbose unfinished npm timer reify 1669380890822
958 verbose unfinished npm timer reify:build 1669380951178
959 verbose unfinished npm timer build 1669380951191
960 verbose unfinished npm timer build:deps 1669380951193
961 verbose unfinished npm timer build:run:install 1669380951408
962 verbose unfinished npm timer build:run:install:node_modules/@serialport/bindings 1669380951410
963 verbose code 1
964 error A complete log of this run can be found in:
```

The complete log is attached

Anybody has an idea to get this going?

### Zigbee2MQTT version

master version e66fe8534e81b6199dbb4dfdacaf3e5b8ba4210b / 1.28.2

### Adapter firmware version

20190425

### Adapter

CC2531

### Debug log

[zigbee2mqtt.zip](https://github.com/Koenkk/zigbee2mqtt/files/10093052/zigbee2mqtt.zip)

rduivenvoorde, 2022-11-26:
I gave up on this one. Got the feeling that RasberryPi 3B and Node are not playing well together (or at least the serial part of Node...).

If somebody  is able to make this work (with current Raspbian/Node etc etc), please let me know below in a comment...

formtapez, 2022-12-04:
I got a similar error with `@serialport` on ArchLinux with nodejs v19.2.0 and npm v9.1.3.
As a workaround i edited `package-lock.json` and changed the marked line as follows:

```
   "node_modules/@serialport/bindings": {
      "version": "9.2.8",
      "resolved": "https://registry.npmjs.org/@serialport/bindings/-/bindings-9.2.8.tgz",
      "integrity": "sha512-hSLxTe0tADZ3LMMGwvEJWOC/TaFQTyPeFalUCsJ1lSQ0k6bPF04JwrtB/C81GetmDBTNRY0GlD0SNtKCc7Dr5g==",
      "hasInstallScript": true,
      "dependencies": {
        "@serialport/binding-abstract": "9.2.3",
        "@serialport/parser-readline": "9.2.4",
        "bindings": "^1.5.0",
        "debug": "^4.3.2",
        "nan": "github:jkleinsc/nan#remove_accessor_signature",   <------- !!!!!!!!
        "prebuild-install": "^7.0.0"
      },

```

Thanks to @MRayermannMSFT for this idea at https://github.com/electron/electron/issues/35193#issuecomment-1212384522

Koenkk, 2022-12-04:
Serialport has been updated to 10.5.0 now, can you check if this fixes the issue?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

rduivenvoorde, 2022-12-04:
@Koenkk I pulled latest master, cleaned up my ~/.cache/... and tried `npm ci` again. About the same result.

About logging: what is it that you want to see?  I've not yet started zigbee2mqtt there as I'm not able to get it installed?
Or am I missing your point?

I'm not a node-person, so it is very well possible that I'm making a mess of my node setup...?

Koenkk, 2022-12-05:
You should pull the latest dev  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-01-12 by github-actions[bot]