Koenkk/zigbee2mqtt #15180: TS0601_rcbo

CheFuma, 2022-11-25:
### What happened?

I bought a RCBO 3P , designed for Tuya zigbee . made by HOCH.

In the manual it marked as ZJSBL7-100Z
but zigbee2mqtt recognise it like ZJSBL7-100Z.

also. in  the tuya app i can use all functionality: monitoring current, voltage, power, cumulative power, ad all of this py every phase. but! z2M can provide access only to common comsumption,  and i can't monitor each phase. 

### Zigbee2MQTT version

1.28.2-1 

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

Zigbee2MQTT:info  2022-11-25 21:59:53: Ввод в дом (0x84ba20fffe323182): TS0601_rcbo - TuYa DIN mount RCBO with smart energy metering (Router)
Zigbee2MQTT:info  2022-11-25 21:59:55: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/meter_number/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"name":"Ввод в дом meter number","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_meter_number_zigbee2mqtt","value_template":"{{ value_json.meter_number }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:55: MQTT publish: topic 'homeassistant/switch/0x84ba20fffe323182/switch_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Ввод в дом/set/state","device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"name":"Ввод в дом state","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_switch_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:55: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/alarm/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"name":"Ввод в дом alarm","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_alarm_zigbee2mqtt","value_template":"{{ value_json.alarm }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:55: MQTT publish: topic 'homeassistant/switch/0x84ba20fffe323182/switch_trip/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Ввод в дом/set/trip","device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"name":"Ввод в дом trip","payload_off":"clear","payload_on":"trip","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_switch_trip_zigbee2mqtt","value_template":"{{ value_json.trip }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:55: MQTT publish: topic 'homeassistant/switch/0x84ba20fffe323182/switch_child_lock/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Ввод в дом/set/child_lock","device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"entity_category":"config","icon":"mdi:account-lock","name":"Ввод в дом child_lock","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_switch_child_lock_zigbee2mqtt","value_template":"{{ value_json.child_lock }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"Ввод в дом power on behavior","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/select/0x84ba20fffe323182/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Ввод в дом/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"entity_category":"config","icon":"mdi:power-settings","name":"Ввод в дом power on behavior","options":["off","on","previous"],"state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/countdown_timer/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"enabled_by_default":false,"name":"Ввод в дом countdown timer","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_countdown_timer_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.countdown_timer }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/number/0x84ba20fffe323182/countdown_timer/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Ввод в дом/set/countdown_timer","device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"max":86400,"min":0,"name":"Ввод в дом countdown timer","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_countdown_timer_zigbee2mqtt","unit_of_measurement":"s","value_template":"{{ value_json.countdown_timer }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/voltage_rms/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"enabled_by_default":true,"name":"Ввод в дом voltage rms","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_voltage_rms_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage_rms }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","name":"Ввод в дом current","state_class":"measurement","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
Zigbee2MQTT:info  2022-11-25 21:59:56: MQTT publish: topic 'homeassistant/sensor/0x84ba20fffe323182/current_average/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"enabled_by_default":true,"name":"Ввод в дом current average","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_current_average_zigbee2mqtt","unit_of_measurement":"A","value_templat [... cut ...] lue_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x84ba20fffe323182"],"manufacturer":"TuYa","model":"DIN mount RCBO with smart energy metering (TS0601_rcbo)","name":"Ввод в дом"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Ввод в дом linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Ввод в дом","unique_id":"0x84ba20fffe323182_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

CheFuma, 2022-12-30:
What should i comment? I reported bug, but i have no reactions back.

ericvb, 2023-01-14:
Hi @CheFuma,
I was searching the issues to see if there was already an issue for my problem and I saw your issue.

Generally what can help to get answers from others, is a title covering/explaining in short the problem...

When I search your device in the Z2H website in supported devices : https://www.zigbee2mqtt.io/devices/TS0601_rcbo.html
I don't see any mention about supporting values per phase in the description, so I fear this is not (yet?) supported.

I recommend you close this issue and create a new one to ask to extend support for this device.
If you have technical knowledge, you can try to extend yourselves by reading this : https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

CheFuma, 2023-01-29:
hi, @ericvb 

 I'm absolutely sure it can support by-phase monitoring, because int the native app i can do it.
 
  ok. I'll create new task to extend support of this device.

closed 2023-01-29 by CheFuma