Koenkk/zigbee2mqtt #15181: TV02 window_open after ON to OFF

Wikibear, 2022-11-25:
### What happened?

After a time  I can't switch on window_open. After ON it needs 2-3 sec and switch to OFF.

Same as reported here:
https://github.com/Koenkk/zigbee2mqtt/issues/14163

After restart of z2m it will work again.

### Zigbee2MQTT version

1.28.2-dev commit: 3b221d4f

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

vladi1234, 2022-11-26:
https://github.com/Koenkk/zigbee2mqtt/issues/15164#issuecomment-1326806395

Koenkk, 2022-11-26:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/15164

closed 2022-11-26 by Koenkk