Koenkk/zigbee2mqtt #15182: Moes IP55 switch (see #12507)

markxroberts, 2022-11-25:
### Link

https://www.moeshouse.com/en-gb/products/ip55-waterproof-smart-zigbee-scene-switch-wireless-remote-dimmer?_pos=1&_sid=04466010a&_ss=r

### Database entry

{"id":82,"type":"EndDevice","ieeeAddr":"0xdc8e95fffece3217","nwkAddr":40327,"manufId":4098,"manufName":"_TZ3000_ja5osu5g","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096,57345],"outClusterList":[25,10,3,4,6,8,4096],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":1,"65534":0,"modelId":"TS004F","manufacturerName":"_TZ3000_ja5osu5g","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"manuSpecificTuya_3":{"attributes":{"53265":"�\u0000\u0003�\u0003�\u0000Z\u0003�\u0003�\u0000\u0003�\u0000d\u0000�\u0003�\u0003�\u0000\u0001�\u0003 \u0001\u000e\u0003�\u0003�\u0001\u0001�\u0000\u0000\u0000�\u0000�\u0000�"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff077d3e","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff077d3e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":927531739},"lastSeen":1669398163230,"defaultSendRequestWhen":"immediate"}

### Comments

I have created a new external converter based on the template at #12507
The manufacturer ID for this is _TZ3000_ja5osu5g
This exposes the read endpoints, but doesn't expose the configuration endpoint.

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        fingerprint: [{modelID: 'TS004F', manufacturerName: '_TZ3000_ja5osu5g'}],
        model: 'ERS-10TZBVB-AA',
        vendor: 'Moes',
        description: 'Smart Button',
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold'])],
        fromZigbee: [fz.battery, fz.tuya_on_off_action],
        toZigbee: [tz.tuya_operation_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', [0x0004, 0x000, 0x0001, 0x0005, 0x0007, 0xfffe]);
            await endpoint.write('genOnOff', {'tuyaOperationMode': 1});
            await endpoint.read('genOnOff', ['tuyaOperationMode']);
            try {
                await endpoint.read(0xE001, [0xD011]);
            } catch (err) {/* do nothing */}
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        },
    },
];
```

Silvenga, 2022-12-26:
(saving the issue from the stale issue bot)

What's needed for this to be merged? Do we just need a PR e.g. merged into the Tuya js file in the coverters repo?

markxroberts, 2022-12-26:
Thanks for the prompt. Had forgotten about this.  Yes, just needs a single extra line identifying this particular model, I think.  I’ll take a look.

markxroberts, 2022-12-26:
Addressed in #15730 and closed

closed 2022-12-26 by markxroberts

k-zakhariy, 2023-02-17:
For some reason it can’t be paired for me

2023-02-17 20:55:55Failed to interview '0x70ac08fffe2565ed', device has not successfully been paired
Info 2023-02-17 20:55:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70ac08fffe2565ed","ieee_address":"0x70ac08fffe2565ed","status":"failed"},"type":"device_interview"}'
Info 2023-02-17 20:55:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x70ac08fffe2565ed"},"type":"pairing"}'

naamah75, 2023-06-24:
When I try to switch from command mode to event mode I get the following error:
`Publish 'set' 'operation_mode' to 'Interruttore (sala)' failed: 'Error: Write 0x70ac08fffea11972/1 genOnOff({"tuyaOperationMode":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`