Koenkk/zigbee2mqtt #15184: Device model 'TS0601' and manufacturer name '_TZE200_atpwqgml' is NOT supported

RtoRoc, 2022-11-25:
### Is your feature request related to a problem? Please describe

Good evening everyone. It's my first issue so forgive me if I'm wrong or omit something.
I'm trying to pair a device 'TS0601' and manufacturer name '_TZE200_atpwqgml' .
although the paring goes well. the log always gives me this error: Device '0xa4c138526908fb43' with Zigbee model 'TS0601' and manufacturer name '_TZE200_atpwqgml' is NOT supported
how can I solve this problem???

thank you all

### Describe the solution you'd like

I would like the device to be supported

### Describe alternatives you've considered

non conosco altre soluzioni

### Additional context

.

danieledwardgeorgehitchcock, 2022-11-26:
Hi,

Please could you provide a link to the device that you are trying to pair? TS0601 means a wide array of devices in TuYa land...

Could you also upload the database.db entry too?

alesoft73, 2022-11-28:
I think work with this .js..

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_atpwqgml']),
    model: 'TS0601_switch_3',
    vendor: 'TuYa',
    description: 'Zigbee 3 gang smart switch',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1'), 
        e.switch().withEndpoint('l2'), 
        e.switch().withEndpoint('l3')
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;

alesoft73, 2022-11-28:
I have add with pull request ;)
https://github.com/Koenkk/zigbee-herdsman-converters/compare/master...alesoft73:zigbee-herdsman-converters:patch-1

closed 2023-01-05 by github-actions[bot]