Koenkk/zigbee2mqtt #15191: [New device support]: Tuya Soil Sensor _TZE200_myd45weu

kris969, 2022-11-26:
### Link

https://fr.aliexpress.com/item/1005004705106578.html?spm=a2g0o.order_list.0.0.6e935e5bqgigNf&gatewayAdapt=glo2fra

### Database entry

{"id":29,"type":"EndDevice","ieeeAddr":"0xa4c138bcc914bc7f","nwkAddr":34574,"manufId":4417,"manufName":"_TZE200_myd45weu","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"'�\u0014+i(�\u0014+i+�\u0014+i,�\u0014+i;�\u0014+i<�\u0014+i��\u0014+f��\u0014+\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff071444","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1669455534902,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
I have just receive this Tuya Soil Sensor that seems to be new on the market.
I tried to pair it using the dev branch, but it is not recognized yet.

Although I was coding in the past, I'm not comfortable with js. I begun writing an external converter but I thing it will be more efficient if someone else will do this job.

I give you my first work on converter, even job is'nt finished.

I give you here after the debug log I got:

2022-11-26 10:07:20Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":57856}' from endpoint 1 with groupID null

2022-11-26 10:07:21Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":58112}' from endpoint 1 with groupID null

many thanks for all ;)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_myd45weu'
        },
    ],
    model: 'QT-07S',
    vendor: 'Unknown',
    description: 'Soil Sensor',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    // Here you should put all functionality that your device exposes
    exposes: [],

};

module.exports = definition;
```

kris969, 2022-11-26:
Here after an extract of the relative tuya dump file:

1669453325405 0xa4c138bcc914bc7f c800 00 0f 02 00 00 00 64
1669453325614 0xa4c138bcc914bc7f c900 00 0e 04 02
1669453346768 0xa4c138bcc914bc7f ca00 00 0f 02 00 00 00 64
1669453346988 0xa4c138bcc914bc7f cb00 00 0e 04 02
1669453368112 0xa4c138bcc914bc7f cc00 00 0f 02 00 00 00 64
1669453368338 0xa4c138bcc914bc7f cd00 00 0e 04 02
1669453409403 0xa4c138bcc914bc7f ce00 00 0f 02 00 00 00 64
1669453410215 0xa4c138bcc914bc7f cf00 00 0e 04 02
1669453430621 0xa4c138bcc914bc7f d000 00 0f 02 00 00 00 64
1669453430762 0xa4c138bcc914bc7f d100 00 0e 04 02
1669453471848 0xa4c138bcc914bc7f d200 00 0f 02 00 00 00 64
1669453472598 0xa4c138bcc914bc7f d300 00 0e 04 02
1669453474154 0xa4c138bcc914bc7f d400 00 0f 02 00 00 00 64
1669453474264 0xa4c138bcc914bc7f d500 00 0e 04 02
1669453513475 0xa4c138bcc914bc7f d600 00 0f 02 00 00 00 64
1669453513704 0xa4c138bcc914bc7f d700 00 0e 04 02
1669453534712 0xa4c138bcc914bc7f d800 00 0f 02 00 00 00 64
1669453534925 0xa4c138bcc914bc7f d900 00 0e 04 02
1669453536839 0xa4c138bcc914bc7f da00 00 0f 02 00 00 00 64
1669453537074 0xa4c138bcc914bc7f db00 00 0e 04 02
1669453557164 0xa4c138bcc914bc7f dc00 00 0f 02 00 00 00 64
1669453557928 0xa4c138bcc914bc7f dd00 00 0e 04 02
1669453578421 0xa4c138bcc914bc7f de00 00 0f 02 00 00 00 64
1669453578647 0xa4c138bcc914bc7f df00 00 0e 04 02
1669453599622 0xa4c138bcc914bc7f e000 00 0f 02 00 00 00 64
1669453599755 0xa4c138bcc914bc7f e100 00 0e 04 02
1669453640845 0xa4c138bcc914bc7f e200 00 0f 02 00 00 00 64
1669453641076 0xa4c138bcc914bc7f e300 00 0e 04 02
1669453642938 0xa4c138bcc914bc7f e400 00 0f 02 00 00 00 64
1669453643269 0xa4c138bcc914bc7f e500 00 0e 04 02
1669453683226 0xa4c138bcc914bc7f e600 00 0f 02 00 00 00 64
1669453683465 0xa4c138bcc914bc7f e700 00 0e 04 02
1669453704470 0xa4c138bcc914bc7f e800 00 0f 02 00 00 00 64
1669453704694 0xa4c138bcc914bc7f e900 00 0e 04 02
1669453706612 0xa4c138bcc914bc7f ea00 00 0f 02 00 00 00 64
1669453706748 0xa4c138bcc914bc7f eb00 00 0e 04 02
1669453745967 0xa4c138bcc914bc7f ec00 00 0f 02 00 00 00 64
1669453746214 0xa4c138bcc914bc7f ed00 00 0e 04 02
1669453773155 0xa4c138bcc914bc7f ee00 00 0f 02 00 00 00 64
1669453773298 0xa4c138bcc914bc7f ee00 00 0f 02 00 00 00 64
1669453773310 0xa4c138bcc914bc7f ee00 00 0f 02 00 00 00 64
1669453788458 0xa4c138bcc914bc7f f000 00 0f 02 00 00 00 64
1669453788620 0xa4c138bcc914bc7f f100 00 0e 04 02
1669453809677 0xa4c138bcc914bc7f f200 00 0f 02 00 00 00 64
1669453809898 0xa4c138bcc914bc7f f300 00 0e 04 02
1669453831035 0xa4c138bcc914bc7f f400 00 0f 02 00 00 00 64
1669453831171 0xa4c138bcc914bc7f f500 00 0e 04 02
1669453872369 0xa4c138bcc914bc7f f600 00 0f 02 00 00 00 64
1669453872582 0xa4c138bcc914bc7f f700 00 0e 04 02
1669453913606 0xa4c138bcc914bc7f f800 00 0f 02 00 00 00 64
1669453913822 0xa4c138bcc914bc7f f900 00 0e 04 02
1669453954805 0xa4c138bcc914bc7f fa00 00 0f 02 00 00 00 64
1669453955003 0xa4c138bcc914bc7f fb00 00 0e 04 02
1669453976169 0xa4c138bcc914bc7f fc00 00 0f 02 00 00 00 64
1669453976411 0xa4c138bcc914bc7f fd00 00 0e 04 02
1669453997452 0xa4c138bcc914bc7f fe00 00 0f 02 00 00 00 64
1669453997675 0xa4c138bcc914bc7f ff00 00 0e 04 02
1669454038722 0xa4c138bcc914bc7f 00 00 0f 02 00 00 00 64
1669454038947 0xa4c138bcc914bc7f 100 00 0e 04 02
1669454059871 0xa4c138bcc914bc7f 200 00 0f 02 00 00 00 64
1669454060123 0xa4c138bcc914bc7f 300 00 0e 04 02
1669454061996 0xa4c138bcc914bc7f 400 00 0f 02 00 00 00 64
1669454062130 0xa4c138bcc914bc7f 500 00 0e 04 02
1669454102320 0xa4c138bcc914bc7f 600 00 0f 02 00 00 00 64
1669454102551 0xa4c138bcc914bc7f 700 00 0e 04 02
1669454143574 0xa4c138bcc914bc7f 800 00 0f 02 00 00 00 64
1669454143827 0xa4c138bcc914bc7f 900 00 0e 04 02
1669454164746 0xa4c138bcc914bc7f a00 00 0f 02 00 00 00 64
[... 379 lines cut ...]
1669459567262 0xa4c138bcc914bc7f aa00 00 0f 02 00 00 00 64
1669459567488 0xa4c138bcc914bc7f ab00 00 0e 04 02
1669459569456 0xa4c138bcc914bc7f ac00 00 0f 02 00 00 00 64
1669459569595 0xa4c138bcc914bc7f ad00 00 0e 04 02
1669459609884 0xa4c138bcc914bc7f ae00 00 0f 02 00 00 00 64
1669459610105 0xa4c138bcc914bc7f af00 00 0e 04 02
1669459631192 0xa4c138bcc914bc7f b000 00 0f 02 00 00 00 64
1669459631908 0xa4c138bcc914bc7f b100 00 0e 04 02
1669459672477 0xa4c138bcc914bc7f b200 00 0f 02 00 00 00 64
1669459672694 0xa4c138bcc914bc7f b300 00 0e 04 02
1669459693961 0xa4c138bcc914bc7f b400 00 0f 02 00 00 00 64
1669459694175 0xa4c138bcc914bc7f b500 00 0e 04 02
1669459734365 0xa4c138bcc914bc7f b600 00 0f 02 00 00 00 64
1669459735129 0xa4c138bcc914bc7f b700 00 0e 04 02
1669459755673 0xa4c138bcc914bc7f b800 00 0f 02 00 00 00 64
1669459755807 0xa4c138bcc914bc7f b900 00 0e 04 02
1669459796958 0xa4c138bcc914bc7f ba00 00 0f 02 00 00 00 64
1669459797671 0xa4c138bcc914bc7f bb00 00 0e 04 02
1669459818208 0xa4c138bcc914bc7f bc00 00 0f 02 00 00 00 64
1669459818365 0xa4c138bcc914bc7f bd00 00 0e 04 02
1669459839493 0xa4c138bcc914bc7f be00 00 0f 02 00 00 00 64
1669459839698 0xa4c138bcc914bc7f bf00 00 0e 04 02
1669459860924 0xa4c138bcc914bc7f c000 00 0f 02 00 00 00 64
1669459861163 0xa4c138bcc914bc7f c100 00 0e 04 02
1669459882317 0xa4c138bcc914bc7f c200 00 0f 02 00 00 00 64

Koenkk, 2022-11-27:
I suggest to start with the following converter and then adding the datapoints to `tuyaDatapoints`. The hardest part is figuring out what the datapoints mean, usually this is done via the TuYa gatway: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_myd45weu']),
    model: 'TS0601_soil',
    vendor: 'TuYa',
    description: 'Soil sensor',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [14, 'dp_14', tuya.valueConverter.raw],
            [15, 'dp_15', tuya.valueConverter.raw],
        ],
    },
    whiteLabel: [
        {vendor: 'Yieryi', model: 'THE01840'},
    ],
};

module.exports = definition;
```

kris969, 2022-11-27:
Many thanks for your reply in such a short time!

On my side I am available to continue on this topic, and to be as useful as possible, even if ti's a new domain for me ;(

Using your template converter, it gives the following return, in the zigbee2mqtt state screen:
 "dp_14": 2,
 "dp_15": 100,
that is, of course, conform to tuya dump log

My assumption is that "100" must be battery level, cause the batteries I'm using are new.

I don't know what's value "2" is relative to. This sensor is supposed to report temperature, humidity an battery levels, but only values OxOE (dp_14) and OxOF (dp_15) are present in the dump log. I suppose the value "2" must be relative to humidity, a relative low value indeed. Actually it's rainy outside and the soil is very wet.
I don't understand why there is no data relative to temperature.

I tried to map these data in the expose section : exposes: [e.humidity(), e.battery() ]. It's evident that something is missing and that my knowledge need to be improved.

I will spend time on  https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html to identify more information about this sensor.

at least but not last, could you give me your understanding about this extract from dump log : 
1669454228416 0xa4c138bcc914bc7f 1000 00 0f 02 00 00 00 64
1669454228620 0xa4c138bcc914bc7f 1100 00 0e 04 02
1669454269599 0xa4c138bcc914bc7f 1200 00 0f 02 00 00 00 64
1669454269835 0xa4c138bcc914bc7f 1300 00 0e 04 02
1669454290986 0xa4c138bcc914bc7f 1400 00 0f 02 00 00 00 64
1669454291126 0xa4c138bcc914bc7f 1500 00 0e 04 02
1669454332351 0xa4c138bcc914bc7f 1600 00 0f 02 00 00 00 64
1669454332510 0xa4c138bcc914bc7f 1700 00 0e 04 02
1669454373686 0xa4c138bcc914bc7f 1800 00 0f 02 00 00 00 64
1669454373893 0xa4c138bcc914bc7f 1900 00 0e 04 02
1669454394911 0xa4c138bcc914bc7f 1a00 00 0f 02 00 00 00 64
1669454395048 0xa4c138bcc914bc7f 1b00 00 0e 04 02
1669454436173 0xa4c138bcc914bc7f 1c00 00 0f 02 00 00 00 64
1669454436411 0xa4c138bcc914bc7f 1d00 00 0e 04 02
1669454438334 0xa4c138bcc914bc7f 1e00 00 0f 02 00 00 00 64
1669454438453 0xa4c138bcc914bc7f 1f00 00 0e 04 02
1669454478711 0xa4c138bcc914bc7f 2000 00 0f 02 00 00 00 64
1669454478943 0xa4c138bcc914bc7f 2100 00 0e 04 02
1669454499997 0xa4c138bcc914bc7f 2200 00 0f 02 00 00 00 64
1669454500155 0xa4c138bcc914bc7f 2300 00 0e 04 02
1669454541256 0xa4c138bcc914bc7f 2400 00 0f 02 00 00 00 64
1669454541411 0xa4c138bcc914bc7f 2500 00 0e 04 02
1669454582513 0xa4c138bcc914bc7f 2600 00 0f 02 00 00 00 64

Koenkk, 2022-11-27:
The `exposes` part will come later, we first need to figure out the data points.

Are there any more datapoints send by the device? You can check this by activating the debug log, you should see `Datapoint XX not defined for..`

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

kris969, 2022-11-27:
My debug level is already activated ;)

Other datapoints seems not to be sent by the device.
so far, I haven't seen "Datapoint XX not defined for"

Here is the log  filtering debug info:
2022-11-27 14:06:53Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":12288}' from endpoint 1 with groupID null
2022-11-27 14:06:53Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":12544}' from endpoint 1 with groupID null
2022-11-27 14:06:55Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":12800}' from endpoint 1 with groupID null
2022-11-27 14:06:55Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":13056}' from endpoint 1 with groupID null
2022-11-27 14:07:35Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":13312}' from endpoint 1 with groupID null
2022-11-27 14:07:36Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":13568}' from endpoint 1 with groupID null
2022-11-27 14:08:17Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":13824}' from endpoint 1 with groupID null
2022-11-27 14:08:17Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":14080}' from endpoint 1 with groupID null
2022-11-27 14:08:58Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":14336}' from endpoint 1 with groupID null
2022-11-27 14:08:58Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":14592}' from endpoint 1 with groupID null
2022-11-27 14:09:39Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":14848}' from endpoint 1 with groupID null
2022-11-27 14:09:40Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":15104}' from endpoint 1 with groupID null
2022-11-27 14:10:01Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":15360}' from endpoint 1 with groupID null
2022-11-27 14:10:01Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":15616}' from endpoint 1 with groupID null
2022-11-27 14:10:42Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":15}],"seq":15872}' from endpoint 1 with groupID null
2022-11-27 14:10:42Received Zigbee message from 'GARDEN_SOIL_01', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14}],"seq":16128}' from endpoint 1 with groupID null

kris969, 2022-11-27:
here are technical specifications, 
available on: https://www.aliexpress.com/item/1005004705106578.html?spm=a2g0o.order_list.0.0.21ef5e5bfgxp4o&gatewayAdapt=glo2fra

Specifications:
Moisture measurement range:0~100%
Moisture accuracy: 0~50% (±3%) , 50~100% (±5%)
Temperature measurement range: -20°C~60°C
Temperature accuracy: 1°C
Communication protocol: Zigbee
APP response time: 60s
Power supply: 2 pcs AA dry batteries
Size: 120*180*46.5mm
Probe Length: 60mm
Protection class: IP67

kris969, 2022-11-28:
For your information, I tried to connect this device to a Tuya hub (G2H camera) with no success. I didn't find this device in Tuya sensors database.
The initial idea was to check if a new firmware is available for update.

Koenkk, 2022-11-29:
Without the datapoints there is little I can do here. Maybe you can try with the white rectangular TuYa hub.

kris969, 2022-12-02:
My hub was not the correct one!
I purchase a new one, it takes a couple of days to have it.
I follow your guide to identify datapoint IDs, and that's it :

humidity : 3
Current temperature : 4
Temp unit convert : 9 --> remark : I don't see what it could be relative to???
Battery level state : 14
Battery level : 15

kris969, 2022-12-02:
Because no temperature as been reported, I looked closer to understand why it was not working.
I found I have introduce a mistake in the temperature datapoint IDs I provided to you. Sorry for this waste of time ;(

So here is the right template:
humidity : 3
Current temperature : 5
Temp unit convert : 9 --> remark : I don't see what it could be relative to???
Battery level state : 14
Battery level : 15

after correction, the state becomes:
{
    "battery": 90,
    "battery_state": 2,
    "humidity": 41,
    "last_seen": "2022-12-02T20:13:20.551Z",
    "linkquality": 224,
    "temperature_unit": "celsius",
    "device": {
        "applicationVersion": 72,
        "dateCode": "",
        "friendlyName": "GARDEN_SOIL_01",
        "hardwareVersion": 1,
        "ieeeAddr": "0xa4c138bcc914bc7f",
        "manufacturerID": 4417,
        "manufacturerName": "_TZE200_myd45weu",
        "model": "TS0601_soil",
        "networkAddress": 23478,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "temperature": 11
}

my question is: why the temperature appears at the end, after the device block?
I still have an misunderstanding that's why temperature value is 11? Of course it's not a z2m issue cause I got exactly the same value with Tuya application. I will have a look later to check if this value is changing over the time.

kris969, 2022-12-03:
It is clearly explained: "Wireless soil temperature and humidity measuring instrument, which can measure soil humidity and temperature at the same time, and connect to Tuya APP through Zigbee."

It is confirm the humidity AND the temperature values are the ones from the soil.
The temperature accuracy is 1°C, that's enough. During the last night the external temperature drop from 6°C to 3°C in free air. During this time the soil sensor goes from 11°C to 10°C. That's normal, cause soil temperature move slowly due to inertia.

I will in the next days do some more tests with different batteries levels to confirm how the battery_state changes.

That's all ;)

Koenkk, 2022-12-03:
Great, note that we can also leave out the `battery_state` since we already have the `battery` (%)

kris969, 2022-12-03:
I have done it, this morning. It was not so long ;)
I got 2 values for `battery_state`:
when `battery_level` is reported as 90% or 100% --> `battery_state` is 2
when `battery_level` is reported as 10% or lower --> `battery_sate` is 0

I wanted to use my supply lab to easily change the voltage and find the intermediate threshold, it was not a good idea!
Due to a wrong action I killed the sensor ;(

I open the sensor, and I saw it's really well done.
As supposed, there is a very little Negative Temperature Coefficient (NTC) resistance that is placed in the metallic lower part of the sensor to measure soil temperature.

The damage is around a MOSFET transistor and 2 resistors in the neighborhood. The MOSFET is easy to replace cause reference is writing on, but it's not the case concerning resistor. I's know impossible to get the values!
I bought a new sensor. When I will receive it, I will be able to measure resistors values and to fix the damaged sensor.
So I will not be able to continue with you until I will have received the new one.

Yes, I think you can publish it.
Concerning `battery_state` maybe you can start with something like:
{2: 'battery_full',  1: 'battery_medium', 0: 'battery_low', 4}

In a couple of days I will be able to continue to validate `battery_sate` and I will come back to you.

denveronly, 2022-12-03:
looking for this topic
have the same sensor with no support

Koenkk, 2022-12-04:
I added the device!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kris969, 2022-12-04:
Great!

I will test it as soon as possible...

cloudbr34k84, 2022-12-19:
When will this be available in Z2M as i bought and its not supported??

cloudbr34k84, 2022-12-19:
> looking for this topic have the same sensor with no support

same

kris969, 2022-12-24:
Hi, I'm back ;)

It takes a long time to have the new device...
I have received it this morning, paired it to a Tuya hub, just to check if firmware was ok: it was the case.
Then I paired it to z2m
For battery and humidity values it's ok.
Unfortunately the temperature is not reporting in the z2m state frame even in the log. the value displayed is still zero.

Here after an example:
`Zigbee2MQTT:info  2022-12-24 11:13:34: MQTT publish: topic 'zigbee2mqtt/GARDEN_SOIL_01', payload '{"battery":100,"battery_state":"high","device":{"applicationVersion":72,"dateCode":"","friendlyName":"GARDEN_SOIL_01","hardwareVersion":1,"ieeeAddr":"0xa4c13805901f6d3e","manufacturerID":4417,"manufacturerName":"_TZE200_myd45weu","model":"TS0601_soil","networkAddress":30848,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"humidity":39,"last_seen":"2022-12-24T11:13:34.598Z","linkquality":255,"temperature_unit":"celsius"}'
`

The device picture is not available yet, but this must be normal, of course isn't totally complete at this step of integration.
I will be available for more investigations as needed.

kris969, 2022-12-24:
@cloudbr34k84 

You need to use the dev branch, in docker-compose:
image: koenkk/zigbee2mqtt:latest-dev

you will found this sensor is supported, its integration in z2m is on going.

kris969, 2022-12-24:
@Koenkk 

I have found where the issue is coming from.

with z2m version: 1.28.4-dev commit: [6d08a2f]

in file : zigbee-herdsman-converters/devices/tuya.js

we have:
```
{                                                                 
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_myd45weu']),
        model: 'TS0601_soil',                       
        vendor: 'TuYa',                                 
        description: 'Soil sensor',                                   
        fromZigbee: [tuya.fz.datapoints],                                                                                  
        toZigbee: [tuya.tz.datapoints],                                                                                    
        configure: tuya.configureMagicPacket,                                                                              
        exposes: [e.temperature(), e.humidity(), tuya.exposes.temperatureUnit(), e.battery(), tuya.exposes.batteryState()],
        meta: {                                                       
            tuyaDatapoints: [                                                
                [3, 'humidity', tuya.valueConverter.raw],                                                                  
                [4, 'temperature', tuya.valueConverter.divideBy10],          
                [9, 'temperature_unit', tuya.valueConverter.temperatureUnit],
                [14, 'battery_state', tuya.valueConverter.batteryState],
                [15, 'battery', tuya.valueConverter.raw],          
            ],                                                                                                             
        },                                                                                                                 
    },           
```
the line relative to temperature must be changed to:
`[5, 'temperature', tuya.valueConverter.divideBy10],   `

After having removed my device from z2m and pairing it again, it works !!!

So let me know when you will have time to integrate this last change in the dev branch, I will do a last check.

cloudbr34k84, 2022-12-24:
> @cloudbr34k84
> 
> You need to use the dev branch, in docker-compose: image: koenkk/zigbee2mqtt:latest-dev
> 
> you will found this sensor is supported, its integration in z2m is on going.

ill just wait for the non dev update to come though.

Koenkk, 2022-12-25:
Great, assumed this can be closed now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-12-25 by Koenkk

kris969, 2022-12-26:
Hi,

A last change about temperature that don't need to be devided.
I send a pull request around this, expecting I have done as it shoud?
let me know.

I made testing this morning: it's ok!

regards ;)

brianegge, 2023-05-30:
Thanks for the change! I purchased the same soil sensors, and they worked fine after I updated my setup.

fliespl, 2023-06-28:
I just bought one of these and seeing strange thing about temperature. It reports 27 outside while its really 19. Added negative calibration of 8 and it went down to 11 on next report.

Also precision doesn't seem to work on my end. Only get full degrees.

Currently on edge since it adds support for my model.

kris969, 2023-06-29:
The temperature reported is the soil one, don't compare it with the air
one. It takes a lot of time, after putting it in the soil, to be
stabilized.
The temperature precision is given to be around 1°C. Let's say it can
only be use to get an approximate soil temperature. 
I have 2 of them since 6 month, it gives useful results to manage
watering.

fliespl, 2023-06-29:
@kris969 thanks for explaining! That makes sense. Btw. Do you mind sharing thresholds used for irrigation?

kris969, 2023-06-29:
About thresholds, I am not sure to have the right tune, cause there is
a lake of rain where I am living and I need more feedback.

I can give you in the attach file the graph representing moisture in
green and rain drops in yellow.
Based on this, and considering soil aspect, I stop irrigation when
moisture is above 40%, it's my baseline for now. 

My moisture sensor is placed in the garden out of the scope of
irrigation. So I only consider water coming from rain. I am not saying
it must be done like this, just telling you how I used it until now.
Not sure I will not change this in the future.

We also have to consider the plants you have in your garden, some of
them needs more or less water to be healthy. so we could consider
different thresholds for different irrigation zones. 

So just to say there is not an unique solution, and we need time to
understand.

Finally, don't forget, the best way to do it is the one you prefer ;)

Enjoy!