Koenkk/zigbee2mqtt #15192: Aqara FP1 sensor immediately joins and leaves the network with Sonoff ZBDONGLE-E adapter (EFR32MG21)

olivr2s, 2022-11-26:
### What happened?

I tried several times to add the aqara FP1 sensor in Z2M and each time it is recognized and configured but immediately removed from the network.

[image]

### Zigbee2MQTT version

1.28.2-dev (commit #66c4ca0)

### Adapter firmware version

7.1.1.0 build 273

### Adapter

SONOFF ZBDONGLE-E (EFR32MG21)

### Debug log

info  2022-11-26 10:45:27: Logging to console and directory: '/app/data/log/2022-11-26.10-45-27' filename: log.txt
info  2022-11-26 10:45:27: Starting Zigbee2MQTT version 1.28.2-dev (commit #66c4ca0)
info  2022-11-26 10:45:27: Starting zigbee-herdsman (0.14.76)
info  2022-11-26 10:45:30: zigbee-herdsman started (resumed)
info  2022-11-26 10:45:30: Coordinator firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"1","product":9,"revision":"7.1.1.0 build 273"},"type":"EZSP v9"}'
info  2022-11-26 10:45:30: Currently 60 devices are joined:
info  2022-11-26 11:18:08: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"3u0de-4"}'
debug 2022-11-26 11:18:19: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Coordinator","time":254,"transaction":"3u0de-5","value":true}'
info  2022-11-26 11:18:19: Zigbee: allowing new devices to join via Coordinator.
info  2022-11-26 11:18:41: Device '0x54ef441000560b0c' joined
info  2022-11-26 11:18:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_joined"}'
info  2022-11-26 11:18:41: Starting interview of '0x54ef441000560b0c'
info  2022-11-26 11:18:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c","status":"started"},"type":"device_interview"}'
warn  2022-11-26 11:18:42: Device '0x54ef441000560b0c' left the network
info  2022-11-26 11:18:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_leave"}'
warn  2022-11-26 11:18:42: Device '0x54ef441000560b0c' left the network
info  2022-11-26 11:18:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_leave"}'
info  2022-11-26 11:18:58: Device '0x54ef441000560b0c' joined
info  2022-11-26 11:18:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_joined"}'
info  2022-11-26 11:18:58: Starting interview of '0x54ef441000560b0c'
info  2022-11-26 11:18:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c","status":"started"},"type":"device_interview"}'
debug 2022-11-26 11:18:58: Received Zigbee message from '0x54ef441000560b0c', type 'readResponse', cluster 'genBasic', data '{"swBuildId":""}' from endpoint 1 with groupID 0
info  2022-11-26 11:18:58: Successfully interviewed '0x54ef441000560b0c', device has successfully been paired
info  2022-11-26 11:18:58: Device '0x54ef441000560b0c' is supported, identified as: Xiaomi Aqara presence detector FP1 (regions not supported for now) (RTCZCGQ11LM)
info  2022-11-26 11:18:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara presence detector FP1 (regions not supported for now)","exposes":[{"access":5,"description":"Indicates whether the device detected presence","name":"presence","property":"presence","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Presence events: \"enter\", \"leave\", \"left_enter\", \"right_leave\", \"right_enter\", \"left_leave\", \"approach\", \"away\"","name":"presence_event","property":"presence_event","type":"enum","values":["enter","leave","left_enter","right_leave","right_enter","left_leave","approach","away"]},{"access":7,"description":"Monitoring mode with or without considering right and left sides","name":"monitoring_mode","property":"monitoring_mode","type":"enum","values":["undirected","left_right"]},{"access":7,"description":"The distance at which the sensor detects approaching","name":"approach_distance","property":"approach_distance","type":"enum","values":["far","medium","near"]},{"access":7,"description":"Different sensitivities means different static human body recognition rate and response speed of occupied","name":"motion_sensitivity","property":"motion_sensitivity","type":"enum","values":["low","medium","high"]},{"access":2,"description":"Reset the status of no presence","name":"reset_nopresence_status","property":"reset_nopresence_status","type":"enum","values":[""]},{"access":1,"description":"Temperature of the device","name":"device_temperature","property":"device_temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Number of power outages (since last pairing)","name":"power_outage_count","property":"power_outage_count","type":"numeric"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"RTCZCGQ11LM","options":[{"access":2,"description":"Calibrates the device_temperature value (absolute offset), takes into effect on next report of device.","name":"device_temperature_calibration","property":"device_temperature_calibration","type":"numeric"}],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c","status":"successful","supported":true},"type":"device_interview"}'
info  2022-11-26 11:18:58: Configuring '0x54ef441000560b0c'
debug 2022-11-26 11:18:58: Received Zigbee message from '0x54ef441000560b0c', type 'readResponse', cluster 'aqaraOpple', data '{"268":3}' from endpoint 1 with groupID 0
debug 2022-11-26 11:18:58: lumi.motion.ac01: Processed data into payload {"motion_sensitivity":"high"}
info  2022-11-26 11:18:58: MQTT publish: topic 'zigbee2mqtt/0x54ef441000560b0c', payload '{"device":{"applicationVersion":54,"dateCode":"20210121","friendlyName":"0x54ef441000560b0c","hardwareVersion":1,"ieeeAddr":"0x54ef441000560b0c","manufacturerID":4660,"manufacturerName":"aqara","model":"RTCZCGQ11LM","networkAddress":52026,"powerSource":"DC Source","softwareBuildID":"","stackVersion":2,"type":"EndDevice","zclVersion":3},"linkquality":255,"motion_sensitivity":"high"}'
warn  2022-11-26 11:18:58: Device '0x54ef441000560b0c' left the network
info  2022-11-26 11:18:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_leave"}'
warn  2022-11-26 11:18:58: Device '0x54ef441000560b0c' left the network
info  2022-11-26 11:18:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000560b0c","ieee_address":"0x54ef441000560b0c"},"type":"device_leave"}'
debug 2022-11-26 11:19:07: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":"Coordinator","time":254,"transaction":"3u0de-6","value":false}'
info  2022-11-26 11:19:07: Zigbee: disabling joining new devices.
debug 2022-11-26 11:19:07: Received Zigbee message from 'Coordinator', type 'commandNotification', cluster 'greenPower', data '{"data":[25,23,2,10,0,0],"type":"Buffer"}' from endpoint 242 with groupID null, ignoring since it is from coordinator
info  2022-11-26 11:19:07: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"device":"Coordinator","time":254,"value":false},"status":"ok","transaction":"3u0de-6"}'
error 2022-11-26 11:19:19: Failed to configure '0x54ef441000560b0c', attempt 3 (Error: Read 0x54ef441000560b0c/1 aqaraOpple([322], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 52026 - 1 - 22 - 64704 - 1 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7))

Hessenpower01, 2022-11-26:
Is this the actual firmware? Try updating.

olivr2s, 2022-11-26:
> Is this the actual firmware? Try updating.

This is the firmware from this github:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

Recommended in this thread:
https://github.com/Koenkk/zigbee-herdsman/issues/319

It's with this firmware that I have the best stability and integration of sensors.

closed 2023-01-25 by github-actions[bot]